pschatzmann/ESP32-A2DP #472: PCM5102 DAC board - additional information for wiki

benjaminaigner, 2023-09-07:
Dear @pschatzmann ,

thank you very much for this library and the additional documentation!

Regarding this board:
https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC#pcm5102-dac

I've got mine, but struggled to get something out of it.

I noticed 4 jumpers on the back side, which are connected to:
* __H1L__: FLT pin (GND)
* __H2L__: DEMP pin (3V3)
* __H3L__: XSMT pin (3V3)
* __H4L__: FMT pin (GND)

[image]

Connecting them as shown, this board works (or maybe differently, if you recommend otherwise)

I would suggest to add this info to the wiki (you can use this picture of course).

Greetings

pschatzmann, 2023-09-07:
Could you proved some more links that show what module exactly you talk about ?

closed 2023-09-07 by pschatzmann

joba-1, 2023-09-07:
I just reordered these, so I’m sure he means those:  

Just found this amazing item on AliExpress. Check it out! 2,66€  17％ Off | Interface I2S PCM5102A DAC Decoder GY-PCM5102 I2S Player Module For Raspberry Pi pHAT Format Board Digital PCM5102 Audio Board
https://a.aliexpress.com/_Ey4nsgb

There seems to be no manual or datasheet, so I had the same doubts when I first used them, cleared up by google: https://raspberrypi.stackexchange.com/a/94951

benjaminaigner, 2023-09-07:
> I just reordered these, so I’m sure he means those:
> 
> Just found this amazing item on AliExpress. Check it out! 2,66€ 17％ Off | Interface I2S PCM5102A DAC Decoder GY-PCM5102 I2S Player Module For Raspberry Pi pHAT Format Board Digital PCM5102 Audio Board https://a.aliexpress.com/_Ey4nsgb
> 
Exactly one of those, the front looks exactly like the picture from the wiki.

> There seems to be no manual or datasheet, so I had the same doubts when I first used them, cleared up by google: https://raspberrypi.stackexchange.com/a/94951

I should have searched a little bit longer before getting the multimeter :-).
In this thread, the jumpers have a default connection. In my case they were floating, so it was necessary to connect them.

> I don't see any reason why you would need to change the De-emphasis control or Soft mute control.
I don't see any reason to change it as well, but on these boards the pins are floating, and there is no statement in the datasheet for an internal pull up/down.

As these pins are not broken out on most boards, I thought they can be hardwired.

> Maybe your module has just a bug or something is not good with your wiring ?
After soldering the 4 jumpers, it works as expected, no sound output without soldered jumpers.

pschatzmann, 2023-09-07:
OK, got it. Unfortunately I don't find the module that I used for testing so that I can double check how it was delivered. But I am sure it was just working as purchased.

The documentation has been updated.

johangroe, 2023-11-27:
Hi, I suggest to add some information to the last paragrph regarding this board. Just a simple hint to check, if the board has the solder bridges already presoldered. (my board from ebay didn't have any solder bridge already soldered at all.)