pschatzmann/ESP32-A2DP #483: c:\Users\LENOVO\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSource.cpp:538:42: error: missing binary operator before token "("  #if ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(4, 4, 4)                                           ^  exit status 1  Compilation error: exit status 1

dhdhdhdhdhdhhd, 2023-10-05:
### Problem Description

error

### Device Description

esp32

### Sketch

```cpp
c:\Users\LENOVO\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSource.cpp:538:42: error: missing binary operator before token "("
 #if ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(4, 4, 4)
                                          ^

exit status 1

Compilation error: exit status 1
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-05:
Why don't you just upgrade to a recent version ?
We are corrently at 2.0.14!

pschatzmann, 2023-10-05:
I tried to reproduced your error with ESP32 2.0.4 and Arduino IDE 2.2.1 and I can't reproduce your error!

dhdhdhdhdhdhhd, 2023-10-05:
> 我试图用 ESP32 2.0.4 和 Arduino IDE 2.2.1 重现您的错误，但我无法重现这个！
I have already commented out some code, and it can compile successfully. The commented-out code is the following：
and it comes from BluetoothA2DPSource.cpp:538
/*#if ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(4, 4, 4)
  case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
    ESP_LOGI( BT_AV_TAG,"ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT");
    break;

  case ESP_BT_GAP_MODE_CHG_EVT:
    ESP_LOGI(BT_AV_TAG, "ESP_BT_GAP_MODE_CHG_EVT mode:%d",
             param->mode_chg.mode);
    break;
#endif*/

pschatzmann, 2023-10-05:
I would rather like to understand and reproduce why you are getting this error in order to fix it.
What Arduino Verion are you using ?

pschatzmann, 2023-10-05:
Are you using the right offical board url ?
https://espressif.github.io/arduino-esp32/package_esp32_index.json

pschatzmann, 2023-10-05:
Just tried to reproduce with Arduino 1.8.19 and I see that ESP32 2.0.4 is not supported.
So I am closing this issue

closed 2023-10-05 by pschatzmann

dhdhdhdhdhdhhd, 2023-10-05:
> Just tried to reproduce with Arduino 1.8.19 and I see that ESP32 2.0.4 is not supported! So I am closing this issue since this is working fine with all supported versions

2.2.2

dhdhdhdhdhdhhd, 2023-10-05:
> Just tried to reproduce with Arduino 1.8.19 and I see that ESP32 2.0.4 is not supported! So I am closing this issue since this is working fine with all supported versions

Arduino--2.2.2
esp32--1.0.4

pschatzmann, 2023-10-05:
Strange: I cant reproduce your issue.
Why are you still using this old version which has a lot of bugs that have been fixed

dhdhdhdhdhdhhd, 2023-10-06:
Thank you very much, the above compilation error problem has been solved, but there is another problem, the sound quality of the music played is relatively poor.

dhdhdhdhdhdhhd, 2023-10-06:
> Strange: I cant reproduce your issue. Why are you still using this old version which has a lot of bugs that have been fixed

Thank you very much, the above compilation error problem has been solved, but there is another problem, the sound quality of the music played is relatively poor.

pschatzmann, 2023-10-07:
Did you test the audio output properly before using my library ?
I am getting perfect quality!

truedat101, 2024-03-08:
Oddly , I see this same problem on 2.0.14 ESP and IDE 2.1.0.