pschatzmann/ESP32-A2DP #485: Weird noises when playing music from Spotify or YouTube

Razzorback225, 2023-10-06:
### Problem Description

Hi all,

I just tested the library today in a hurry and I just use a old board that I developped a few years ago.
This board uses a ESP32-WROOM32D module and I soldered a 3.5mm audio jack to the pins 25 and 26 of the module.
The Bluetooth part seems to work as I can connect to the ESP32, but the audio out is not working properly.
I captured what I can hear when plugin some earphones or speaker to the 3.5mm audio jack. [Sound](https://drive.google.com/file/d/19KzFXCmrwOesQEHzrlSfXby-xVJ5HFes/view?usp=sharing)

I used the latest release of the library.

### Device Description

Custom ESP32 board using an ESP32-WROOM32D (4MB of Flash)

### Sketch

```cpp
Sketch used comes from the examples folder of the Github.
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Can't remember at this time

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-07:
I am hardly hearing anything. 
Did you try to increase the volume on the phone or by calling set_volume() ? 
Looking at your picture, I can't see the ground connection to the audio jack

Razzorback225, 2023-10-09:
@pschatzmann Yeah sorry for that I tried to be as close as possible of the earphones when I recorded the sample.

I set the volume at max level on my phone when I tried but not in the code.

The jack is connected to ground through the PTH at its right. It is connected to ground.

The board you mentionned is the one with Yellow mask and TFT or the one above?

pschatzmann, 2023-10-09:
The one with the TFT screen is using the internal DAC of the ESP32 with an amplifier to output audio.
To be honest I tested it with my AudioTools. Maybe it helps if you do some tests
- https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-analog/streams-generator-analog.ino (if you dont get a proper sine signal you have a hardware issue)

mrhempman69, 2023-10-17:
What is the phone that you are using?

Razzorback225, 2023-10-26:
@mrhempman69 

Sorry for the late answer.

I've got an android phone (a Motorola Edge).
I'did not try anything else than these 2 applications. 
I put them in the title because they are the 2 applications I used to test the setup.

pschatzmann, 2023-10-26:
Maybe is it just noise from your power supply ? 
Did you try to power it from an (e.g. laptop) battery

Razzorback225, 2023-10-26:
I also tried to power it from 5V coming from USB to UART adapter but it makes no changes.

The sound I add to this topic is the only thing that comes out of the speakers when I play a song on my phone. There is no music just this sound

pschatzmann, 2023-10-26:
Did you try another ESP32 to figure out if the issue is your phone or your Processor ?
Did you run the test sketch to figure out if the issue is with the A2DP library ?

Razzorback225, 2023-10-26:
I did not tried yet on another ESP32 due to missing components.

I run the test sketch for internalDac and this was not working but I cannot guarantee that this does not have to do with my setup.

I tryied the library with an I2S DAC (PCM5102A) using a custom pin assignement and It worked like a charm.

Once I'll have all the component I'll try again.

CC0126, 2024-02-03:
This is due to the lack of CPU processing ability or the BREDR signal quality difference leading to the loss of blue sound frequency, different solutions:
1. Setting CPU speed 80MHz or higher (recommended 240MHz)
2. Increased DMA Buffer length
3.Increasing firing rate or shooting distance

closed 2024-04-06 by pschatzmann