pschatzmann/ESP32-A2DP #490: "a2dp_sink.set_address_validator(check);" function currently ALWAYS allows PAIRING

Rob58329, 2023-10-14:
### Problem Description

re: https://github.com/pschatzmann/ESP32-A2DP as at 14Oct23:

and: https://github.com/pschatzmann/ESP32-A2DP/wiki/Access-Protection---Authentication

A nice library - thank you.

Note that the "a2dp_sink.set_address_validator(check);" function does not currently work, as it always allows PAIRING what ever the "bool check(esp_bd_addr_t adr)" returns.
(I'm using Adruino 1.8.19 with github.com/espressif/arduino-esp32 v16Jan23,)

  

### Device Description

NodeMCU_32S

### Sketch

```cpp
BluetoothA2DPSink a2dp_sink;

bool check(esp_bd_addr_t adr){
  // this will prevent all connections
  return false;
}

void setup() {
  a2dp_sink.set_address_validator(check);
  a2dp_sink.start("MyMusic", false);  
}

void loop() {
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Adruino 1.8.19 with github.com/espressif/arduino-esp32 v16Jan23

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-14:
The application works as designed: The best thing I found to do, is to issue a disconnect when the validator fails. In order to do this validation I need to receive the address.

Please feel free to improve the functionality if you see any better solution.

closed 2023-10-14 by pschatzmann

Rob58329, 2023-10-19:
@pscatzmann:  Many thanks for grtting back to me.

-> My bad.  The above example seems to be working now.

Specifically although PAIRING IS ALLOWED, whenever you try to connect (after pairing) the connection either fails, or is immediately disconnected.  If however you change the above code from “return false;” to “return true;” then you can connect fine.

[I’m not sure why the above didn't seem to be the case when I originally tried it!...)

Thanks again.