pschatzmann/ESP32-A2DP #493: set_volume does not change the volume until the music is paused

mrhempman69, 2023-10-17:
### Problem Description

Volume changes are received by the device, but the actual volume doesn't change until the music is paused then played. It doesn't update the volume during streaming.

Based on serial logs, the set_volume inside A2DPVolumeControl doesn't fire when music is playing. Only when you pause the song, will "set_volume" function be fired.

### Device Description

ESP32 Wroom https://www.amazon.com/gp/product/B0C7C2HQ7P/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Sketch

```cpp
Part 1:

#include "BluetoothA2DPSink.h"
#include "./data/mrhemVolumeControl.h"
BluetoothA2DPSink a2dp_sink;
A2DPMrhemVolumeControl mrhemVolumeControl;
void setup() {   
 
  Serial.begin(9600);
  int mclkPin=3;
  i2s_pin_config_t external_pin_config = {
        .mck_io_num = mclkPin,
        .bck_io_num =33,
        .ws_io_num = 32,
        .data_out_num = 22,
        .data_in_num = I2S_PIN_NO_CHANGE,
    };
  static const i2s_config_t externalDAC_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
        .sample_rate =  44100, // corrected by info from bluetooth
        .bits_per_sample = (i2s_bits_per_sample_t) 16, 
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .use_apll = true,
        .tx_desc_auto_clear = true, // avoiding noise in case of data unavailability
        .mclk_multiple=(i2s_mclk_multiple_t) 256,

  };

  a2dp_sink.set_volume_control(&mrhemVolumeControl); 

  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
  bool externalDAC=true;

    a2dp_sink.set_pin_config(external_pin_config);
    a2dp_sink.set_i2s_config(externalDAC_config);
    a2dp_sink.i2s_mclk_pin_select(mclkPin);

  a2dp_sink.start("mrhemMusicBasic");
}
```

./data/mrhemVolumeControl.h:

```
#include "BluetoothA2DPSink.h"
#include "SoundData.h"
#include "esp_log.h"

class A2DPMrhemVolumeControl : public A2DPVolumeControl {

        public: A2DPMrhemVolumeControl() {
            volumeFactorMax = 4096;
        }

        virtual void set_volume(uint8_t volume) override {
            constexpr double base = 1.4;
            constexpr double bits = 12;
            constexpr double zero_ofs = pow(base, -bits);
            constexpr double scale = pow(2.0, bits);
            double volumeFactorFloat = (pow(base, volume * bits / 127.0 - bits) - zero_ofs) * scale / (1.0 - zero_ofs);
            volumeFactor = volumeFactorFloat;
            if (volumeFactor > 0x1000) {
                volumeFactor = 0x1000;
            }
            Serial.print("mrhem default Volume is now"); 
            Serial.println(volumeFactor);
        }
};

```
```

### Other Steps to Reproduce

I monited the serial output in the Arduino IDE's Serial Monitor. I press the volume change buttons on the iphone, but "mrhem default volume is now" does not output. It only outputs much later with the newest value, when I press pause or play.

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

esp32 by Espressif: 2.0.11

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-17:
In the light that this is working perfectly with the provided default implementation, I tend to assume that you messed something up!
 
What are you doing in the loop ?

mrhempman69, 2023-10-17:
The loop is nothing:
```
int8_t IntStat = 0;
int8_t IntConnected = 0;
int16_t runningMean=0;
int16_t absRunningMean=0;
void loop() {

   if (a2dp_sink.is_connected() != IntConnected) {
    IntConnected = a2dp_sink.is_connected();
    if (IntConnected == 0) {
      Serial.println("Not Connected");
    }
    if (IntConnected == 1) {
      Serial.println("Connected");
    }
  }
  if (a2dp_sink.get_audio_state() != IntStat) {
    IntStat = a2dp_sink.get_audio_state();
    if (IntStat == 0) {
      Serial.println("Stop");
    }
    if (IntStat == 2) {
      Serial.println("Play");
    }
  }

}
// btw, I have commented out the following function & the line in the "setup" function that references it, and problem still persists too:
void avrc_metadata_callback(uint8_t data1, const uint8_t *data2) {
  Serial.printf("AVRC metadata rsp: attribute id 0x%x, %s\n", data1, data2);
}
```

The reason for providing my own implementation, was only to determine why the volume wasn't working 😆. I have trimmed down to basically just the default implementation.

When I activate the verbose logging, there is a huge delay in the volume being changed. Also, the music is skipping because the controller seems like it is logging too much to also be able to stream correctly. However, the volume does change without me pressing pause, it just is like a 10 second delay.

pschatzmann, 2023-10-17:
Did you double check that you are not starving the A2DP task with your loop ?
Just add some delay calls in your loop ....

mrhempman69, 2023-10-17:
I thought that was going to fix it but even using ` void loop() { } ` did not help... 

Here is some of the log from when I press "play" and tap "volume up" about once per second:

```
16:18:12.065 -> [455564][D][BluetoothA2DPSink.cpp:1435] ccall_av_hdl_avrc_tg_evt(): [BT_AV] ccall_av_hdl_avrc_tg_evt
16:18:12.162 -> [455670][D][BluetoothA2DPSink.cpp:570] app_rc_ct_callback(): [BT_AV] app_rc_ct_callback
16:18:12.258 -> [455776][D][BluetoothA2DPSink.cpp:1443] av_hdl_avrc_tg_evt(): [BT_AV] av_hdl_avrc_tg_evt evt 3
16:18:12.354 -> [455869][D][BluetoothA2DPSink.cpp:587] app_rc_ct_callback(): [BT_AV] app_rc_ct_callback ESP_AVRC_CT_CHANGE_NOTIFY_EVT
16:18:12.483 -> [455969][I][BluetoothA2DPSink.cpp:1459] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 87%
16:18:12.611 -> [456093][D][BluetoothA2DPSink.cpp:358] app_work_dispatch(): [BT_API] app_work_dispatch event 0x4, param len 12
16:18:12.707 -> [456205][I][BluetoothA2DPSink.cpp:1399] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 87
16:18:12.835 -> [456322][D][BluetoothA2DPSink.cpp:390] app_send_msg(): [BT_AV] app_send_msg
16:18:12.931 -> mrhem default Volume is now2433
16:18:12.963 -> [456528][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:13.059 -> [456648][D][BluetoothA2DPSink.cpp:413] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x3
16:18:13.155 -> [456656][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:13.250 -> [456760][I][BluetoothA2DPSink.cpp:416] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:18:13.379 -> [456870][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:13.476 -> [456976][D][BluetoothA2DPSink.cpp:381] app_work_dispatched(): [BT_AV] app_work_dispatched
16:18:13.572 -> [457072][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:13.668 -> [457167][D][BluetoothA2DPSink.cpp:1435] ccall_av_hdl_avrc_tg_evt(): [BT_AV] ccall_av_hdl_avrc_tg_evt
16:18:13.765 -> [457278][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:13.861 -> [457369][D][BluetoothA2DPSink.cpp:1443] av_hdl_avrc_tg_evt(): [BT_AV] av_hdl_avrc_tg_evt evt 3
16:18:13.957 -> [457465][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:14.052 -> [457565][I][BluetoothA2DPSink.cpp:1459] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 93%
16:18:14.181 -> [457677][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:14.277 -> [457773][I][BluetoothA2DPSink.cpp:1399] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 93
16:18:14.405 -> [457869][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:14.500 -> mrhem default Volume is now3159
16:18:14.533 -> [458103][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:14.628 -> [458211][D][BluetoothA2DPSink.cpp:413] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x3
16:18:14.724 -> [458218][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:14.820 -> [458322][I][BluetoothA2DPSink.cpp:416] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:18:14.948 -> [458433][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:15.044 -> [458539][D][BluetoothA2DPSink.cpp:381] app_work_dispatched(): [BT_AV] app_work_dispatched
16:18:15.140 -> [458635][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:15.237 -> [458729][D][BluetoothA2DPSink.cpp:1435] ccall_av_hdl_avrc_tg_evt(): [BT_AV] ccall_av_hdl_avrc_tg_evt
16:18:15.333 -> [458841][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:15.430 -> [458932][D][BluetoothA2DPSink.cpp:1443] av_hdl_avrc_tg_evt(): [BT_AV] av_hdl_avrc_tg_evt evt 3
16:18:15.525 -> [459027][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:15.621 -> [459127][I][BluetoothA2DPSink.cpp:1459] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 100%
16:18:15.749 -> [459238][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:15.845 -> [459337][I][BluetoothA2DPSink.cpp:1399] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 100
16:18:15.941 -> [459433][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:16.037 -> mrhem default Volume is now4096
16:18:16.102 -> [459668][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:16.165 -> [459775][D][BluetoothA2DPSink.cpp:413] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x4
16:18:16.293 -> [459783][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:16.390 -> [459887][I][BluetoothA2DPSink.cpp:416] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:18:16.487 -> [459997][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback
16:18:16.583 -> [460103][D][BluetoothA2DPSink.cpp:381] app_work_dispatched(): [BT_AV] app_work_dispatched
16:18:16.678 -> [460199][D][A2DPVolumeControl.h:43] update_audio_data(): [VolumeControl] update_audio_data
16:18:16.774 -> [460294][D][BluetoothA2DPSink.cpp:1316] ccall_av_hdl_avrc_evt(): [BT_AV] ccall_av_hdl_avrc_evt
16:18:16.903 -> [460405][D][BluetoothA2DPSink.cpp:1116] audio_data_callback(): [BT_AV] audio_data_callback

```
The volume change is about 10 seconds late

pschatzmann, 2023-10-17:
Yes: an empty loop is equally bad!

mrhempman69, 2023-10-17:
Right true, I got that from the homepage, but i also just tried `void loop(){ delay(100); }` and it also did not help...

pschatzmann, 2023-10-17:
Did you check the log it you get the events in time from the IPhone ?

mrhempman69, 2023-10-17:
Well, I don't know how to check the timing of those events, but it works on my android tablet, also the audio quality is much better..... My iphone is fairly old too (iphone 6S) but it's not surprising knowing Apple that it would make up a problem like this for non-Apple bluetooth.... I will try that class.

mrhempman69, 2023-10-17:
The problem persists when using Apple Music (perhaps not related to  https://github.com/pschatzmann/ESP32-A2DP/issues/485 )

The problem persists in iOS 16.6 (iPhone SE)

mrhempman69, 2023-10-18:
This issue is likely related to the last few detailed comments in https://github.com/pschatzmann/ESP32-A2DP/issues/493, as the behaviors are quite similar

pschatzmann, 2023-10-19:
I committed a new version of BluetoothA2DPSinkQueued which might resolve your issues...
You will need to run your sketch with this class instead of the simple BluetoothA2DPSink!

Since I don't have any recent IPhone, please let me know if this help ?

mrhempman69, 2023-10-19:
I will do some more investigation, but I wanted to paste this log after doing what you suggested, in case there is an easy fix.

```
16:59:08.404 -> ��ѽ�ѡ�): [BT_AV] bluedroid initialized
16:59:08.489 -> [   838][I][BluetoothA2DPSink.cpp:338] init_bluetooth(): [BT_AV] bluedroid enabled
16:59:08.531 -> [   838][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:08.659 -> [   935][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
16:59:08.787 -> [   940][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
16:59:08.850 -> [  1050][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:08.978 -> [  1131][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
16:59:09.075 -> [  1252][I][BluetoothA2DPSink.cpp:643] av_hdl_a2d_evt(): [BT_AV] A2DP PROF STATE: Init Compl
16:59:09.171 -> 
16:59:09.171 -> [  1333][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
16:59:09.235 -> [  1050][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 117540
16:59:09.330 -> [  1514][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
16:59:13.905 -> [  6314][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 16
16:59:14.306 -> [  6699][I][BluetoothA2DPSink.cpp:497] app_gap_callback(): [BT_AV] authentication success: iPhone (16)
16:59:14.402 -> [  6813][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:14.530 -> [  6813][I][BluetoothA2DPSink.cpp:753] handle_connection_state(): [BT_AV] partner address: c0:b6:58:63:bf:5d
16:59:14.657 -> [  6913][I][BluetoothA2DPSink.cpp:755] handle_connection_state(): [BT_AV] A2DP connection state: Connecting, [c0:b6:58:63:bf:5d]
16:59:14.785 -> [  7049][I][BluetoothA2DPSink.cpp:814] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTING
16:59:14.881 -> [  7164][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:15.009 -> [  7285][I][BluetoothA2DPSink.cpp:661] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , codec type 0
16:59:15.105 -> [  7390][I][BluetoothA2DPSink.cpp:673] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , sample_rate 44100
16:59:15.233 -> [  7501][I][BluetoothA2DPSink.cpp:684] handle_audio_cfg(): [BT_AV] configure audio player 21-15-2-35
16:59:15.330 -> 
16:59:15.330 -> [  7609][I][BluetoothA2DPSink.cpp:692] handle_audio_cfg(): [BT_AV] audio player configured, samplerate=44100
16:59:15.457 -> [  7724][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:15.585 -> [  7844][I][BluetoothA2DPSink.cpp:1463] av_hdl_avrc_tg_evt(): [BT_AV] AVRC conn_state evt: state 1, [c0:b6:58:63:bf:5d]
16:59:15.714 -> [  7970][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:15.810 -> [  8091][I][BluetoothA2DPSink.cpp:1479] av_hdl_avrc_tg_evt(): [BT_AV] AVRC register event notification: 13, param: 0x0
16:59:15.938 -> [  8216][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:16.066 -> [  8337][I][BluetoothA2DPSink.cpp:753] handle_connection_state(): [BT_AV] partner address: c0:b6:58:63:bf:5d
16:59:16.195 -> [  8452][I][BluetoothA2DPSink.cpp:755] handle_connection_state(): [BT_AV] A2DP connection state: Connected, [c0:b6:58:63:bf:5d]
16:59:16.323 -> [  8586][I][BluetoothA2DPSink.cpp:819] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTED
16:59:16.419 -> [  8701][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable false
16:59:16.547 -> [  8816][I][BluetoothA2DPSinkQueued.cpp:6] bt_i2s_task_start_up(): [BT_API] ringbuffer data empty! mode changed: RINGBUFFER_MODE_PREFETCHING
16:59:16.706 -> [  8964][I][BluetoothA2DPSinkQueued.cpp:21] bt_i2s_task_start_up(): [BT_AV] BtI2STask Started
16:59:16.803 -> [  9063][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:16.899 -> [  9179][I][BluetoothA2DPSink.cpp:550] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT stat:0
16:59:17.027 -> [  9184][I][BluetoothA2DPSink.cpp:939] av_hdl_avrc_evt(): [BT_AV] AVRC conn_state evt: state 1, [c0:b6:58:63:bf:5d]
16:59:17.155 -> [  9295][I][BluetoothA2DPSink.cpp:552] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT remote name:iPhone (16)
16:59:17.283 -> [  9417][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:17.411 -> [  9667][I][BluetoothA2DPSink.cpp:984] av_hdl_avrc_evt(): [BT_AV] AVRC remote features 25b
16:59:17.507 -> [  9763][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:17.604 -> [  9884][I][BluetoothA2DPSink.cpp:1494] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote features 25b, CT features 2
16:59:17.731 -> [ 10000][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
16:59:17.859 -> [ 10120][I][BluetoothA2DPSink.cpp:992] av_hdl_avrc_evt(): [BT_AV] remote rn_cap: count 7, bitmask 0x1f06
16:59:17.955 -> [ 10260][I][BluetoothA2DPSink.cpp:559] app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT
[... cut ...]
16:59:35.037 -> [  1133][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
16:59:35.133 -> [  1254][I][BluetoothA2DPSink.cpp:643] av_hdl_a2d_evt(): [BT_AV] A2DP PROF STATE: Init Compl
16:59:35.229 -> 
16:59:35.229 -> [  1335][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
16:59:35.293 -> [  1052][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 117548
16:59:35.389 -> [  1517][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
```

pschatzmann, 2023-10-19:
Strange: I tested it on Android and there it was working. It seems that the I2S writes are not working, so the queue is never consumed.

I will compare your log with mine tomorrow...

mrhempman69, 2023-10-19:
The last log was using the PCM5102 breakout board from the wiki. I actually am not getting audio from that board. But this is what happens using a different external DAC (NAU8402) where i am getting audio (still i2s_write_data failed). 

```
17:26:53.116 -> [ 43314][I][BluetoothA2DPSink.cpp:692] handle_audio_cfg(): [BT_AV] audio player configured, samplerate=44100
17:26:53.212 -> [ 43428][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
17:26:53.341 -> [ 43549][I][BluetoothA2DPSink.cpp:753] handle_connection_state(): [BT_AV] partner address: c0:b6:58:63:bf:5d
17:26:53.469 -> [ 43664][I][BluetoothA2DPSink.cpp:755] handle_connection_state(): [BT_AV] A2DP connection state: Connected, [c0:b6:58:63:bf:5d]
17:26:53.596 -> [ 43798][I][BluetoothA2DPSink.cpp:819] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTED
17:26:53.693 -> [ 43913][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable false
17:26:53.821 -> [ 44028][I][BluetoothA2DPSinkQueued.cpp:6] bt_i2s_task_start_up(): [BT_API] ringbuffer data empty! mode changed: RINGBUFFER_MODE_PREFETCHING
17:26:53.982 -> [ 44176][I][BluetoothA2DPSinkQueued.cpp:21] bt_i2s_task_start_up(): [BT_AV] BtI2STask Started
17:26:54.078 -> [ 44275][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
17:26:54.174 -> [ 44300][I][BluetoothA2DPSink.cpp:550] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT stat:0
17:26:54.302 -> [ 44396][I][BluetoothA2DPSink.cpp:984] av_hdl_avrc_evt(): [BT_AV] AVRC remote features 25b
17:26:54.398 -> [ 44507][I][BluetoothA2DPSink.cpp:552] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT remote name:iPhone (16)
17:26:54.527 -> [ 44603][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
17:26:54.655 -> [ 44741][I][BluetoothA2DPSink.cpp:559] app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT
17:26:54.752 -> [ 44853][I][BluetoothA2DPSink.cpp:1494] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote features 25b, CT features 2
17:26:54.848 -> [ 44949][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 71896
17:26:54.944 -> [ 45065][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
17:26:55.073 -> [ 45278][I][BluetoothA2DPSink.cpp:992] av_hdl_avrc_evt(): [BT_AV] remote rn_cap: count 7, bitmask 0x1f06
17:26:56.871 -> [ 47199][I][BluetoothA2DPSink.cpp:559] app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT
17:26:56.967 -> [ 47200][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 72080
17:26:57.031 -> [ 47254][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
17:26:57.159 -> [ 47374][I][BluetoothA2DPSink.cpp:1473] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 100%
17:26:57.287 -> [ 47412][I][BluetoothA2DPSinkQueued.cpp:100] write_audio(): [BT_API] ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING
17:26:57.415 -> [ 47488][I][BluetoothA2DPSink.cpp:1413] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 100
17:26:57.544 -> [ 47632][E][BluetoothA2DPSink.cpp:1344] i2s_write_data(): [BT_AV] i2s_write_data failed - inactive
17:26:57.639 -> [ 47641][W][BluetoothA2DPSinkQueued.cpp:93] write_audio(): [BT_API] ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING
17:26:57.799 -> [ 47860][E][BluetoothA2DPSink.cpp:1344] i2s_write_data(): [BT_AV] i2s_write_data failed - inactive
17:26:57.927 -> [ 48024][W][BluetoothA2DPSinkQueued.cpp:81] write_audio(): [BT_API] ringbuffer is full, drop this packet!
17:26:58.023 -> [ 48126][E][BluetoothA2DPSink.cpp:1344] i2s_write_data(): [BT_AV] i2s_write_data failed - inactive
17:26:58.119 -> [ 48239][W][BluetoothA2DPSinkQueued.cpp:81] write_audio(): [BT_API] ringbuffer is full, drop this packet!
17:26:58.248 -> [ 48342][E][BluetoothA2DPSink.cpp:1344] i2s_write_data(): [BT_AV] i2s_write_data failed - inactive
17:26:58.344 -> [ 48455][W][BluetoothA2DPSinkQueued.cpp:81] write_audio(): [BT_API] ringbuffer is full, drop this packet!
```

 The last few lines repeat...

I am pretty concerned based on your last message that there is some kind of hardware error here. Almost like the breadboards that I'm using are flawed, or the jumper wires are too thin, or something like that. So many potential variables...

pschatzmann, 2023-10-19:
I guess the access to is_i2s_active needs to by synchronized.
Can you check if is good enought to define this variable as volatile ?

mrhempman69, 2023-10-19:
> I guess the access to **is_i2s_active** needs to by synchronized. Can you check if is good enought to define this variable as volatile ?

I tried that here. This is the full log: 
```
18:23:29.012 -> �(���.��L��$5�[    36][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 222316
18:23:29.511 -> [    36][I][BluetoothA2DPSink.cpp:172] start(): [BT_AV] Device name will be set to 'mrhemMusicBasic7'
18:23:29.608 -> [   103][I][BluetoothA2DPSink.cpp:238] init_i2s(): [BT_AV] init_i2s
18:23:29.705 -> [   174][I][BluetoothA2DPSink.cpp:240] init_i2s(): [BT_AV] init_i2s is_i2s_output
18:23:29.768 -> [   356][I][BluetoothA2DPSink.cpp:320] init_bluetooth(): [BT_AV] controller initialized
18:23:30.078 -> [   680][I][BluetoothA2DPSink.cpp:329] init_bluetooth(): [BT_AV] bluedroid initialized
18:23:30.143 -> [   749][I][BluetoothA2DPSink.cpp:338] init_bluetooth(): [BT_AV] bluedroid enabled
18:23:30.238 -> [   749][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:30.366 -> [   845][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
18:23:30.495 -> [   852][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
18:23:30.558 -> [   960][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:30.686 -> [  1041][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
18:23:30.751 -> [  1162][I][BluetoothA2DPSink.cpp:643] av_hdl_a2d_evt(): [BT_AV] A2DP PROF STATE: Init Compl
18:23:30.847 -> 
18:23:30.878 -> [  1243][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
18:23:30.942 -> [   960][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 116652
18:23:31.038 -> [  1424][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
18:23:38.186 -> [  8812][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 16
18:23:38.418 -> [  9046][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:38.546 -> [  9046][I][BluetoothA2DPSink.cpp:753] handle_connection_state(): [BT_AV] partner address: c0:b6:58:63:bf:5d
18:23:38.673 -> [  9146][I][BluetoothA2DPSink.cpp:755] handle_connection_state(): [BT_AV] A2DP connection state: Connecting, [c0:b6:58:63:bf:5d]
18:23:38.801 -> [  9282][I][BluetoothA2DPSink.cpp:814] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTING
18:23:38.929 -> [  9397][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:39.025 -> [  9518][I][BluetoothA2DPSink.cpp:1463] av_hdl_avrc_tg_evt(): [BT_AV] AVRC conn_state evt: state 1, [c0:b6:58:63:bf:5d]
18:23:39.153 -> [  9644][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:39.281 -> [  9765][I][BluetoothA2DPSink.cpp:1479] av_hdl_avrc_tg_evt(): [BT_AV] AVRC register event notification: 13, param: 0x0
18:23:39.409 -> [  9890][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:39.538 -> [ 10011][I][BluetoothA2DPSink.cpp:661] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , codec type 0
18:23:39.634 -> [ 10116][I][BluetoothA2DPSink.cpp:673] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , sample_rate 44100
18:23:39.763 -> [ 10227][I][BluetoothA2DPSink.cpp:684] handle_audio_cfg(): [BT_AV] configure audio player 21-15-2-35
18:23:39.859 -> 
18:23:39.859 -> [ 10335][I][BluetoothA2DPSink.cpp:692] handle_audio_cfg(): [BT_AV] audio player configured, samplerate=44100
18:23:39.956 -> [ 10449][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:40.084 -> [ 10570][I][BluetoothA2DPSink.cpp:939] av_hdl_avrc_evt(): [BT_AV] AVRC conn_state evt: state 1, [c0:b6:58:63:bf:5d]
18:23:40.213 -> [ 10692][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:40.341 -> [ 10813][I][BluetoothA2DPSink.cpp:984] av_hdl_avrc_evt(): [BT_AV] AVRC remote features 25b
18:23:40.438 -> [ 10909][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:40.567 -> [ 11030][I][BluetoothA2DPSink.cpp:1494] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote features 25b, CT features 2
18:23:40.662 -> [ 11145][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
18:23:40.791 -> [ 11266][I][BluetoothA2DPSink.cpp:753] handle_connection_state(): [BT_AV] partner address: c0:b6:58:63:bf:5d
18:23:40.886 -> [ 11381][I][BluetoothA2DPSink.cpp:755] handle_connection_state(): [BT_AV] A2DP connection state: Connected, [c0:b6:58:63:bf:5d]
18:23:41.046 -> [ 11515][I][BluetoothA2DPSink.cpp:819] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTED
18:23:41.046 -> [ 11515][I][BluetoothA2DPSink.cpp:819] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTED
18:23:41.142 -> [ 11630][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable false
18:23:41.271 -> [ 11745][I][BluetoothA2DPSinkQueued.cpp:6] bt_i2s_task_start_up(): [BT_API] ringbuffer data empty! mode changed: RINGBUFFER_MODE_PREFETCHING
18:23:41.398 -> [ 11893][I][BluetoothA2DPSinkQueued.cpp:21] bt_i2s_task_start_up(): [BT_AV] BtI2STask Started
[... cut ...]
18:23:46.209 -> [ 16578][W][BluetoothA2DPSinkQueued.cpp:81] write_audio(): [BT_API] ringbuffer is full, drop this packet!
18:23:46.305 -> [ 16680][E][BluetoothA2DPSink.cpp:1344] i2s_write_data(): [BT_AV] i2s_write_data failed - inactive
```
--- repeats endlessly

> But it is strange: the in the above log the entry for i2s_start is missing...
If a line is missing from the log, may be due to arduino ide maintainers closing discussion for a huge issue, then having its assignee leave, without being replaced https://github.com/arduino/arduino-ide/issues/2262 https://github.com/arduino/arduino-ide/issues/812

pschatzmann, 2023-10-19:
That should be good enough. I am still confused from what I see: some critical events seem to be missing.
Can you try to add the logic from https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_status_callback/bt_music_receiver_status_callback.ino

I really need the confirmation that the playback status is working....

mrhempman69, 2023-10-19:
Here is the log from the code you requested. Hopefully the issue was not that I had "Info" logging instead of "Debug". Debug is active now..:
```
19:18:18.063 -> ets Jul 29 2019 12:21:46
19:18:18.108 -> 
19:18:18.108 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:18:18.108 -> configsip: 0, SPIWP:0xee
19:18:18.108 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:18:18.108 -> mode:DIO, clock div:1
19:18:18.108 -> load:0x3fff0030,len:1344
19:18:18.108 -> load:0x40078000,len:13964
19:18:18.108 -> load:0x40080400,len:3600
19:18:18.108 -> entry 0x400805f0
19:18:18.471 -> [    25][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
19:18:18.503 -> [    36][D][BluetoothA2DPSink.cpp:160] start(): [BT_AV] start
19:18:18.503 -> [    36][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 222372
19:18:18.503 -> [    45][I][BluetoothA2DPSink.cpp:172] start(): [BT_AV] Device name will be set to 'MyMusic'
19:18:18.503 -> [    53][D][BluetoothA2DPSink.cpp:1163] init_nvs(): [BT_AV] init_nvs
19:18:18.503 -> [    59][D][BluetoothA2DPCommon.cpp:152] get_last_connection(): [BT_AV] get_last_connection
19:18:18.535 -> [    68][D][BluetoothA2DPCommon.cpp:175] get_last_connection(): [BT_AV] => c0:b6:58:63:bf:5d
19:18:18.535 -> [    75][I][BluetoothA2DPSink.cpp:238] init_i2s(): [BT_AV] init_i2s
19:18:18.535 -> [    81][I][BluetoothA2DPSink.cpp:240] init_i2s(): [BT_AV] init_i2s is_i2s_output
19:18:18.535 -> [    89][D][BluetoothA2DPSink.cpp:315] init_bluetooth(): [BT_AV] init_bluetooth
19:18:18.658 -> [   190][I][BluetoothA2DPSink.cpp:320] init_bluetooth(): [BT_AV] controller initialized
19:18:18.964 -> [   502][I][BluetoothA2DPSink.cpp:329] init_bluetooth(): [BT_AV] bluedroid initialized
19:18:19.028 -> [   572][I][BluetoothA2DPSink.cpp:338] init_bluetooth(): [BT_AV] bluedroid enabled
19:18:19.028 -> [   572][D][BluetoothA2DPSink.cpp:455] app_task_start_up(): [BT_AV] app_task_start_up
19:18:19.028 -> [   575][D][BluetoothA2DPSink.cpp:1289] ccall_app_task_handler(): [BT_AV] ccall_app_task_handler
19:18:19.028 -> [   584][D][BluetoothA2DPSink.cpp:411] app_task_handler(): [BT_AV] app_task_handler
19:18:19.060 -> [   591][D][BluetoothA2DPSink.cpp:363] app_work_dispatch(): [BT_API] app_work_dispatch event 0x0, param len 0
19:18:19.060 -> [   601][D][BluetoothA2DPSink.cpp:395] app_send_msg(): [BT_AV] app_send_msg
19:18:19.060 -> [   608][D][BluetoothA2DPSink.cpp:418] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x0
19:18:19.060 -> [   616][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
19:18:19.092 -> [   626][D][BluetoothA2DPSink.cpp:386] app_work_dispatched(): [BT_AV] app_work_dispatched
19:18:19.092 -> [   634][D][BluetoothA2DPSink.cpp:1319] ccall_av_hdl_stack_evt(): [BT_AV] ccall_av_hdl_stack_evt
19:18:19.092 -> [   643][D][BluetoothA2DPSink.cpp:1015] av_hdl_stack_evt(): [BT_AV] av_hdl_stack_evt evt 0
19:18:19.092 -> [   651][D][BluetoothA2DPSink.cpp:1020] av_hdl_stack_evt(): [BT_AV] av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
19:18:19.124 -> [   661][D][BluetoothA2DPSink.cpp:1029] av_hdl_stack_evt(): [BT_AV] AVRCP controller initialized!
19:18:19.124 -> [   662][D][BluetoothA2DPSink.cpp:1313] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
19:18:19.124 -> [   670][D][BluetoothA2DPSink.cpp:1072] av_hdl_stack_evt(): [BT_AV] set_scan_mode_connectable(true)
19:18:19.156 -> [   678][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
19:18:19.156 -> [   687][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
19:18:19.156 -> [   696][D][BluetoothA2DPSink.cpp:1301] ccall_app_a2d_callback(): [BT_AV] ccall_app_a2d_callback
19:18:19.156 -> [   670][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 130044
19:18:19.188 -> [   712][D][BluetoothA2DPSink.cpp:1088] app_a2d_callback(): [BT_AV] app_a2d_callback
19:18:19.188 -> [   727][D][BluetoothA2DPSink.cpp:1108] app_a2d_callback(): [BT_AV] app_a2d_callback ESP_A2D_AUDIO_CFG_EVT
19:18:19.188 -> [   737][D][BluetoothA2DPSink.cpp:363] app_work_dispatch(): [BT_API] app_work_dispatch event 0x4, param len 16
19:18:19.188 -> [   746][D][BluetoothA2DPSink.cpp:395] app_send_msg(): [BT_AV] app_send_msg
19:18:19.220 -> [   753][D][BluetoothA2DPSink.cpp:418] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x4
19:18:19.220 -> [   754][D][BluetoothA2DPSink.cpp:1313] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
19:18:19.220 -> [   762][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
19:18:19.220 -> [   770][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
19:18:19.252 -> [   780][D][BluetoothA2DPSink.cpp:386] app_work_dispatched(): [BT_AV] app_work_dispatched
19:18:19.252 -> [   787][D][BluetoothA2DPSink.cpp:1313] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
19:18:19.252 -> [   795][D][BluetoothA2DPSink.cpp:1333] ccall_av_hdl_a2d_evt(): [BT_AV] ccall_av_hdl_a2d_evt
19:18:19.252 -> [   804][I][BluetoothA2DPSink.cpp:566] app_gap_callback(): [BT_AV] event: 10
19:18:19.284 -> [   812][D][BluetoothA2DPSink.cpp:619] av_hdl_a2d_evt(): [BT_AV] av_hdl_a2d_evt evt 4
19:18:19.284 -> [   818][D][BluetoothA2DPSink.cpp:1313] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
[... cut ...]
19:18:38.045 -> [ 19609][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
19:18:38.117 -> [ 19650][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
19:18:38.117 -> [ 19666][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
```
The above logs are from just the page you linked to.  In the next Github comment, I will provide logs for some code that has some more of my code integrated (will be show the code there)

mrhempman69, 2023-10-19:
So in the last comment, I just used the code that you linked to. But this is with some more of my code integrated.

The code: 

```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

// for esp_a2d_connection_state_t see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_a2dp.html#_CPPv426esp_a2d_connection_state_t
void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  Serial.println(a2dp_sink.to_str(state));
}

// for esp_a2d_audio_state_t see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_a2dp.html#_CPPv421esp_a2d_audio_state_t
void audio_state_changed(esp_a2d_audio_state_t state, void *ptr){
  Serial.println(a2dp_sink.to_str(state));
}

void setup() {
  Serial.begin(115200);
  a2dp_sink.set_on_connection_state_changed(connection_state_changed);
  a2dp_sink.set_on_audio_state_changed(audio_state_changed);
  
  int mclkPin=3;
  i2s_pin_config_t external_pin_config = {
        .mck_io_num = mclkPin,
        .bck_io_num =33,
        .ws_io_num = 32,
        .data_out_num = 22,
        .data_in_num = I2S_PIN_NO_CHANGE,
    };
  static const i2s_config_t externalDAC_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
        .sample_rate =  44100, // corrected by info from bluetooth
        .bits_per_sample = (i2s_bits_per_sample_t) 16, 
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .use_apll = true,
        .tx_desc_auto_clear = true, // avoiding noise in case of data unavailability
        .mclk_multiple=(i2s_mclk_multiple_t) 256,

  };

  bool externalDAC=true;

    a2dp_sink.set_pin_config(external_pin_config);
    a2dp_sink.set_i2s_config(externalDAC_config);
    a2dp_sink.i2s_mclk_pin_select(mclkPin);

  a2dp_sink.start("MyMusic");  
}

void loop() {
  delay(1000); // do nothing
}
```
The log:

```
19:24:16.755 -> ets Jul 29 2019 12:21:46
19:24:16.755 -> 
19:24:16.755 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:24:16.755 -> configsip: 0, SPIWP:0xee
19:24:16.755 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:24:16.755 -> mode:DIO, clock div:1
19:24:16.755 -> load:0x3fff0030,len:1344
[... 385 lines cut ...]
19:25:01.614 -> [ 44522][D][BluetoothA2DPSink.cpp:363] app_work_dispatch(): [BT_API] app_work_dispatch event 0x3, param len 12
19:25:01.646 -> [ 44532][D][BluetoothA2DPSink.cpp:395] app_send_msg(): [BT_AV] app_send_msg
19:25:01.646 -> [ 44539][D][BluetoothA2DPSink.cpp:418] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x3
19:25:01.646 -> [ 44539][D][BluetoothA2DPSink.cpp:1307] ccall_app_rc_ct_callback(): [BT_AV] ccall_app_rc_ct_callback
19:25:01.646 -> [ 44547][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
19:25:01.678 -> [ 44556][D][BluetoothA2DPSink.cpp:575] app_rc_ct_callback(): [BT_AV] app_rc_ct_callback
19:25:01.678 -> [ 44566][D][BluetoothA2DPSink.cpp:386] app_work_dispatched(): [BT_AV] app_work_dispatched
19:25:01.678 -> [ 44574][D][BluetoothA2DPSink.cpp:592] app_rc_ct_callback(): [BT_AV] app_rc_ct_callback ESP_AVRC_CT_CHANGE_NOTIFY_EVT
19:25:01.678 -> [ 44582][D][BluetoothA2DPSink.cpp:1449] ccall_av_hdl_avrc_tg_evt(): [BT_AV] ccall_av_hdl_avrc_tg_evt
19:25:01.710 -> [ 44592][D][BluetoothA2DPSink.cpp:363] app_work_dispatch(): [BT_API] app_work_dispatch event 0x4, param len 12
19:25:01.710 -> [ 44601][D][BluetoothA2DPSink.cpp:1457] av_hdl_avrc_tg_evt(): [BT_AV] av_hdl_avrc_tg_evt evt 3
19:25:01.710 -> [ 44611][D][BluetoothA2DPSink.cpp:395] app_send_msg(): [BT_AV] app_send_msg
19:25:01.742 -> [ 44619][I][BluetoothA2DPSink.cpp:1473] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 11%
19:25:01.742 -> [ 44628][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
19:25:01.742 -> [ 44635][I][BluetoothA2DPSink.cpp:1413] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 11
19:25:01.742 -> [ 44653][D][BluetoothA2DPSink.cpp:418] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x4
19:25:01.774 -> [ 44660][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
19:25:01.774 -> [ 44662][I][BluetoothA2DPSink.cpp:421] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
19:25:01.774 -> [ 44680][D][BluetoothA2DPSink.cpp:386] app_work_dispatched(): [BT_AV] app_work_dispatched
19:25:01.774 -> [ 44685][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
19:25:01.806 -> [ 44688][D][BluetoothA2DPSink.cpp:1326] ccall_av_hdl_avrc_evt(): [BT_AV] ccall_av_hdl_avrc_evt
19:25:01.806 -> [ 44704][D][BluetoothA2DPSink.cpp:935] av_hdl_avrc_evt(): [BT_AV] av_hdl_avrc_evt evt 4
19:25:01.806 -> [ 44711][D][BluetoothA2DPSink.cpp:1121] audio_data_callback(): [BT_AV] audio_data_callback
19:25:01.806 -> [ 44712][D][BluetoothA2DPSink.cpp:911] av_notify_evt_handler(): [BT_AV] av_notify_evt_handler
19:25:01.838 -> [ 44728][D][BluetoothA2DPSink.cpp:918] av_notify_evt_handler(): [BT_AV] av_notify_evt_handler ESP_AVRC_RN_PLAY_STATUS_CHANGE 1, to 2

pschatzmann, 2023-10-20:
I could not really reproduce your issue on my andoid or old IPhones. All were playing the audio.
However after some start and tops, there was the risk that things were getting bad.
I committed some correction hoping that this might help you as well.

Were/Are you getting no audio at all ?

A couple of comments:
- please remove the a2dp_sink.i2s_mclk_pin_select(mclkPin); This method will be obsoleted and is not necessary when you can specify the pin in the i2s_pin_config_t
- Log level Info should give a better overview then Debug

===> Startup:
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[    48][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=33 scl=32 freq=100000
Starting A2DP...
[    69][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 219332
[    69][I][BluetoothA2DPSink.cpp:172] start(): [BT_AV] Device name will be set to 'AudioKit'
[    75][I][BluetoothA2DPCommon.cpp:166] get_last_connection(): [BT_AV] nvs_blob does not exist
[    83][I][BluetoothA2DPSink.cpp:240] init_i2s(): [BT_AV] init_i2s
[    89][I][BluetoothA2DPSink.cpp:242] init_i2s(): [BT_AV] init_i2s is_i2s_output
[   162][I][BluetoothA2DPSink.cpp:322] init_bluetooth(): [BT_AV] controller initialized
[   483][I][BluetoothA2DPSink.cpp:331] init_bluetooth(): [BT_AV] bluedroid initialized
[   552][I][BluetoothA2DPSink.cpp:340] init_bluetooth(): [BT_AV] bluedroid enabled
[   553][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   558][I][BluetoothA2DPSink.cpp:232] start(): [BT_AV] IDF Version 4.4
[   559][I][BluetoothA2DPSink.cpp:568] app_gap_callback(): [BT_AV] event: 10
[   559][I][BluetoothA2DPSink.cpp:645] av_hdl_a2d_evt(): [BT_AV] A2DP PROF STATE: Init Compl

[   571][I][BluetoothA2DPSink.cpp:568] app_gap_callback(): [BT_AV] event: 10
[   564][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 112696
[   586][I][BluetoothA2DPSink.cpp:568] app_gap_callback(): [BT_AV] event: 10
[   602][I][BluetoothA2DPSink.cpp:568] app_gap_callback(): [BT_AV] event: 10

===> Connecting:

[ 85301][I][BluetoothA2DPSink.cpp:568] app_gap_callback(): [BT_AV] event: 16
[ 85611][I][BluetoothA2DPSink.cpp:765] handle_connection_state(): [BT_AV] partner address: 98:0d:6f:b9:ad:d9
[ 85611][I][BluetoothA2DPSink.cpp:767] handle_connection_state(): [BT_AV] A2DP connection state: Connecting, [98:0d:6f:b9:ad:d9]
[ 85621][I][BluetoothA2DPSink.cpp:826] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTING
[ 85673][I][BluetoothA2DPSink.cpp:663] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , codec type 0
[ 85673][I][BluetoothA2DPSink.cpp:675] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , sample_rate 44100
[ 85679][I][BluetoothA2DPSink.cpp:686] handle_audio_cfg(): [BT_AV] configure audio player 21-15-2-35

[ 85689][I][BluetoothA2DPSink.cpp:694] handle_audio_cfg(): [BT_AV] audio player configured, samplerate=44100
[ 85724][I][BluetoothA2DPSink.cpp:765] handle_connection_state(): [BT_AV] partner address: 98:0d:6f:b9:ad:d9
[ 85724][I][BluetoothA2DPSink.cpp:767] handle_connection_state(): [BT_AV] A2DP connection state: Connected, [98:0d:6f:b9:ad:d9]
[ 85734][I][BluetoothA2DPSink.cpp:831] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTED
[ 85743][I][BluetoothA2DPCommon.cpp:289] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable false
[ 85753][I][BluetoothA2DPSinkQueued.cpp:6] bt_i2s_task_start_up(): [BT_API] ringbuffer data empty! mode changed: RINGBUFFER_MODE_PREFETCHING
[ 85765][I][BluetoothA2DPSinkQueued.cpp:21] bt_i2s_task_start_up(): [BT_AV] BtI2STask Started
[ 85812][I][BluetoothA2DPSink.cpp:552] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT stat:0
[ 85813][I][BluetoothA2DPSink.cpp:554] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT remote name:A34 von Phil
[ 85829][I][BluetoothA2DPSink.cpp:951] av_hdl_avrc_evt(): [BT_AV] AVRC conn_state evt: state 1, [98:0d:6f:b9:ad:d9]
[ 85832][I][BluetoothA2DPSink.cpp:1481] av_hdl_avrc_tg_evt(): [BT_AV] AVRC conn_state evt: state 1, [98:0d:6f:b9:ad:d9]
[ 85842][I][BluetoothA2DPSink.cpp:996] av_hdl_avrc_evt(): [BT_AV] AVRC remote features 25b
[ 85850][I][BluetoothA2DPSink.cpp:1512] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote features 25b, CT features 2
[ 85860][I][BluetoothA2DPSink.cpp:1004] av_hdl_avrc_evt(): [BT_AV] remote rn_cap: count 7, bitmask 0xf26
[ 85886][I][BluetoothA2DPSink.cpp:1497] av_hdl_avrc_tg_evt(): [BT_AV] AVRC register event notification: 13, param: 0x0
[ 86063][I][BluetoothA2DPSink.cpp:977] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x1, New Age [Live]
[ 86063][I][BluetoothA2DPSink.cpp:977] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x2, The Velvet Underground
[ 86073][I][BluetoothA2DPSink.cpp:977] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x4, Gold [Disc 2]
[ 86083][I][BluetoothA2DPSink.cpp:977] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x8, 311
[ 86093][I][BluetoothA2DPSink.cpp:977] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x10, 501
[ 86102][I][BluetoothA2DPSink.cpp:977] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x20, Rock
[... 7 lines cut ...]
[... cut ...]
[215044][I][BluetoothA2DPSink.cpp:1431] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 57
[216529][I][BluetoothA2DPSink.cpp:1491] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 84%
[216529][I][BluetoothA2DPSink.cpp:1431] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 84
[218014][I][BluetoothA2DPSink.cpp:1491] av_hdl_avrc_tg_evt(): [BT_AV] AVRC remote set absolute volume: 100%
[218014][I][BluetoothA2DPSink.cpp:1431] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 100

mrhempman69, 2023-10-20:
**Queued class**

I do have audio with the queued class after those changes, but I don't think I did before. The volume changes are still delayed.

When music is played, this is output constantly (almost like it's taking up the full processor):
```
13:23:46.970 -> [ 76312][I][BluetoothA2DPSinkQueued.cpp:69] i2s_task_handler(): [BT_API] ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING
```

If I pause the music, I get:
```
13:25:24.739 -> [174077][I][BluetoothA2DPSinkQueued.cpp:121] write_audio(): [BT_API] ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING
13:25:24.778 -> [174099][W][BluetoothA2DPSinkQueued.cpp:114] write_audio(): [BT_API] ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING
13:25:24.810 -> [174128][W][BluetoothA2DPSinkQueued.cpp:102] write_audio(): [BT_API] ringbuffer is full, drop this packet!
13:25:24.810 -> [174133][W][BluetoothA2DPSinkQueued.cpp:102] write_audio(): [BT_API] ringbuffer is full, drop this packet!
13:25:24.845 -> [174160][W][BluetoothA2DPSinkQueued.cpp:102] write_audio(): [BT_API] ringbuffer is full, drop this packet!
13:25:24.845 -> [174160][I][BluetoothA2DPSinkQueued.cpp:105] write_audio(): [BT_API] ringbuffer data decreased! mode changed: RINGBUFFER_MODE_PROCESSING
```
For only 10-20 seconds!!! Then it may go back to "PREFETCHING" repeating over and over.

The volume changes work instantly when it is paused. But again, it goes back to "PREFETCHING" even when paused for a while especially after volume changes.

Btw, I'm assuming your last bullet point meant me changing the config values to this:
```
        .dma_buf_count = 6,
        .dma_buf_len = 512,
```
which I did also btw.

**Non-Queued Class**

For the regular a2dp sink class, the functionality is the same, but I get no ringbuffer results.

**One more thing**

I touched on this in one of the logs earlier, but I should mention. When I set the volume to 0.0, **the controller receives this one instantly**.  So any value between 1 and 100 requires pausing the music for the volume to update. But volume 0 is immediately received, for some reason. This makes me think you may need an iPhone 6S or  newer to debug this further. Or you can tell me what to do to try to see what the iPhone is communicating here.

pschatzmann, 2023-10-20:
Maybe it's time to go back to the Espressif example to try to reproduce the issue and file a bug report.

You can also try to play around with some parameters. 
- increase the message queue size
- increase the i2s ringbuffer size and or percentage. The audio starts playing after the ringbuffer is filled > percentage
- in the BluetoothA2DPSinkQueued::i2s_task_handler I have added a delay(5). You can try to move this before the xRingbufferReceiveUpTo() and maybe increase the value. The idea is to give the task which is filling the ringbuffer some opportunity to do it's job.

mrhempman69, 2023-10-20:
> * in the BluetoothA2DPSinkQueued::i2s_task_handler I have added a delay(5). You can try to move this before the xRingbufferReceiveUpTo() and maybe increase the value. The idea is to give the task which is filling the ringbuffer some opportunity to do it's job.

Lol, this did something. Delay of "1" was constant "underflow". Delay of "25" was constant "overflow". Delay of "7" was a little bit of underflow, with volume changes only coming at about a 1 second delay. "This is fine", I thought, "good enough". Then the device rebooted itself! 

> It is also strange that if you press pause, it continues to receive data..

Any ideas on how I could debug why this is happening? Technically, it is receiving "item_size==0" worth of data.

> 
> You can also try to play around with some parameters.
> 
> * increase the message queue size
> * increase the i2s ringbuffer size and or percentage. The audio starts playing after the ringbuffer is filled > percentage

Yes, I'll keep trying. If you have any comments or suggestions, please share them. For some reason your last suggestions made me understand a bit more about what is going on

pschatzmann, 2023-10-20:
Not sure if this makes a difference, but I was testing with ESP32 2.0.14.
I am not sure in which task the audio data is written, so it might make sense to try to add some delays to the BluetoothA2DPSinkQueued::write_audio

mrhempman69, 2023-10-20:
> It is also strange that if you press pause, it continues to receive data..

Just want to add this comment, that my Android tablet works perfectly with the queued class, but it does enter the underflow condition when the music is paused too.

mrhempman69, 2023-10-20:
I think to solve this issue, @pschatzmann, even though the above code works, I have to know why it sounds like the singer is gargling water when he talks? It's not a "bass" issue, it's like the whole audio file is choppy. this happens in both the queued and non-queued classes, before and after the above changes were made. 

1) https://drive.google.com/file/d/1CR9PSLW2CMQPmCYgI9psTNVKhguCAZ4U/view?usp=share_link

2) https://drive.google.com/file/d/1DxDCHhuKqY5LDVzHzNBXlpdmr9Mu-MZt/view?usp=share_link

pschatzmann, 2023-10-21:
I noticed that the original design from apple was working with different task priorities. I changed the approach and made sure that all tasks are running with the same priority and I am forcing a task switch with yield().

I have also added some more debug log statements which print the size of the audio data. This should make it possible to find if there are any differences between Apple and Android (in debug level).

This changes have been committed in the __issue-apple branch__.  I am wondering if this helps even when using the BluetoothA2DPSink class.

I am getting the same audio effect when I use the debug log level, so when testing the audio quality I recommend to use the Warning log level.

mrhempman69, 2023-10-21:
> I noticed that the original design from Espressif was working with different task priorities. I changed the approach and made sure that all tasks are running with the same priority and I am forcing a task switch with yield().
> 
> This changes have been committed in the **issue-apple branch**. I am wondering if this helps even when using the BluetoothA2DPSink class.
> 
> I am getting the same audio effect like you when I use the debug log level, so when testing the audio quality I recommend to use the Warning or Info log level.

First pass, the audio is still the same as before, but....

>  I have also added some more debug log statements which print the size of the audio data. This should make it possible to find if there are any differences between Apple and Android (in debug level). See A2DP_DEBUG_AUDIO in config.h I am wondering if Apple is providing different array sizes?

See below logs..

IPhone:
```09:55:35.028 -> Started
09:55:35.028 -> [ 45292][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
09:55:35.028 -> [ 45299][I][BluetoothA2DPSink.cpp:763] set_i2s_active(): [BT] i2s_start
09:55:35.028 -> [ 45299][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
09:55:35.060 -> [ 45305][D][BluetoothA2DPSink.cpp:429] app_task_handler(): [BT] app_task_handler, sig 0x1, 0x4
09:55:35.060 -> [ 45314][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 4096
09:55:35.060 -> [ 45322][D][BluetoothA2DPSink.cpp:432] app_task_handler(): [BT] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
09:55:35.060 -> [ 45332][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
09:55:35.092 -> [ 45339][D][BluetoothA2DPSink.cpp:397] app_work_dispatched(): [BT] app_work_dispatched
09:55:35.092 -> [ 45348][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
09:55:35.092 -> [ 45356][D][BluetoothA2DPSink.cpp:1378] ccall_av_hdl_avrc_evt(): [BT] ccall_av_hdl_avrc_evt
09:55:35.124 -> [ 45364][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 4096
09:55:35.124 -> [ 45372][D][BluetoothA2DPSink.cpp:972] av_hdl_avrc_evt(): [BT] av_hdl_avrc_evt evt 4
09:55:35.124 -> [ 45382][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
09:55:35.124 -> [ 45387][D][BluetoothA2DPSink.cpp:948] av_notify_evt_handler(): [BT] av_notify_evt_handler
09:55:35.156 -> [ 45396][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
09:55:35.156 -> [ 45404][D][BluetoothA2DPSink.cpp:955] av_notify_evt_handler(): [BT] av_notify_evt_handler ESP_AVRC_RN_PLAY_STATUS_CHANGE 1, to 1
09:55:35.156 -> [ 45413][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 4096
09:55:35.156 -> [ 45424][D][BluetoothA2DPSink.cpp:933] av_playback_changed(): [BT] av_playback_changed
09:55:35.188 -> [ 45434][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
09:55:35.188 -> [ 45448][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
09:55:35.188 -> [ 45457][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 4096
09:55:35.188 -> [ 45466][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
09:55:35.220 -> [ 45473][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
09:55:35.220 -> [ 45482][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 4096
09:55:35.220 -> [ 45491][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
09:55:35.220 -> [ 45498][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
```

Android: 
```
09:57:26.267 -> Started
09:57:26.267 -> [101118][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
09:57:26.267 -> [101125][I][BluetoothA2DPSink.cpp:763] set_i2s_active(): [BT] i2s_start
09:57:26.299 -> [101125][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
09:57:26.299 -> [101139][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 3584
09:57:26.299 -> [101149][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
09:57:26.299 -> [101156][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
09:57:26.331 -> [101164][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 3584
09:57:26.331 -> [101177][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
09:57:26.331 -> [101181][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
09:57:26.331 -> [101189][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 3584
09:57:26.363 -> [101199][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
09:57:26.363 -> [101206][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
09:57:26.363 -> [101214][D][BluetoothA2DPSink.cpp:1398] i2s_write_data(): [BT] i2s_write_data: 3584
09:57:26.363 -> [101223][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3072
[... cut ...]
> I have also updated the [Optimizations Chapter](https://github.com/pschatzmann/ESP32-A2DP/wiki/A2DP-Sink-Optimizations) in the Wiki....

This was useful... Let me know if you think I should change any of these values based on that log.... I'm tempted to try setting MAX_PRIORITIES higher... but I should be addressing something with that changing data callback

I read what you wrote about newer versions of Espressif using more RAM. I should probably test with an older version of Espressif at some point... Of course I could try a bunch of different values for these as well.

pschatzmann, 2023-10-21:
I tend to think that the i2s buffer should be at least 4096 bytes or more.
I don't feel comfortable with these big array sizes, so will try to split this up into individual 1k writes...

pschatzmann, 2023-10-21:
I committed the change of making max 1k i2s writes...

mrhempman69, 2023-10-21:
Not sure why these didn't throw as errors before, but a couple typos:

Typo on line 1414 of BluetoothA2DPSink.cpp: Should be "BT_TAG" not "BTe_TAG" - said BTe_TAG not declared in scope

On 1399: Should be printing "size" not "item_size" for this log statement (?) - said item_size not declared in scope

I made those changes and...

On iPhone:

```
11:00:52.214 -> [ 39267][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
11:00:52.214 -> [ 39274][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
11:00:52.246 -> [ 39283][D][BluetoothA2DPSink.cpp:1399] write_audio(): [BT] write_audio: 4096
11:00:52.246 -> [ 39289][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:00:52.246 -> [ 39296][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:00:52.246 -> [ 39304][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:00:52.246 -> [ 39311][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
```

On Android:

```
11:02:11.753 -> [ 76693][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
11:02:11.753 -> [ 76701][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
11:02:11.753 -> [ 76709][D][BluetoothA2DPSink.cpp:1399] write_audio(): [BT] write_audio: 3584
11:02:11.753 -> [ 76716][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:02:11.785 -> [ 76723][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:02:11.785 -> [ 76730][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:02:11.785 -> [ 76738][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 512
11:02:11.785 -> [ 76748][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3072
11:02:11.817 -> [ 76754][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3072
11:02:11.817 -> [ 76763][D][BluetoothA2DPSink.cpp:1399] write_audio(): [BT] write_audio: 3072
11:02:11.817 -> [ 76769][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:02:11.817 -> [ 76777][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:02:11.849 -> [ 76785][D][BluetoothA2DPSink.cpp:1414] i2s_write_data(): [BT] i2s_write_data: 1024
11:02:11.849 -> [ 76794][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
11:02:11.849 -> [ 76801][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
```

Audio qualities are pretty similar. Some pops on Android that are noticeable. On iOS, a lot more pops, still slightly choppy audio.

You don't think iOS is padding their data with zeros do you? (Imagine how dumb that would be lol).  How can I use ESP_LOGD to look at the value of *data? (Sorry, my cpp is pretty bad).

pschatzmann, 2023-10-21:
I think you were too quick. The latest commit should not have any bugs and issues in the Audio...

mrhempman69, 2023-10-21:
Here's a look at the raw data (from the same point in the same song)....
On iPhone:
```
11:32:11.381 -> 0,0,0,0,0,0,0,0[ 75543][D][BluetoothA2DPSink.cpp:972] av_hdl_avrc_evt(): [BT] av_hdl_avrc_evt evt 4
11:32:11.381 -> ,0,0,0,0,0,0,[ 75554][D][BluetoothA2DPSink.cpp:948] av_notify_evt_handler(): [BT] av_notify_evt_handler
11:32:11.413 -> 0,0[ 75561][D][BluetoothA2DPSink.cpp:955] av_notify_evt_handler(): [BT] av_notify_evt_handler ESP_AVRC_RN_PLAY_STATUS_CHANGE 1, to 1
11:32:11.413 -> ,0,0,0,0,0,[ 75572][D][BluetoothA2DPSink.cpp:933] av_playback_changed(): [BT] av_playback_changed
11:32:11.413 -> 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,
11:32:11.605 -> [ 75759][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
11:32:11.605 -> [ 75764][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
11:32:11.605 -> [ 75773][D][BluetoothA2DPSink.cpp:1185] audio_data_callback(): [BT] stream_reader
11:32:11.605 -> 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,
11:32:11.797 -> [ 75967][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 4096
11:32:11.797 -> [ 75967][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 4096
11:32:11.829 -> [ 75976][D][BluetoothA2DPSink.cpp:1185] audio_data_callback(): [BT] stream_reader
[... cut ...] 9,119,119,119,119,120,120,120,120,120,120,120,119,119,118,117,115,114,113,111,109,107,105,103,101,98,95,92,89,86,82,79,75,72,68,64,60,56,52,48,44,40,36,31,27,22,18,13,9,4,0,-5,-11,-16,-21,-26,-31,-37,-41,-46,-50,-54,-58,-61,-64,-67,-69,-71,-72,-73,-75,-76,-76,-77,
11:33:46.260 -> [170431][D][BluetoothA2DPSink.cpp:1335] ccall_audio_data_callback(): [BT] ccall_audio_data_callback: 3584
11:33:46.260 -> [170431][D][BluetoothA2DPSink.cpp:1159] audio_data_callback(): [BT] audio_data_callback: 3584
```

pschatzmann, 2023-10-21:
If you have some SD driver you could save the raw data in a file and analyze it e.g. in Audacity

mrhempman69, 2023-10-31:
I have since tested with a second brand (HiLetGo) and the result is the same, functional audio for Android, and staticy / delayed volume controls for iOS.

> If you have some SD drive you could save the raw data in a file and analyze it e.g. in Audacity. The values in the beginning are usually very small - and not audible. If you use the full volume the range should be between -32768 and 32767

Is there an example code on how to do this? I have an SD card module now... 

> With my sketch you can also try to analyze it in the Serial Plotter in Arduino

I think I will try the SD card instead since I don't know how to do this

> Did you check if the BluetoothA2DPSinkQueued makes a difference ?

Yes many times

mrhempman69, 2023-11-02:
If you are interested, the cheapest phone to test with is the iPhone SE (2020 version). this one goes for about $88: https://www.ebay.com/itm/126111383313?hash=item1d5cd2ff11:g:jM0AAOSwyWdlEzKj&amdata=enc%3AAQAIAAAAwGf3wQcU9TdfnyVYhi6Dl2pmBVlartaIW%2BX%2BJ0V3ZVmaiIJ1XmDmygY1rc7ldXkwIKdtUi1pB5Z6%2B%2BmSK%2Fd31FOpWtFIb%2BuShO87gpLKDGJHAKmt1tbdf5QcVxuosaymsgWSK8vXMypu8l5TwHthQMj%2BoyjCjvNqik1%2F68%2FZaTDCrjgIlF8KKy1YyUM1Yn9GO25IgdyLOv9RtHzIMHrAdJGLwH%2BK1pLLmiPdxRTLK5hj1Oeb25HAw87xqEwaE4c1CA%3D%3D%7Ctkp%3ABk9SR9LuwKfyYg

Leo-Ha, 2024-04-07:
> If you are interested, the cheapest phone to test with is the iPhone SE (2020 version). this one goes for about $88: https://www.ebay.com/itm/126111383313?hash=item1d5cd2ff11:g:jM0AAOSwyWdlEzKj&amdata=enc%3AAQAIAAAAwGf3wQcU9TdfnyVYhi6Dl2pmBVlartaIW%2BX%2BJ0V3ZVmaiIJ1XmDmygY1rc7ldXkwIKdtUi1pB5Z6%2B%2BmSK%2Fd31FOpWtFIb%2BuShO87gpLKDGJHAKmt1tbdf5QcVxuosaymsgWSK8vXMypu8l5TwHthQMj%2BoyjCjvNqik1%2F68%2FZaTDCrjgIlF8KKy1YyUM1Yn9GO25IgdyLOv9RtHzIMHrAdJGLwH%2BK1pLLmiPdxRTLK5hj1Oeb25HAw87xqEwaE4c1CA%3D%3D%7Ctkp%3ABk9SR9LuwKfyYg
> 
> You may be able to find it for cheaper, but this is the cheapest one that I could find that doesn't appear to have a problem (bad screen, missing home button, etc)
> 
> I stand by my offer that I would donate the money for it!!

Sorry for this not being directly related to the issue but what iphone are you using and does the audio work otherwise since i cant get audio to work right