pschatzmann/ESP32-A2DP #511: Interupted sound

kuzmanzs, 2024-01-03:
### Problem Description

Hi,

I am trying to run bt_music_sender.ino
with following setup:
ESP32-WROOM32
ESP-IDF 4.3.6
Visual studio + platformio

I hear the sinus sound but it is interupted in every lets say 100-200msec. It sounds like a morse code.

I have tried with example from ESP-IDF the noise as well and it also sound also interupted.
```
static int32_t bt_app_a2d_data_cb(uint8_t *data, int32_t len)
{
    if (data == NULL || len < 0) {
        return 0;
    }

    int16_t *p_buf = (int16_t *)data;
    for (int i = 0; i < (len >> 1); i++) {
        p_buf[i] = rand() % (1 << 16);
    }

    return len;
}

```
1) If I just build I run the example source code of ESP-IDF it sounds without any interuption
2) If I run your "bt_music_sender_write" there is no any interruption. It sound well.

So what I am wrong with the "bt_music_sender" example?

I have another question. If I run the ESP-IDF example sender my speaker stops the connection after 120sec.
I think heartbeat is running, what I miss?

Thx

### Device Description

ESP32-WROOM32
ESP-IDF 4.3.6
Visual studio + platformio

### Sketch

```cpp
bt_music_sender
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

ESP-IDF 4.3.6

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

BlackyBlacksky, 2024-01-15:
Same problem here. :/

pschatzmann, 2024-04-06:
As far as I understand you generate a buffer overflow and write past the valid data!

closed 2024-04-06 by pschatzmann