pschatzmann/ESP32-A2DP #523: a2dp on windows 11 fails

eRTK, 2024-02-24:
### Problem Description

Hello, i used the a2dp lib sample projects on a standard esp32 compiled with actual arduino-ide. 
After down load when connecting to the a2dp sink with an android device a pairing without pin or so is possible. 
It does pairing without entering anything and plays a sound and all ist fine.
If i try to connect from windows 11, tried on several computers, i was always be asked for a "pin". 
Trying 0000 or 1234 or so always fails.
My question is, what to do to connect from windows 10/11 or such things to esp32 a2dp.
I finally indead want  to give a pin to the device, but at first it must do pairing without this i think.
Thanks
  Thomas

### Device Description

Vanilla ESP32 or NINA

### Sketch

```cpp
see the provided samples
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.13

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

eRTK, 2024-02-25:
Or want to know what the password is.

pschatzmann, 2024-04-06:
How about reading the [documentation](https://github.com/pschatzmann/ESP32-A2DP/wiki/Access-Protection---Authentication) ?

closed 2024-04-06 by pschatzmann