pschatzmann/ESP32-A2DP #531: RSSI cannot be updated in real-time

523273195, 2024-04-01:
### Problem Description

a2dp_sink.set_rssi_active(true);   a2dp_sink.set_rssi_callback(rssi);RSSI cannot be updated in real-time

### Device Description

ESP32

### Sketch

```cpp
bt_music_receiver_rssi
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

0

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-04-06:
I am sorry, I don't understand your problem. 
Could you please a bit more specific what your issue is ?

523273195, 2024-04-07:
This code updates the rssi value only once when the Bluetooth device is successfully connected, and the rssi value will not change regardless of how much the device distance changes. There is no way to update rssi values at regular intervals.

pschatzmann, 2024-04-07:
Using the latest version you can call [update_rssi](https://pschatzmann.github.io/ESP32-A2DP/html/class_bluetooth_a2_d_p_sink.html#a3b43ee67031c300d07638f9c969fa4ef) ().

In your version you can call  esp_bt_gap_read_rssi_delta(*get_current_peer_address()); to request an update

closed 2024-04-10 by pschatzmann