pschatzmann/ESP32-A2DP #552: default AVRCP raises volume instead of lowering if volume is below ~128

modpotato, 2024-05-10:
### Problem Description

when pressing the volume down button on a bluetooth speaker (using a2dp source) if the volume is greater than 127 (approx) it raises the volume instead. 

### Device Description

esp32-wroom-32d

### Sketch

```cpp
#include "BluetoothA2DPSource.h"
#include <Arduino.h>  // for random()

BluetoothA2DPSource a2dp_source;

// callback
int32_t get_sound_data(Frame *data, int32_t frameCount) {
  // generate white noise
  for (int i = 0; i < frameCount * 2; i++) {
    int16_t sample = random(-32768, 32767);
    data[i] = sample * 0.01;  // apply volume level
  }

  return frameCount;
}

void setup() {
  randomSeed(analogRead(A0));  // seed the random number generator
  a2dp_source.start("SRS-XB23", get_sound_data);
}

void loop() {
}
```

### Other Steps to Reproduce

specifically using a sony SRS-XB23 bluetooth speaker

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.13

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-11:
Strange that your speaker is sending values > 127.
I committed a correction,  but I am not able to test this myself since I don't have any speaker with volume control.

modpotato, 2024-05-11:
sorry for the late reply, it seems to have caused a different problem. when connecting the volume is set to 0 and from there + and - both add more volume until it hits the limit, which then the + button does nothing and the - button resets it to 0 (or 2 for some reason)

the following logs went as follows:
(speaker) power on -> (speaker) bluetooth pairing -> (esp) connect -> (speaker) volume down x10 ->(speaker) volume up x20 

https://logpaste.com/PG0DSl7e

pschatzmann, 2024-05-11:
[Second trial](https://github.com/pschatzmann/ESP32-A2DP/commit/aeb07ef844d6c307ea6fb70610bd0abee6690dad)

modpotato, 2024-05-11:
thanks, thats solved the volume behaving oddly at high levels but the volume down button still doesnt work, these are the logs for when pressing volume down: https://logpaste.com/B1yOlSk7

would it be possible to add a callback any time an avrcp request is made for a volume change?

pschatzmann, 2024-05-11:
I am printing all events that have been received...

modpotato, 2024-05-11:
the speaker works fine with changing the volume of my phone (ios 16) so i assume its not at fault, my only other idea is something with these lines

`15:24:10.894 -> [ 73759][I][BluetoothA2DPSource.cpp:928] bt_av_notify_evt_handler(): [RCCT] Volume changed: 125
15:24:10.894 -> [ 73769][I][BluetoothA2DPSource.cpp:931] bt_av_notify_evt_handler(): [RCCT] Set absolute volume: volume 127`

it *seems* as though its being notified to lower the volume, yet sets the volume to another value though i could be wrong as i havent really looked at the source

pschatzmann, 2024-05-11:
This logic has been taken from the Espressif example code: For some reason they increase the received value always by 5.
If you press the volume down, I would expect that we should receive decreased [RCCT] Volume changed: values.

There is no logic in place to increase or decrease values: we just take the values that have been received. You can try to play around with the code to figure out what is going on...

modpotato, 2024-05-11:
it appears all calls increase volume by 5 for literally by no reason.
https://github.com/pschatzmann/ESP32-A2DP/blob/aeb07ef844d6c307ea6fb70610bd0abee6690dad/src/BluetoothA2DPSource.cpp#L930

my speaker goes through values 3 at a time already so it doesnt seem to need to be multiplied, but upon removing the +5 (rather replacing +5 for + 0 because i dont remember much c++) it works perfectly fine, if you could implement this in a way that isnt incredibly scuffed like mine id greatly appreciate it!

pschatzmann, 2024-05-11:
[I removed the addition of 5](https://github.com/pschatzmann/ESP32-A2DP/commit/5af2573f42c7cba49bf4fdff0b2a922205164e32)

modpotato, 2024-05-11:
my reason for doing + 0 was because removing it threw this same error
[image]

pschatzmann, 2024-05-11:
Ok, I see. It is compiling now again...

My guess the +5 is was supposed to prevent a user from setting the volume to 0.  A 0 value has the risk that the user might forget that the device is still switched on

closed 2024-05-11 by modpotato