pschatzmann/ESP32-A2DP #557: Sending music from Atom Echo (ESP32) to Ubuntu VM.

xG3nesis, 2024-05-23:
### Problem Description

Hello ! 
First, I would like to thank you for this wonderful repository ! 

I'm attempting to flash my M5 Atom Echo (ESP32) with the "bt_music_sender" example to stream music to my Ubuntu VM (22.04) on which there is a bluetooth dongle connected. I've set my ubuntu VM so that it can act as an A2DP sink by following this [tutorial](https://mendel-nykorowycz.pl/2021/05/01/raspberry-pi-debian-a2dp-sink-or-how-to-make-your-own-bluetooth-speaker/). Once configured, I can listen to music on the VM when i stream audio through my phone.

But when i attempt to stream "music" by using the "bt_music_sender" example, i'm not able to hear anything 🤔.

I'm quite a noob, so I'm really sorry if I've misunderstood a step or any kind of information that might help me to stream music.

Here are some logs : 
```
[ 51981][I][BluetoothA2DPSource.cpp:475] bt_app_gap_callback(): [BT_AV] Discovery started.
[ 60628][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 60628][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
[ 60638][I][BluetoothA2DPSource.cpp:655] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_DISCOVERING, evt 0xff00
[ 60649][D][BluetoothA2DPSource.cpp:622] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[ 64165][D][BluetoothA2DPSource.cpp:456] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
[ 64166][I][BluetoothA2DPSource.cpp:389] filter_inquiry_scan_result(): [BT_AV] Scanned device: 00:e0:4c:9a:a9:03
[ 64173][I][BluetoothA2DPSource.cpp:395] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x7c0000
[ 64182][I][BluetoothA2DPSource.cpp:399] filter_inquiry_scan_result(): [BT_AV] --RSSI: -48
[ 64190][I][BluetoothA2DPSource.cpp:419] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
[ 64200][I][BluetoothA2DPSource.cpp:421] filter_inquiry_scan_result(): [BT_AV] --Name: AudioTest
[ 64208][D][BluetoothA2DPSource.cpp:433] filter_inquiry_scan_result(): [BT_AV] --Checking match: AudioTest
[ 64218][I][BluetoothA2DPSource.cpp:442] filter_inquiry_scan_result(): [BT_AV] --Result: Target device found
[ 64227][D][BluetoothA2DPCommon.cpp:180] set_last_connection(): [BT_AV] set_last_connection: 00:e0:4c:9a:a9:03
[ 64237][D][BluetoothA2DPCommon.cpp:184] set_last_connection(): [BT_AV] no change!
[ 64244][I][BluetoothA2DPSource.cpp:446] filter_inquiry_scan_result(): [BT_AV] Cancel device discovery ...
[ 64256][D][BluetoothA2DPSource.cpp:456] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
[ 64262][I][BluetoothA2DPSource.cpp:466] bt_app_gap_callback(): [BT_AV] Device discovery stopped.
[ 64271][I][BluetoothA2DPSource.cpp:467] bt_app_gap_callback(): [BT_AV] a2dp connecting to peer: AudioTest
[ 64280][I][BluetoothA2DPSource.h:271] esp_a2d_connect(): [BT_AV] a2dp connecting to: 00:e0:4c:9a:a9:03
[ 64295][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 64300][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
[ 64310][I][BluetoothA2DPSource.cpp:655] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTING, evt 0x0
[ 64321][D][BluetoothA2DPSource.cpp:622] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[ 64331][D][BluetoothA2DPSource.cpp:631] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connecting
[ 64344][D][BluetoothA2DPSource.cpp:712] bt_app_av_state_connecting_hdlr(): [BT_AV] bt_app_av_state_connecting_hdlr evt 0
[ 65713][D][BluetoothA2DPSource.cpp:456] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 16
[ 65714][I][BluetoothA2DPSource.cpp:536] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT
[ 66493][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[ 66493][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
[ 66503][I][BluetoothA2DPSource.cpp:655] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTING, evt 0x0
[ 66514][D][BluetoothA2DPSource.cpp:622] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[ 66524][D][BluetoothA2DPSource.cpp:631] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connected
[ 66537][D][BluetoothA2DPSource.cpp:712] bt_app_av_state_connecting_hdlr(): [BT_AV] bt_app_av_state_connecting_hdlr evt 0
[ 66547][I][BluetoothA2DPSource.cpp:720] bt_app_av_state_connecting_hdlr(): [BT_AV] a2dp connected
[ 68843][D][BluetoothA2DPSource.cpp:889] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 0
[ 68844][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 12
[ 6 681]5D][DluetluehA2DhA2DPSeocrc:307] p:_8pp_ ask_hppnrlect_: [B: API] bT] bpp_tpp__hancler, sigt 5:
0[16 8ve0t: 0x][Bl[ etoot][2DPBlurtootcp2pP257rc .c_ap942_ or__vdidpaavrc): _evtT_A [RC bt _t_av_orlkadic_ct_ech  vt 0
thA2DPSource.cpp:952] bt_av_hdl_avrc_ct_evt(): [RCCT] AVRC conn_state event: state 1, [00:e0:4c:9a:a9:03]
[ 68881][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x5
[ 68891][D][BluetoothA2DPSource.cpp:942] bt_av_hdl_avrc_ct_evt(): [RCCT] bt_av_hdl_avrc_ct_evt evt 5
[ 68900][I][BluetoothA2DPSource.cpp:991] bt_av_hdl_avrc_ct_evt(): [RCCT] AVRC remote features 25b, TG features 5f
[ 68910][D][BluetoothA2DPSource.cpp:889] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 6
[ 68918][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x6, param len 12
[ 68928][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x6
[ 68939][D][BluetoothA2DPSource.cpp:942] bt_av_hdl_avrc_ct_evt(): [RCCT] bt_av_hdl_avrc_ct_evt evt 6
[ 68948][I][BluetoothA2DPSource.cpp:1000] bt_av_hdl_avrc_ct_evt(): [RCCT] remote rn_cap: count 8, bitmask 0x2d1e
[ 70628][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[ 70628][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
[ 70638][I][BluetoothA2DPSource.cpp:655] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0xff00
[ 70649][D][BluetoothA2DPSource.cpp:622] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
[ 70659][D][BluetoothA2DPSource.cpp:755] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 65280
[ 70670][D][BluetoothA2DPSource.cpp:828] bt_app_av_media_proc(): [BT_AV] bt_app_av_media_proc evt 65280
[ 70679][I][BluetoothA2DPSource.cpp:833] bt_app_av_media_proc(): [BT_AV] a2dp media ready checking ...
[ 70688][D][BluetoothA2DPSource.cpp:257] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x3, param len 16
[ 70699][D][BluetoothA2DPSource.cpp:307] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x3
[... 67 lines cut ...]
    }
    // to prevent watchdog
    delay(1);

    return frame_count;
}

void setup() {
  a2dp_source.set_auto_reconnect(false);
  a2dp_source.start("AudioTest", get_data_frames);  
  a2dp_source.set_volume(30);
}

void loop() {
  delay(1000);
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.14

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-05-24:
I have never tried this and I don't think that I will be able to provide some support for linux
What are the error messages you get in the Linux log ?

xG3nesis, 2024-06-08:
Hi @pschatzmann,

Thank you so much for your help. I successfully got everything working a week ago, but I've been too busy to update you until now.

In a few days, I plan to post a GitHub repository detailing how I got my "Atom Echo" to function as a Bluetooth microphone, using your repository as a reference. It will include instructions on how to have it recognized on Linux machines (specifically Raspberry Pi OS on a Raspberry Pi 5) as an A2DP-Source, allowing voice data to stream directly to an STT engine in Home Assistant.

If you're interested, I'd be happy to keep you informed.

Have a great day !
@xG3nesis

closed 2024-06-08 by xG3nesis