pschatzmann/ESP32-A2DP #564: Audio state starts with 'playing', but no Bluetooth Connection exist

harryberlin, 2024-06-03:
### Problem Description

It's a bit confuse thing for me. 
After startup, i have added a simple audio state request `a2dp_sink.get_audio_state()`, which returns `1` for state play. 
Bluetooth is not connected at this moment.

### Device Description

ESP Wroom 32D

### Sketch

```cpp
//simple state request in setup() 
a2dp_sink.get_audio_state()
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.16

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-06-04:
You obviously did not check the [documentation](https://pschatzmann.github.io/ESP32-A2DP/html/class_bluetooth_a2_d_p_common.html#a74eadbd69b5c7adf1b190c7e41b75b10): This method is returning a esp_a2d_audio_state_t and when you [double check in the IDF documentation](https://docs.espressif.com/projects/esp-idf/en/release-v3.0/api-reference/bluetooth/esp_a2dp.html#_CPPv421esp_a2d_audio_state_t)  1 really means ESP_A2D_AUDIO_STATE_STOPPED

closed 2024-06-04 by pschatzmann

harryberlin, 2024-06-04:
sorry for wasting your time.
found out what's my mistake. i did use the two examples `bt_music_receiver_status_callback` and `bt_music_receiver_playing_status_callbacks`
I thought the a2dp_sink.get_audio_state() does return the same state like callback.