pschatzmann/ESP32-A2DP #571: Cant upload sketch to esp32

CasualGuy33, 2024-06-09:
### Problem Description

So i bought the esp32 wroom 32d devboard and an amplifier which has a 3.5 mm jack in so i needed a sketch that will allow my esp32 to output jack signal. I also provided my sketch. My problem is when i press upload it starts uploading but stops and says "fatal error: xtensa_api.h: No such file or directory" i been searching for some fixes but i didnt found any. :c 

### Device Description

ESP32 WROOM 32D

### Sketch

```cpp
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream i2s;
BluetoothA2DPSink a2dp_sink(i2s);

void setup() {
    a2dp_sink.start("MyMusic");
}

void loop() {
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

idk where i could find EP32 Arduino Core or IDF version, sorry :c

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

jberezin, 2024-06-11:
I am having the same problem using  Arduino IDE and expressif board support 3.0.1.  I tried with Wroom, firebeetle 32, and ttgo t1 boards with identical error.  I had to go to an older 2.x version of expressif to work around it for the time being.

pschatzmann, 2024-06-11:
I committed a [correction](https://github.com/pschatzmann/ESP32-A2DP/commit/e206656606a3fbc0244a90c09a4b13ca093e517e).

closed 2024-06-11 by pschatzmann