pschatzmann/ESP32-A2DP #572: Arduino print not writw data and stuck in a loop

DenysChuhlib, 2024-06-11:
### Problem Description

I use I2SClass in ESP_I2S.h as output and noticed that they do not output data at all and get stuck, so I looked at the code and found the reason:
```cpp
/*
//In BluetoothA2DPOutput.h
*/

class BluetoothA2DPOutputAudioTools : public BluetoothA2DPOutput {
 public:
  BluetoothA2DPOutputAudioTools() = default;
  bool begin() override;
  size_t write(const uint8_t *data, size_t len) override;
  void end() override;
  void set_sample_rate(int rate) override;
  void set_output_active(bool active) override;

#if A2DP_I2S_AUDIOTOOLS
  operator bool() { return p_print != nullptr; }

  /// Output AudioStream using AudioTools library
  void set_output(AudioOutput &output) {
    p_print = &output;
    p_audio_print = &output;
  }

  /// Output AudioStream using AudioTools library
  void set_output(AudioStream &output) {
    static AdapterAudioStreamToAudioOutput adapter(output);
    adapter.setStream(output);
    p_print = &output;
    p_audio_print = &adapter;
  }
#elif ARDUINO   //START NEW CODE
  operator bool() { return !!p_print; }

  /// Output to Arduino Print
  void set_output(Print &output) { p_print = &output; }
#else
    
  operator bool() { return false; }
  
#endif //END NEW CODE

 protected:
#if defined(ARDUINO) || A2DP_I2S_AUDIOTOOLS
  Print *p_print = nullptr;
#endif

#if A2DP_I2S_AUDIOTOOLS
  AudioOutput *p_audio_print = nullptr;
#endif
};

/*
//In BluetoothA2DPOutput.cpp
*/

size_t BluetoothA2DPOutputAudioTools::write(const uint8_t *data, size_t item_size) {
  size_t i2s_bytes_written = 0;

#if defined(ARDUINO) || A2DP_I2S_AUDIOTOOLS
  if (p_print != nullptr) {
    i2s_bytes_written = p_print->write(data, item_size);
  }
#endif
  return i2s_bytes_written;
}
```

### Device Description

ESP32 Wroom

### Sketch

```cpp
#include "ESP_I2S.h"
#include "BluetoothA2DPSink.h"

I2SClass i2s;
BluetoothA2DPSink a2dp_sink(i2s);

void setup() {

    if (!i2s.begin(I2S_MODE_STD, 44100, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO)) {
      Serial.println("Failed to initialize I2S!");
      while (1); // do nothing
    }

    a2dp_sink.start("MyMusic");
}

void loop() {
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.1

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-06-12:
I [committed a correction](https://github.com/pschatzmann/ESP32-A2DP/commit/2d64e7747893842bd9b86150967f46dfd094d03c). Unfortunately you did not describe what exactly the issues were so I hope I got it right

DenysChuhlib, 2024-06-12:
So you got it right. The problem is that he still thought there was no print, and if there was, he didn't record it.

closed 2024-06-12 by DenysChuhlib