pschatzmann/ESP32-A2DP #574: Example keeps resetting

Erklei, 2024-06-12:
### Problem Description

On arduino bt_music_receiver_simple keeps resetting. All libraries downloaded on 2024-06-11. So they are latest. 

To get it compile, I had to change  line 51 in "BluetoothA2DPCommon.h" first to #include "xtensa/xtensa_api.h"

if I comment out a2dp_sink.start("MyMusic");  it does not reset.

Something is broken, but I have no clue, what. 

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1448
load:0x40078000,len:14828
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
[    24][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 196160
[    32][I][BluetoothA2DPSink.cpp:102] start(): [BT_AV] Device name will be set to 'MyMusic'
[    43][I][BluetoothA2DPCommon.cpp:166] get_last_connection(): [BT_AV] nvs_blob does not exist
[    53][I][BluetoothA2DPSink.cpp:165] init_i2s(): [BT_AV] init_i2s
[    61][I][BluetoothA2DPOutput.cpp:199] begin(): [BT_AV] begin
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1448
load:0x40078000,len:14828
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
18:05:45: [    24][I][BluetoothA2DPCommon.cpp:246] log_free_heap(): [BT_AV] Available Heap: 196160
[    32][I][BluetoothA2DPSink.cpp:102] start(): [BT_AV] Device name will be set to 'MyMusic'
[    43][I][BluetoothA2DPCommon.cpp:166] get_last_connection(): [BT_AV] nvs_blob does not exist
[    53][I][BluetoothA2DPSink.cpp:165] init_i2s(): [BT_AV] init_i2s
[    61][I][BluetoothA2DPOutput.cpp:199] begin(): [BT_AV] begin

### Device Description

ESP32-WROOM-32E
[image]

### Sketch

```cpp
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream out;
BluetoothA2DPSink a2dp_sink(out);

void setup() {
  a2dp_sink.start("MyMusic");  
}

void loop() {
  delay(1000); // do nothing
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.1

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

Erklei, 2024-06-12:
Ok, this one I figured out - 3.3V LDO could not deliver enough current at the moment of BT turn on.

Now I got another watchdog reset:

rst:0x7 (TG0WDT_SYS_RESET),boot:0xc (SPI_FLASH_BOOT)
ets_main.c 397 
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0xc (SPI_FLASH_BOOT)
ets_main.c 397 
18:43:18: ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0xc (SPI_FLASH_BOOT)
ets_main.c 397 
ets Jul 29 2019 12:21:46

Arduino main file does not have line 397...

Erklei, 2024-06-12:
Ok, I close it. The thing works after solving power issues. It takes 40 - 50 mA @ 3.3V after BT is started. I assume ESP has very high current spice for a short time. Need more capacitors.

closed 2024-06-12 by Erklei

bobisaperson1, 2024-06-25:
I am having an identical issue but power isnt a problem and commenting out "a2dp_sink.start("MyMusic");" doesnt solve the issue like it did for Erklei

pschatzmann, 2024-06-25:
Are you sure that a regular imple hallo world sketch is working with your current settings ? 
W/o knowing your hardware a flash size of 8MB seems strange to me...

bobisaperson1, 2024-06-25:
I found the solution: I had the "ESP32 Dev Module" selected as the board (is a esp32 dev board) in the IDE, changed it to "ESP32-WROOM-DA Module" and made sure to change to Huge APP partition scheme again and its now working