pschatzmann/ESP32-A2DP #576: Example triggering a Sketch too big Error

NiNeu01, 2024-06-15:
### Problem Description

I tried run the bt_music_receiver_to_internal_dac example and it gave me following error message:

Sketch uses 1902589 bytes (145%) of program storage space. Maximum is 1310720 bytes.
Global variables use 62404 bytes (19%) of dynamic memory, leaving 265276 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

### Device Description

ESP-WROOM-32

### Sketch

```cpp
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

AnalogAudioStream out;
BluetoothA2DPSink a2dp_sink(out);

void setup() {
  a2dp_sink.start("InternalDAC");  
}

void loop() {
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.1

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-06-15:
https://github.com/pschatzmann/ESP32-A2DP/issues/575

closed 2024-06-15 by pschatzmann

Sskki-exe, 2024-06-18:
I am new to Github and I didn't know that completed issues were organised in a separate tab. Sorry for my mistake and thank you for answering anyway.