pschatzmann/ESP32-A2DP #580: Compilation Error

nemoscaos, 2024-06-20:
### Problem Description

the program worked correctly
then it started to fail the compilation
I tried on two different computers but the error is the same
maybe there are new updates that create problems for the library??

### Device Description

Esp32 Wroover

### Sketch

```cpp
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_av_hdl_stack_evt(uint16_t, void*)':
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSource.cpp:566:97: error: invalid conversion from 'void (*)(void*)' to 'TimerCallbackFunction_t' {aka 'void (*)(tmrTimerControl*)'} [-fpermissive]
  566 |             s_tmr = xTimerCreate("connTmr", (10000 / portTICK_RATE_MS), pdTRUE, (void *)tmr_id, ccall_a2d_app_heart_beat);
      |                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                 |
      |                                                                                                 void (*)(void*)
In file included from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPCommon.h:26,
                 from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSource.h:22,
                 from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSource.cpp:16:
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/timers.h:241:57: note:   initializing argument 5 of 'tmrTimerControl* xTimerCreate(const char*, TickType_t, UBaseType_t, void*, TimerCallbackFunction_t)'
  241 |                                 TimerCallbackFunction_t pxCallbackFunction ) PRIVILEGED_FUNCTION;
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp: In member function 'virtual esp_err_t BluetoothA2DPSink::i2s_mclk_pin_select(uint8_t)':
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:218:29: error: 'PERIPHS_IO_MUX_GPIO0_U' was not declared in this scope
  218 |             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
      |                             ^~~~~~~~~~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:218:53: error: 'FUNC_GPIO0_CLK_OUT1' was not declared in this scope
  218 |             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
      |                                                     ^~~~~~~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:218:13: error: 'PIN_FUNC_SELECT' was not declared in this scope
  218 |             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
      |             ^~~~~~~~~~~~~~~
In file included from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/esp_common/include/esp_macros.h:13,
                 from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:76,
                 from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPCommon.h:25,
                 from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.h:17,
                 from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:16:
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:219:28: error: 'PIN_CTRL' was not declared in this scope
  219 |             WRITE_PERI_REG(PIN_CTRL, 0xFFF0);
      |                            ^~~~~~~~
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/esp_common/include/esp_assert.h:24:38: note: in definition of macro 'TRY_STATIC_ASSERT'
   24 |             if (__builtin_constant_p(CONDITION) && !(CONDITION)) {          \
      |                                      ^~~~~~~~~
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/soc.h:34:57: note: in expansion of macro 'IS_DPORT_REG'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
      |                                                         ^~~~~~~~~~~~
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/soc.h:113:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
  113 |             ASSERT_IF_DPORT_REG((addr), WRITE_PERI_REG);                                                               \
      |             ^~~~~~~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:219:13: note: in expansion of macro 'WRITE_PERI_REG'
  219 |             WRITE_PERI_REG(PIN_CTRL, 0xFFF0);
      |             ^~~~~~~~~~~~~~
In file included from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/newlib/platform_include/sys/reent.h:9,
                 from /Users/andrea/Library/Arduino15/packages/esp32/tools/esp-x32/2302/xtensa-esp32-elf/sys-include/stdio.h:60,
                 from /Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPCommon.h:19:
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:219:28: error: 'PIN_CTRL' was not declared in this scope
  219 |             WRITE_PERI_REG(PIN_CTRL, 0xFFF0);
      |                            ^~~~~~~~
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/soc.h:34:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
      |                                      ^~~~~~~~~~~~~~~~~
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/soc.h:34:57: note: in expansion of macro 'IS_DPORT_REG'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
[... 94 lines cut ...]
  113 |             ASSERT_IF_DPORT_REG((addr), WRITE_PERI_REG);                                                               \
      |             ^~~~~~~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:227:13: note: in expansion of macro 'WRITE_PERI_REG'
  227 |             WRITE_PERI_REG(PIN_CTRL, 0xFF00);
      |             ^~~~~~~~~~~~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:227:28: error: 'PIN_CTRL' was not declared in this scope
  227 |             WRITE_PERI_REG(PIN_CTRL, 0xFF00);
      |                            ^~~~~~~~
/Users/andrea/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/soc/esp32/include/soc/soc.h:25:34: note: in definition of macro 'ETS_UNCACHED_ADDR'
   25 | #define ETS_UNCACHED_ADDR(addr) (addr)
      |                                  ^~~~
/Users/andrea/Documents/Arduino/libraries/ESP32-A2DP-1.7.0/src/BluetoothA2DPSink.cpp:227:13: note: in expansion of macro 'WRITE_PERI_REG'
  227 |             WRITE_PERI_REG(PIN_CTRL, 0xFF00);
      |             ^~~~~~~~~~~~~~

exit status 1
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Arduino IDE 2.3.2

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

closed 2024-06-20 by pschatzmann

nemoscaos, 2024-06-20:
I updated the libraries and launched the 32 bit receiver but the functions #include "AudioTools.h"
#include "AudioLibs/AudioBoardStream.h" // install https://github.com/pschatzmann/arduino-audio-driver
where am I ? they are not there in the link indicated
Thank you

pschatzmann, 2024-06-20:
I don't understand: what are you trying to do ?
The AudioBoardStream class is part of the AudioTools and requires the audio-driver project. This is however only necessary if you have a configurable codec...

nemoscaos, 2024-06-20:
I'm just trying to make bluetooth work on my Esp32 wrover, the old program doesn't work anymore and the new libraries don't compile because they're missing these drivers

pschatzmann, 2024-06-20:
You must be doing something wrong!
What is your hardware ?

nemoscaos, 2024-06-21:
I use esp32 wroover
first I compiled the bt_music_receiver_32bit_ext example and it worked without problems, now I was trying
bt_music_receiver_32bits which seems the same to me
but it doesn't find the libraries: "AudioTools.h" and "AudioLibs/AudioBoardStream.h", if I search for them manually I can't find them
are in Arduino audio drivers??

pschatzmann, 2024-06-21:
Remove the include to AudioLibs/AudioBoardStream.h: that is not needed!

nemoscaos, 2024-06-21:
HI
I saw that you updated the library,
I remapped the pins, solved the space problem
but I saw that the command a2dp_sink.i2s_mclk_pin_select
has become obsolete, how do I generate an MCLK?

thank you

pschatzmann, 2024-06-21:
It is defined along the other pins  e.g.  config.pin_mck = 0;