pschatzmann/ESP32-A2DP #589: ESP crash after connection

yoann56100, 2024-07-26:
### Problem Description

my ESP crash after connection my phone

### Device Description

ESP32 Wroom

### Sketch

```cpp
#include "BluetoothA2DPSink.h"
#include "FISLib.h"

#define NEXT_BUTTON_PIN 19

#define PREVIOUS_BUTTON_PIN 21

//FIS
#define pinENABLE 34
#define pinCLOCK 32
#define pinDATA 35
FISLib LCD(pinENABLE, pinCLOCK, pinDATA);

int lastState = LOW;  // the previous state from the input pin
int NEXTcurrentState = 1;   // the current reading from the input pin
int PREVIOUScurrentState = 1;   // the current reading from the input pin
int currentMusicState;
int timer = 0;
BluetoothA2DPSink a2dp_sink;

void setup() {
  a2dp_sink.start("Audi A6");
  Serial.begin(9600);
  Serial.print("Serial started");
  pinMode(NEXT_BUTTON_PIN, INPUT_PULLUP);
  pinMode(PREVIOUS_BUTTON_PIN, INPUT_PULLUP);

 //FIS
  for(int i=0; i<8; i++){
    LCD.showText("IS FIS","BLOCKS!");
    delay(500);
  }
  for(int i=0; i<8; i++){
    LCD.showText("HI! I AM","A A6 :)");
    delay(500);
  }

}

void loop() {
  int NEXTcurrentState = digitalRead(NEXT_BUTTON_PIN);
  int PREVIOUScurrentState = digitalRead(PREVIOUS_BUTTON_PIN);

if(NEXTcurrentState == LOW){ 
   timer += 1;
   delay(50);
   //Serial.println(timer);

     if(timer == 1){
      Serial.println("Next button press");
      a2dp_sink.next();
      }

     if (timer >= 2 && NEXTcurrentState == LOW){
      timer += 1;
      //Serial.println("Timer continue1");
      }
      
}

if(PREVIOUScurrentState == HIGH){ 
   timer += 1;
   delay(50);
      //Serial.println(timer);
     if(timer == 1){
      Serial.println("PREVIOUS button press");
      a2dp_sink.previous();
      }

     if (timer >= 2 && PREVIOUScurrentState == HIGH){
      timer += 1;
      //Serial.println("Timer continue2");
      }
}
     if (timer >= 1 && PREVIOUScurrentState == LOW && NEXTcurrentState == HIGH ){
      Serial.println("set timer to 0");

      timer = 0;
  }
 }
```

### Other Steps to Reproduce

just connect to the esp

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.13

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

yoann56100, 2024-07-26:
this is the error in the serial
18:14:19.763 -> E (36932) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
18:14:19.857 -> E (36932) task_wdt:  - IDLE0 (CPU 0)
18:14:19.903 -> E (36932) task_wdt: Tasks currently running:
18:14:19.949 -> E (36932) task_wdt: CPU 0: BTC_TASK
18:14:19.996 -> E (36932) task_wdt: CPU 1: IDLE1
18:14:20.043 -> E (36932) task_wdt: Aborting.
18:14:20.043 -> E (36932) task_wdt: Print CPU 0 (current core) backtrace
18:14:20.136 -> 
18:14:20.136 -> 
18:14:20.136 -> 
18:14:20.136 -> 
18:14:20.136 -> Backtrace: 0x4019832d:0x3ffd0610 0x400d2f71:0x3ffd0630 0x40198159:0x3ffd0650 0x400d24c3:0x3ffd0670 0x401191c1:0x3ffd0690 0x4011184d:0x3ffd06d0 0x401117e9:0x3ffd06f0 0x400963ba:0x3ffd0720
18:14:20.323 -> 
18:14:20.323 -> 
18:14:20.323 -> 
18:14:20.323 -> 
18:14:20.323 -> ELF file SHA256: f95e53a8e1842737
18:14:20.370 -> 
18:14:20.417 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
18:14:20.463 -> Debug exception reason: BREAK instr 
18:14:20.510 -> Core  0 register dump:
18:14:20.556 -> PC      : 0x400803c0  PS      : 0x00040336  A0      : 0x00000000  A1      : 0x3ffc43a0  
18:14:20.649 -> A2      : 0x40091117  A3      : 0x4009110c  A4      : 0x40091117  A5      : 0x00000000  
18:14:20.743 -> A6      : 0x00000000  A7      : 0x3ffd0550  A8      : 0x40083f0a  A9      : 0x00000000  
18:14:20.836 -> A10     : 0x00000000  A11     : 0x435cbe77  A12     : 0x435cbe77  A13     : 0x00060521  
18:14:20.930 -> A14     : 0xfffbfff0  A15     : 0x00000046  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
18:14:21.024 -> EXCVADDR: 0x00060511  LBEG    : 0x40090fa0  LEND    : 0x40090fb6  LCOUNT  : 0xffffffff  
18:14:21.117 -> 
18:14:21.117 -> 
18:14:21.117 -> Backtrace: 0x400803bd:0x3ffc43a0
18:14:21.163 -> 
18:14:21.163 -> 
18:14:21.163 -> 
18:14:21.163 -> 
18:14:21.163 -> ELF file SHA256: f95e53a8e1842737
18:14:21.210 -> 
18:14:21.210 -> Re-entered core dump! Exception happened during core dump!
18:14:21.256 -> Rebooting...

pschatzmann, 2024-07-26:
Issues are supposed to be used for reporting bugs in the A2DP library and not for the bugs that you have introduced in your code.

You are obviously starving the processor in your loop: Try to add some delay.

I also recommend to google the topic denouncing of input or use the [functionality provided by my AudioTools](https://github.com/pschatzmann/arduino-audio-tools/wiki/Interactive-Audio)

closed 2024-07-26 by pschatzmann