pschatzmann/ESP32-A2DP #590: SoundData class declaration/implementation causes linker error

brodisattva, 2024-08-02:
### Problem Description

I built a simple test program based off of the bt_music_sender_write demo. I'm using the esp-idf framework, so I modified the code enough to compile in that environment. My development environment is Arch linux.

When I build using **idf.py build** I get a link error that ends with

```/home/.../.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/ESP32-A2DP/libESP32-A2DP.a(SoundData.cpp.obj):(.literal._ZN19OneChannelSoundDataC2EPslb11ChannelInfo+0x0): undefined reference to `_ZTV9SoundData'```

And when I look the output of **nm --demangle libESP32-A2DP.a**, I can observe that the vtable for SoundData is undefined:
```
00000000 T SoundData::doLoop()
00000000 T SoundData::setLoop(bool)
00000000 V vtable for OneChannelSoundData
00000000 V vtable for TwoChannelSoundData
00000000 V vtable for OneChannel8BitSoundData
         U vtable for SoundData
```
which explains the symptom. The linker can't resolve vtable entries because SoundData declares virtual functions but doesn't provide an implementation. One could write default implementations, or just make the virtual functions pure virtual:
```
class SoundData {
  public:
     virtual int32_t get2ChannelData(int32_t pos, int32_t len, uint8_t *data) = 0;
     virtual int32_t getData(int32_t pos, Frame &channels) = 0;
     virtual void setDataRaw( uint8_t* data, int32_t len) = 0;
     /**
      * Automatic restart playing on end
      */
     bool doLoop();
     void setLoop(bool loop);

  private:
     bool automatic_loop;
};
```
This resolves the linker error.

### Device Description

esp32c3. But I have to do **idf.py set-target esp32** and not **idf.py set-target esp32c3,** or a lot of #defines and such don't get set up properly.

### Sketch

```cpp
#include "BluetoothA2DPSource.h"
#include "piano16bit.h"

BluetoothA2DPSource a2dp_source;
SoundData *data = new OneChannelSoundData((int16_t*)piano16bit_raw, piano16bit_raw_len/2);

void setup() {
  //a2dp_source.set_auto_reconnect(false);
  a2dp_source.start("LEXON MINO L");  
  a2dp_source.set_volume(20);
  a2dp_source.write_data(data);
}

void loop() {
  // to prevent watchdog in release > 1.0.6
  vTaskDelay(1000 / portTICK_PERIOD_MS);
}

extern "C" void app_main(void){
  setup();
  while(true){
    loop();
  }
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

esp-idf 5.3.1

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-02:
You are right, this is a bug and I committed a correction.

However please notice that the c3 does not support Bluetooth classic, so A2DP will not work on that processor! 
You can find this information in my documentation but you can also double check this with the Technical Sheet of your processor.
 
Apart of this, I recommend to use the concepts provided by the AudioTools library over the legacy A2DP SoundData,

closed 2024-08-02 by pschatzmann

brodisattva, 2024-08-02:
Thanks, that's good to know! I will look at AudioTools.