pschatzmann/ESP32-A2DP #599: Task watchdog got triggered on CPU 0

fesch, 2024-08-23:
### Problem Description

While the basic BT receiving example works fine when playing music through my iPad Pro oder Huawei P30, the application crashes as soon as I start a couple of seconds on my Samsung S24+.

> E (74637) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
> E (74637) task_wdt:  - IDLE0 (CPU 0)
> E (74637) task_wdt: Tasks currently running:
> E (74637) task_wdt: CPU 0: BTC_TASK
> E (74637) task_wdt: CPU 1: loopTask
> E (74637) task_wdt: Aborting.
> E (74637) task_wdt: Print CPU 0 (current core) backtrace

### Device Description

I'm using the DOIT ESP32 DEVKIT, but tried also different other WROM32 based boards. Always with the same result :-(

Downgrading the ESP-IDF to 2.0.17 did not change anything.

### Sketch

```cpp
#include "ESP_I2S.h"
#include "BluetoothA2DPSink.h"

const uint8_t I2S_SCK = 26;       /* Audio data bit clock */
const uint8_t I2S_WS = 22;       /* Audio data left and right clock */
const uint8_t I2S_SDOUT = 25;    /* ESP32 audio data output (to speakers) */
I2SClass i2s;

BluetoothA2DPSink a2dp_sink(i2s);

void setup() {
    i2s.setPins(I2S_SCK, I2S_WS, I2S_SDOUT);
    if (!i2s.begin(I2S_MODE_STD, 44100, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO, I2S_STD_SLOT_BOTH)) {
      Serial.println("Failed to initialize I2S!");
      while (1); // do nothing
    }

    a2dp_sink.start("BT-Box");
}

void loop() {
}
```

### Other Steps to Reproduce

I found similar older issues here, but as in my case the same code works for my old phone but not for my new one, I believe there must be something about how the data is being send from the phone. Maybe there is some kind of too much data being send by newer devices?

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-23:
Your empty loop does never yields any control, so you are starving all other processes!

You have to add a delay, like in all my examples

closed 2024-08-23 by pschatzmann

fesch, 2024-08-23:
Even with a delay(up to 1000) in the loop, a call to yield() it does not work.
I have also tried putting a higher value into the watchdog's timeout.

As said: The **same** code works when playing music on any of my devices, except for the S24+.

pschatzmann, 2024-08-23:
I don't think this is an issue with this library, but rather your S24+ or potentially the IDF.
I am using a Samsumg Galaxy A34 5G and don't have any issues...

You could try to activate the logging and compare the difference between a working Android and your non working Phone to get a hint where things start to go wrong...

planevina, 2024-09-16:
I have the same problem. Esp arduino 3.0.4 , phone is  iphone 13.