pschatzmann/ESP32-A2DP #600: Attempt using ESP32 PCM5102A leads to core problems

MrBlazer335, 2024-08-28:
### Problem Description

After I deploy code and try to connect any device, the PCM5102A doesn't make any sound and after few moments my ESP32 reloads itself with this problem Backtrace: 0x4019bbd0:0x3ffd1530 0x400d3aad:0x3ffd1550 0x4019ba05:0x3ffd1570 0x400d2827:0x3ffd1590 0x4011cacd:0x3ffd15b0 0x40115159:0x3ffd15f0 0x401150f5:0x3ffd1610 0x40096966:0x3ffd1640
****

### Device Description

ESP32 Wroom

### Sketch

```cpp
#include <BluetoothA2DPSink.h>
#include <driver/i2s.h>  

BluetoothA2DPSink a2dp_sink;

void setup() {
  Serial.begin(115200);

  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX),
    .sample_rate = 44100,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = false,
    .tx_desc_auto_clear = true,
    .fixed_mclk = 0
  };

  i2s_pin_config_t pin_config = {
    .bck_io_num = 26,
    .ws_io_num = 25,
    .data_out_num = 22,
    .data_in_num = I2S_PIN_NO_CHANGE
  };

  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  i2s_set_pin(I2S_NUM_0, &pin_config);

  a2dp_sink.start("ESP32_A2DP_Transmitter");
  Serial.print("I am working!");
}

void loop() {
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.3

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-08-28:
You must be joking: why are you using the old unsupported I2S API with 3.0.3 ?
Read the Readme again!

closed 2024-08-28 by pschatzmann

MrBlazer335, 2024-08-28:
will 2.0.17 will work?