pschatzmann/ESP32-A2DP #610: compile error in version v1.8.4 related to A2DP_I2S_MAX_WRITE_SIZE

adrobisch, 2024-10-04:
### Problem Description

Thanks for this awesome library, using it to build an bluetooth mp3 player for my daughter.

After upgrading the library to v1.8.4 the compiler produced this error:
```
BluetoothA2DPSink.cpp:1245:61: error: 'A2DP_I2S_MAX_WRITE_SIZE' was not declared in this scope
```
I was indeed not able to find `A2DP_I2S_MAX_WRITE_SIZE` apart from this usage and some docs html.

### Device Description

Using this in a M5Stack Core2 platformio project

### Sketch

```cpp
just compile with the new version
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

espressif32 6.8.1

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-10-04:
I added it back to the config.h

closed 2024-10-04 by pschatzmann

Nutto55, 2024-10-04:
I also started building Bluetooth speakers today and found the same issue. I guess it happened because the developer removed `A2DP_I2S_MAX_WRITE_SIZE 1024 * 5` on config.h file