pschatzmann/ESP32-A2DP #618: Audio constantly cutting out

Legolor, 2024-10-19:
### Problem Description

When I play audio it is extremely choppy. I have tried playing from an iphone and from windows with the same results.

The setup and wiring works and sounds correct when playing directly from an SD card (ESP32-audioI2S library.)

Is there anything that I can try to improve the Bluetooth connection? Is there a specific esp32 board version that is recommended?

### Device Description

Basic ESP32 Dev Module
MAX98357 I2S Audio Amplifier Module

### Sketch

```cpp
Basic example sketch
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Espressif 32 (6.2.0)

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-10-19:
Isn't this just a duplicate of [this issue](https://github.com/pschatzmann/ESP32-A2DP/issues/533) ?

The only thing that you could try is to play around with the A2DP_I2S_MAX_WRITE_SIZE is config.h

Legolor, 2024-10-26:
I didn't seem like an exact duplicate. I have the exact same issue playing from IOS, a windows computer, and Android. My audio output was also choppier.

> How would you be using IDF 6.2.0 when the latest version is only 5.3 ?

I misunderstood, I am using platform-espressif32 version 6.2.0 which corresponds to ESP-IDF - v5.0.1 & Arduino - v2.0.8

> You could also try to use the Arduino API of I2S instead my AudioTools to rule out that the issue is in my I2SStream.

Same issue.

[Changing the data pin to 19 solved the issue!](https://github.com/pschatzmann/ESP32-A2DP/discussions/582) It now plays perfectly :) For reference I had the same issue with both the MAX98357A and the PCM5102, different speakers, and different power supplies so I believe the issue was with my esp32.

closed 2024-10-26 by Legolor