pschatzmann/ESP32-A2DP #621: Issue with Bluetooth Connection Loss

Paul-TheOctopus, 2024-10-24:
### Problem Description

Hi! Issue with Bluetooth Connection Loss on ESP32 with A2DP Sink
I'm using an ESP32 with the following simple sketch to stream audio via Bluetooth:

However, when I connect two different mobile phones (Samsung A70\Some Honor (not old one)) via Bluetooth and start playing audio, both devices lose the Bluetooth connection immediately after the playback starts or just by vol. control pressing to min and max posisson). The Bluetooth connection seems stable until the audio begins.

I'm using the BluetoothA2DPSink library, and all connection parameters (bit rate, sample rate, etc.) seem to be set correctly. I've checked for potential power supply issues, but the problem persists. Could this be related to buffer settings, resource conflicts, or interference? What could be causing this behavior, and how can I resolve it?

### Device Description

Chip Model: ESP32-D0WD-V3
Chip Cores: 2
Flash Size: 4MB
PSRAM Size: 0MB

Memory Information:
Free Heap: 352664 bytes
Minimum Free Heap: 347284 bytes
Max Alloc Heap: 118772 bytes

System Information:
CPU Frequency: 240 MHz
SDK Version: v5.1.4-828-g33fbade6b8-dirty
Sketch Size: 282032 bytes
Free Sketch Space: 1310720 bytes
Flash Frequency: 80000000 Hz
Cycle Count: 263873208

chip revision: v3.1[0m

### Sketch

```cpp
/*
  Streaming Music from Bluetooth
  
  Copyright (C) 2020 Phil Schatzmann
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

// ==> Example A2DP Receiver which uses I2S to an external DAC
// This creates a new Bluetooth device with the name “MyMusic” and the output will be sent to the following default I2S pins which need to be connected to an external DAC:
//
// bck = 14
// ws = 15
// data_out = 22

#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream out;
BluetoothA2DPSink a2dp_sink(out);

void setup() {
  a2dp_sink.start("MyMusic");  
}

void loop() {
  delay(1000); // do nothing
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Chip Model: ESP32-D0WD-V3

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-10-24:
Never heard of anything like this!

Please provide the ESP32 core version that you can find in the Board Manager.
The first thing to do, would be to actvate the A2DP logging and the AudioTools logging!

- For the AudioToolls add
```C++
Serial.begin(115200);
AudioLogger::instance().begin(Serial, AudioLogger::Debug);
```
- For A2DP, activate the logging in the Tools menu of Arduino

Paul-TheOctopus, 2024-10-24:
My Board selected ESP32 Dev Module.

It seems that BT is not disconnecting now. I have not listened audio yet and am only looking at the SDATA pin with an oscilloscope. 
Here's what's strange - I see spam in the port monitor: 

E (1671429) i2s_common: i2s_channel_write(1109): input parameter 'handle' is NULL [E] I2SESP32V1.h : 111 - size_t audio_tools::I2SDriverESP32V1::writeBytes(const void*, size_t) E (1671444 i2) s_common: i2s_channel_write(1109): input parameter 'handle' is NULL [E] I2SESP32V1.h : 111 - size_t audio_tools::I2SDriverESP32V1::writeBytes(const void*, size_t) E (1671460) i2s_common: i2s_channel_write(1109): input parameter 'handle' is NULL
[E] I2SESP32V1.h : 111 - size_t audio_tools::I2SDriverESP32V1::writeBytes(const void*, size_t)
E (1671475) i2s_common: i2s_channel_write(1109): input parameter 'handle' is NULL

And another strange thing - if I pause the BT stream or turn off the BT on my mobile, I continue to see the SDATA stream with the oscilloscope.

My sketch:

#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream out;
BluetoothA2DPSink a2dp_sink(out);

void setup() {
  Serial.begin(115200);  
  AudioLogger::instance().begin(Serial, AudioLogger::Error); 
  a2dp_sink.start("MyMusic"); 
}

void loop() {
  delay(1000);  

}

And i can't see any A2DP logging settings in the Tools menu of Arduino IDE. Could you clarify what you mean? Thanks!

pschatzmann, 2024-10-24:
I committed a [correction the the AudioTools](https://github.com/pschatzmann/arduino-audio-tools/commit/9b8be07e7d8fefa15f23bf24385446e03bb064e1)...
For some strange reason I2S was opened in read mode....

closed 2024-10-24 by pschatzmann

Paul-TheOctopus, 2024-10-25:
No disconnects! Correct SDATA behavior on stop\pause. TNX!!! Later a wiil connect DAC and listen output. I will remember you after converting the project to coins.