pschatzmann/ESP32-A2DP #622: Namespace problem with AudioTools library

duevil, 2024-10-26:
### Problem Description

When using this library together with [AudioTools](https://github.com/pschatzmann/arduino-audio-tools), declaring `A2DP_I2S_AUDIOTOOLS` to be true and `USE_AUDIOTOOLS_NS` to be false will cause compilation to fail as in some places classes from the AudioTools library are used without heir namespace declaration.

Error output gives these error locations:
- `BluetoothA2DPOutputAudioTools.h:96`
- `BluetoothA2DPOutputAudioTools.cpp:231`

I'm using PlatformIO and am setting the flags directly in the platformio.ini.

### Device Description

DFRobot FireBeetle 2 ESP32-E

### Sketch

```cpp
#include <AudioTools.h>
#include <BluetoothA2DPSink.h>

audio_tools::I2SStream out{};
BluetoothA2DPSink a2dp_sink(out);

void setup() {
    a2dp_sink.start("MyMusic");
}

void loop() {
    delay(1000); // do nothing
}
```

### Other Steps to Reproduce

platformio.ini:
```INI
[env:dfrobot_firebeetle2_esp32e]
platform = espressif32
board = dfrobot_firebeetle2_esp32e
framework = arduino
build_flags =
    -D USE_AUDIOTOOLS_NS=0
    -D A2DP_I2S_AUDIOTOOLS=1
lib_deps =
    https://github.com/pschatzmann/arduino-audio-tools.git#v1.0.0
    https://github.com/pschatzmann/ESP32-A2DP.git
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

5.1.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-10-26:
I committed a correction!
Why would you need this ?

closed 2024-10-26 by pschatzmann

duevil, 2024-10-26:
Thank you for the quick fix!

I am using your library together with [Bounce2](https://github.com/thomasfredericks/Bounce2) and when using the AudioTools namespace I'm running into the problem of multiple `Debouncer` class definitions.

Im therefore quite happy that your library provides an own namespace to avoid this problem :)