pschatzmann/ESP32-A2DP #623: ESP32 as Audio transmitter issue - can't reproduce anything but instead only noise

acarrafa, 2024-10-30:
### Problem Description

https://github.com/pschatzmann/ESP32-A2DP/tree/92a3d8b425c468e32ad9c34dd651800f546454ef/examples/bt_music_sender_write

I was trying to reimplement the code related to the ESP32 transmitter to bluetooth audio device

Could you help me to understand what i'm doing wrong? Thank you for the hard work
Alberto

### Device Description

ESP32 Wroom

### Sketch

```cpp
const unsigned char piano16bit_raw[] = {
  0xf8, 0xff, 0xfe, 0xff, 0xf2, 0xff, 0xfe, 0xff, 0xfb, 0xff, 0xf2, 0xff,
  0xfd, 0xff, 0x01, 0x00, 0xf6, 0xff, 0xfa, 0xff, 0xf4, 0xff, 0xf5, 0xff,
  0xfd, 0xff, 0xff, 0xff, 0xf8, 0xff, 0xfe, 0xff, 0xf4, 0xff, 0x00, 0x00,
  0xf9, 0xff, 0xff, 0xff, 0xfa, 0xff, 0xfb, 0xff, 0xfc, 0xff, 0xf1, 0xff,
  0xff, 0xff, 0xff, 0xff, 0xfa, 0xff, 0xfc, 0xff, 0x01, 0x00, 0xf4, 0xff,
  0xfd, 0xff, 0xfd, 0xff, 0xfc, 0xff, 0xf8, 0xff, 0xf7, 0xff, 0xfc, 0xff,
  0xfe, 0xff, 0xf8, 0xff, 0xfc, 0xff, 0xf9, 0xff, 0xf6, 0xff, 0x03, 0x00,
  0xf5, 0xff, 0xf6, 0xff, 0x00, 0x00, 0xee, 0xff, 0x04, 0x00, 0xfa, 0xff,
  0xff, 0xff, 0xfb, 0xff, 0xf9, 0xff, 0xfa, 0xff, 0xf8, 0xff, 0xfe, 0xff,
  0xfe, 0xff, 0xf5, 0xff, 0xfd, 0xff, 0xfb, 0xff, 0xf9, 0xff, 0xf8, 0xff,
  0xff, 0xff, 0xfb, 0xff, 0xfd, 0xff, 0xfd, 0xff, 0xf4, 0xff, 0xfa, 0xff,
  0xfe, 0xff, 0xf7, 0xff, 0xf7, 0xff, 0xfa, 0xff, 0xf6, 0xff, 0xfe, 0xff,
  0xfc, 0xff, 0xfa, 0xff, 0xfd, 0xff, 0xfd, 0xff, 0xfd, 0xff, 0xf7, 0xff,
  0xfe, 0xff, 0xf7, 0xff, 0xf2, 0xff, 0x05, 0x00, 0xfb, 0xff, 0xf9, 0xff,
  0xfd, 0xff, 0xfb, 0xff, 0xfb, 0xff, 0xf7, 0xff, 0xfb, 0xff, 0x01, 0x00,
  0x00, 0x00, 0xf9, 0xff, 0xff, 0xff, 0xfc, 0xff, 0x02, 0x00, 0xf5, 0xff,
  0xff, 0xff, 0xfe, 0xff, 0x00, 0x00, 0x00, 0x00, 0xfd, 0xff, 0x04, 0x00,
  0xfd, 0xff, 0xfe, 0xff, 0x04, 0x00, 0xfb, 0xff, 0xfe, 0xff, 0xfe, 0xff,
  0xff, 0xff, 0xf7, 0xff, 0x04, 0x00, 0xfa, 0xff, 0xf9, 0xff, 0xff, 0xff,
  0xfb, 0xff, 0x04, 0x00, 0xfb, 0xff, 0xf8, 0xff, 0x07, 0x00, 0xfb, 0xff,
  0xfe, 0xff, 0x06, 0x00, 0xf6, 0xff, 0xff, 0xff, 0x04, 0x00, 0x00, 0x00,
  0x00, 0x00, 0xfc, 0xff, 0x01, 0x00, 0xfd, 0xff, 0xff, 0xff, 0x05, 0x00,
  0xfd, 0xff, 0xfe, 0xff, 0x09, 0x00, 0xf9, 0xff, 0x04, 0x00, 0xfc, 0xff,
  0xfd, 0xff, 0x03, 0x00, 0xff, 0xff, 0x04, 0x00, 0xfc, 0xff, 0xfd, 0xff,
  0x09, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x02, 0x00,
  0x02, 0x00, 0x0b, 0x00, 0xfe, 0xff, 0xfa, 0xff, 0x03, 0x00, 0x02, 0x00,
  0x04, 0x00, 0x04, 0x00, 0xfd, 0xff, 0x03, 0x00, 0x0b, 0x00, 0x02, 0x00,
  0x08, 0x00, 0x06, 0x00, 0xfe, 0xff, 0x04, 0x00, 0x09, 0x00, 0x03, 0x00,
  0x05, 0x00, 0x05, 0x00, 0x05, 0x00, 0x0c, 0x00, 0x03, 0x00, 0x02, 0x00,
  0x03, 0x00, 0x09, 0x00, 0x00, 0x00, 0xff, 0xff, 0x07, 0x00, 0x02, 0x00,
  0x08, 0x00, 0x02, 0x00, 0x04, 0x00, 0x04, 0x00, 0x0f, 0x00, 0xff, 0xff,
  0x08, 0x00, 0x07, 0x00, 0x01, 0x00, 0x07, 0x00, 0x03, 0x00, 0x03, 0x00,
  0x0a, 0x00, 0x06, 0x00, 0x03, 0x00, 0x04, 0x00, 0x0a, 0x00, 0x05, 0x00,
  0xfe, 0xff, 0x0e, 0x00, 0x05, 0x00, 0x00, 0x00, 0x0a, 0x00, 0x0b, 0x00,
  0x05, 0x00, 0x02, 0x00, 0x01, 0x00, 0x07, 0x00, 0xfe, 0xff, 0x09, 0x00,
  0x09, 0x00, 0xfc, 0xff, 0x07, 0x00, 0x07, 0x00, 0x03, 0x00, 0x05, 0x00,
  0x04, 0x00, 0x00, 0x00, 0x0a, 0x00, 0x05, 0x00, 0x02, 0x00, 0x01, 0x00,
  0x01, 0x00, 0x05, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x02, 0x00,
  0xff, 0xff, 0x06, 0x00, 0x01, 0x00, 0xfe, 0xff, 0x02, 0x00, 0x05, 0x00,
  0x02, 0x00, 0xfe, 0xff, 0xff, 0xff, 0x01, 0x00, 0xfc, 0xff, 0xfe, 0xff,
  0x08, 0x00, 0x00, 0x00, 0xfe, 0xff, 0x03, 0x00, 0x05, 0x00, 0x02, 0x00,
  0x01, 0x00, 0x01, 0x00, 0x03, 0x00, 0x03, 0x00, 0x02, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x01, 0x00, 0xfb, 0xff, 0x03, 0x00, 0xfa, 0xff, 0xf8, 0xff,
  0x05, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x00, 0xfd, 0xff, 0xfe, 0xff
};

unsigned int piano16bit_raw_len = sizeof(piano16bit_raw);

/*
  Streaming of sound data with Bluetooth to other Bluetooth device.
  We generate 2 tones which will be sent to the 2 channels.
  
  Copyright (C) 2020 Phil Schatzmann
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#include "BluetoothA2DPSource.h"
#include <math.h> 

#define c3_frequency  400.81

BluetoothA2DPSource a2dp_source;
SoundData *data = new OneChannelSoundData((int16_t*)piano16bit_raw, piano16bit_raw_len/2);

// The supported audio codec in ESP32 A2DP is SBC. SBC audio stream is encoded
// from PCM data normally formatted as 44.1kHz sampling rate, two-channel 16-bit sample data
int32_t get_data_frames(Frame *frame, int32_t frame_count) {
    static float m_time = 0.0;
    float m_amplitude = 32.767;  // -32,768 to 32,767
    float m_deltaTime = 1.0 / 44100.0;
    float m_phase = 0.0;
    float pi_2 = PI * 2.0;
    // fill the channel data
    for (int sample = 0; sample < frame_count; ++sample) {
        float angle = pi_2 * c3_frequency * m_time + m_phase;
        frame[sample].channel1 = m_amplitude * sin(angle);
        frame[sample].channel2 = frame[sample].channel1;
        m_time += m_deltaTime;
    }
    // to prevent watchdog
    delay(1);

    return frame_count;
}

// for esp_a2d_connection_state_t see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_a2dp.html#_CPPv426esp_a2d_connection_state_t
void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  Serial.println(a2dp_source.to_str(state));

  if(a2dp_source.to_str(state) == "Connected"){
    a2dp_source.start(get_data_frames);
    Serial.println("Start playing?");
  }
}

void setup() {
  Serial.begin(115200);

  a2dp_source.set_auto_reconnect(true);
  a2dp_source.set_on_connection_state_changed(connection_state_changed);

  a2dp_source.start("MD 45911"); //Bose Mini SoundLink
  a2dp_source.set_volume(30);
}

void loop() {
  // to prevent watchdog in release > 1.0.6
  delay(1000);
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

last released

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-10-30:
Please open issues only for bugs in my software !

There are plenty of examples for using A2DP as source in the AudioTools

closed 2024-10-30 by pschatzmann

acarrafa, 2024-10-31:
Yeah, it's understandable, but with the following examples it is not possibile to create an "HelloWorld" that basically works. Still thank you for the response and have a nice day