pschatzmann/ESP32-A2DP #626: cannot connect from Android 13 and does not detect any devices except neighbors' TV

electrovoyage, 2024-11-06:
### Problem Description

Tried various examples. No matter what I do, the ESP32 does not show up in the bluetooth menu on Android 13.
Also outputs this.
[image]

The only device that gets picked up as 'valid' is our neighbors' TV. I assume the below image shows our TV getting scanned but it is also considered invalid.
[image]

### Device Description

ESP32-WROOM

### Sketch

```cpp
Tried this example.

#include "BluetoothA2DPSource.h"

BluetoothA2DPSource a2dp_source;

// callback 
int32_t get_sound_data(Frame *data, int32_t frameCount) {
    // generate your sound data 
    // return the effective length (in frames) of the generated sound  (which usually is identical with the requested len)
    // 1 frame is 2 channels * 2 bytes = 4 bytes
    return frameCount;
}

void setup() {
  a2dp_source.start("TECNO SPARK 10", get_sound_data);  
}

void loop() {
}
```
`TECNO SPARK 10` is the name of my phone. Tried setting this in an attempt to get it to work.
Did not work when set to e.g. `MyMusic`.
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.7

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-11-06:
Of cause it does not!
BluetoothA2DPSource will send audio to a bluetooth speaker and your Android 13 is definitly not one

closed 2024-11-06 by pschatzmann

electrovoyage, 2024-11-06:
Is it only possible to use this to transmit to a speaker? I thought this could be used as a bluetooth microphone...

pschatzmann, 2024-11-06:
Please read the Readme! It is quite comprehensive.
If you want to send audio from your Android to the ESP32 you need to use a Sink!

electrovoyage, 2024-11-06:
Yes, i read about the sink. I meant the opposite: sending data from the ESP with a connected microphone module to a phone for sound input, e.g. using the ESP microphone's output to record something in the voice recorder app

pschatzmann, 2024-11-06:
So you are at the wrong place: this is a question for an Android forum.
I guess you need to develop your Bluetooth Sink on Andoid then....