pschatzmann/ESP32-A2DP #628: samples errors: 'I2SClass' does not name a type

josef2600, 2024-11-15:
### Problem Description

in some of the samples you used `I2SClass i2s;` . all of them get `'I2SClass' does not name a type` error. but with `I2SStream `they work. 
and the rest of the codes are deprecated.
check them if you had time.

### Device Description

esp32

### Sketch

```cpp
#include "ESP_I2S.h"
#include "BluetoothA2DPSink.h"

const uint8_t I2S_SCK = 5;       /* Audio data bit clock */
const uint8_t I2S_WS = 25;       /* Audio data left and right clock */
const uint8_t I2S_SDOUT = 26;    /* ESP32 audio data output (to speakers) */
I2SClass i2s;

BluetoothA2DPSink a2dp_sink(i2s);

void setup() {
    i2s.setPins(I2S_SCK, I2S_WS, I2S_SDOUT);
    if (!i2s.begin(I2S_MODE_STD, 44100, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO, I2S_STD_SLOT_BOTH)) {
      Serial.println("Failed to initialize I2S!");
      while (1); // do nothing
    }

    a2dp_sink.start("MyMusic");
}
void loop() {}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Arduino Core 3.0.7

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-11-15:
I2SClass i2s is part of the Arduino I2S API.
If you are using a core < 3.0 you need to adapt the example it to the old Arduio I2S API since it has changed as well! 

This is described in the README

closed 2024-11-15 by pschatzmann

josef2600, 2024-11-15:
> I2SClass i2s is part of the Arduino I2S API. If you are using a core < 3.0 you need to adapt the example it to the old Arduio I2S API since this has changed as well! Have a look at the ESP32 I2S examples provided by Arduino!
> 
> This is described in the README

did you read the post? no! Arduino Core 3.0.7 and it is exactly your sample. i said if you have the time check them. if not, what is the point of reporting?

pschatzmann, 2024-11-15:
I tested in Arduino 3.1rc3 and 3.0.7 and for me it is working w/o issues.

Did you try to compile one of the I2S examples provided by Arduino

josef2600, 2024-11-15:
in Arduino example :
```
void setup() {
  // Create an instance of the I2SClass
  I2SClass i2s;

...
```
`I2SClass i2s;` is inside the setup, not outside.

pschatzmann, 2024-11-15:
Simple_tone.ino compiles w/o issue. What did you try ?

josef2600, 2024-11-15:
no. the problem was from` #include "ESP_I2S.h"` .
when inserting it, all works.

pschatzmann, 2024-11-15:
I don't understand: your example that you pasted above has this include!

If you read the comment is bt_music_receiver_arduino_i2s.ino, it clearly states that it is for  ESP32 I2S < 3.0.0