pschatzmann/ESP32-A2DP #642: bt music sender automatically reconnected in 10 seconds after calling set_connected(false)

flywhc, 2024-12-19:
### Problem Description

set_connected(false) doesn't work as expected because it will reconnect to the device after 10 seconds.
I also tried set_auto_reconnect(false), clean_last_connection(), disconnect() and end() and the results are similar. It will crash if I calls end(true) or delete the object. I need to make sure the BT is completely off so that the ESP32 can switch to the WiFi AP mode.

Steps:

1. load the official example 'bt_music_sender_disconnect' and modify the BT device name to my own Bluetooth speaker 'Tmall Genie BOOM'.
2. Upload the compiled firmware to ESP32
3. My BT speaker prompts 'Bluetooth Device connected' then it plays a low-frequency sound as expected
4. wait for a minute and it stops playing and my BT speaker prompts 'Bluetooth device discounted' - as expected.
5. wait for 10 seconds, the BT speaker prompts 'Bluetooth Device connected' then it plays a low-frequency sound - UNEXPECTED

In the following logs, it disconnected at 18:02:16.028 and reconnected at 18:02:26.586:

18:02:05.953 -> [ 51264][D][BluetoothA2DPSource.cpp:838] bt_app_av_media_proc(): [BT_AV] bt_app_av_media_proc evt 65280
18:02:15.875 -> [ 61181][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
18:02:15.875 -> [ 61210][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
18:02:15.908 -> [ 61221][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0xff00
18:02:15.908 -> [ 61232][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:15.908 -> [ 61242][D][BluetoothA2DPSource.cpp:765] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 65280
18:02:15.947 -> [ 61253][D][BluetoothA2DPSource.cpp:838] bt_app_av_media_proc(): [BT_AV] bt_app_av_media_proc evt 65280
18:02:16.028 -> Connected: 0
**18:02:16.028 -> [ 61349][I][BluetoothA2DPCommon.cpp:67] disconnect(): [BT_AV] disconect a2d: 84:44:af:54:ea:e0**
18:02:16.028 -> [ 61363][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
18:02:16.062 -> [ 61374][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
18:02:16.062 -> [ 61384][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0x0
18:02:16.062 -> [ 61396][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:16.102 -> [ 61405][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnecting
18:02:16.102 -> [ 61419][D][BluetoothA2DPSource.cpp:765] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 0
18:02:16.138 -> [ 61468][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
18:02:16.138 -> [ 61478][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
18:02:16.171 -> [ 61479][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 13
18:02:16.171 -> [ 61497][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0x0
18:02:16.171 -> [ 61497][I][BluetoothA2DPSource.cpp:549] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT mode:0
18:02:16.205 -> [ 61517][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:16.205 -> [ 61527][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnected
18:02:16.028 -> [ 61349][I][BluetoothA2DPCommon.cpp:67] disconnect(): [BT_AV] disconect a2d: 84:44:af:54:ea:e0
18:02:16.028 -> [ 61363][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
18:02:16.062 -> [ 61374][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
18:02:16.062 -> [ 61384][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0x0
18:02:16.062 -> [ 61396][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:16.102 -> [ 61405][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnecting
18:02:16.102 -> [ 61419][D][BluetoothA2DPSource.cpp:765] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 0
18:02:16.138 -> [ 61468][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
18:02:16.138 -> [ 61478][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
18:02:16.171 -> [ 61479][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 13
18:02:16.171 -> [ 61497][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0x0
18:02:16.171 -> [ 61497][I][BluetoothA2DPSource.cpp:549] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT mode:0
18:02:16.205 -> [ 61517][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:16.205 -> [ 61527][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnected
18:02:16.205 -> [ 61541][D][BluetoothA2DPSource.cpp:765] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 0
18:02:16.239 -> [ 61551][I][BluetoothA2DPSource.cpp:773] bt_app_av_state_connected_hdlr(): [BT_AV] a2dp disconnected
18:02:20.236 -> [ 65536][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 17
18:02:20.236 -> [ 65545][I][BluetoothA2DPSource.cpp:555] bt_app_gap_callback(): [BT_AV] event: 17
18:02:25.864 -> [ 71171][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
18:02:25.864 -> [ 71182][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
18:02:25.864 -> [ 71193][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_UNCONNECTED, evt 0xff00
18:02:25.864 -> [ 71204][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:25.902 -> [ 71214][D][BluetoothA2DPSource.cpp:693] bt_app_av_state_unconnected_hdlr(): [BT_AV] bt_app_av_state_unconnected_hdlr evt 65280
18:02:25.902 -> [ 71225][I][BluetoothA2DPSource.h:291] esp_a2d_connect(): [BT_AV] a2dp connecting to: 84:44:af:54:ea:e0
18:02:25.902 -> [ 71237][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
18:02:25.934 -> [ 71248][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
18:02:25.934 -> [ 71258][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTING, evt 0x0
18:02:25.934 -> [ 71269][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
18:02:25.980 -> [ 71279][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connecting
18:02:25.980 -> [ 71292][D][BluetoothA2DPSource.cpp:722] bt_app_av_state_connecting_hdlr(): [BT_AV] bt_app_av_state_connecting_hdlr evt 0
18:02:26.060 -> [ 71381][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 16
18:02:26.060 -> [ 71391][I][BluetoothA2DPSource.cpp:545] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT
18:02:26.553 -> [ 71866][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
[... 77 lines cut ...]
}

void setup() {
  Serial.begin(115200);
  //a2dp_source.set_auto_reconnect(false);
  a2dp_source.start("LEXON MINO L", get_data_channels);  
  a2dp_source.set_volume(20);
}

void loop() {
  delay(60000); // change state every minute
  state = !state;
  Serial.print("Connected: ");
  Serial.println(state);
  a2dp_source.set_connected(state);
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.7

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

flywhc, 2024-12-19:
> Your log is very confusing and does not match with the latest version from the library. Please upgrade first and test again.
> 
> Please also note that the test case is switching between active and inactive!

I just pulled the latest version v1.8.5 and ensured Arduino 2.3.4 has all libraries and SDKs upgraded and then tested again. The issue is still reproduced.

` delay(60000); // change state every minute`

According to the source code, it should switch connected/disconnected every minute but it is actually switched off only for 10 seconds without any function call. 
only can see "Connected: 0" before disconnect but no "Connected: 1" before connecting in the log.

21:22:03.458 -> Connected: 0
21:22:03.458 -> [ 61300][I][BluetoothA2DPCommon.cpp:67] disconnect(): [BT_AV] disconect a2d: 84:44:af:54:ea:e0
...
21:22:13.314 -> [ 71155][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
21:22:13.346 -> [ 71165][D][BluetoothA2DPSource.cpp:693] bt_app_av_state_unconnected_hdlr(): [BT_AV] bt_app_av_state_unconnected_hdlr evt 65280
21:22:13.346 -> [ 71176][I][BluetoothA2DPSource.h:294] esp_a2d_connect(): [BT_AV] a2dp connecting to: 84:44:af:54:ea:e0

Full log:
```
21:21:03.227 -> [  1047][D][BluetoothA2DPSource.cpp:326] bt_app_task_start_up(): [BT_AV] bt_app_task_start_up
21:21:03.227 -> [  1056][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 0
21:21:03.227 -> [  1066][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
21:21:03.261 -> [  1076][D][BluetoothA2DPSource.cpp:563] bt_av_hdl_stack_evt(): [BT_AV] bt_av_hdl_stack_evt event: 0
21:21:03.261 -> [  1086][I][BluetoothA2DPCommon.cpp:368] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable false
21:21:03.261 -> [  1090][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x4, param len 16
21:21:03.261 -> [  1106][I][BluetoothA2DPSource.cpp:600] bt_av_hdl_stack_evt(): [BT_AV] Starting device discovery...
21:21:03.294 -> [  1106][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
21:21:03.294 -> [  1123][I][BluetoothA2DPSource.cpp:555] bt_app_gap_callback(): [BT_AV] event: 10
21:21:03.294 -> [  1123][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x4
21:21:03.294 -> [  1141][I][Blueto[o t h1A124D1P]S[oDu]r[cBel.uceptpo:o6t6h5A]2 DbPtS_oauprpc_ea.vc_pspm:_4h6d1l]r (b)t:_ a[pBpT__gAaVp]_ cbatl_lapp_av:sm_hdlr sta_ae APP_AV_lTATE_DISCOV0
ING, evt 0x4
21:21:03.327 -> [  1157][I][B[l u e1t1o5o7t]h[AD2]D[PBSlouuertcoeo.tchpAp2:D5P5S5o]u rbcte_.acpppp_:g6a3p2_]c aplrlobcaecsks(_)u:s e[rB_Ts_tAaVt] event:b10ks(): [BT_AV] process_user_state_callbacks
21:21:03.327 -> [  1173][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
21:21:03.360 -> [  1173][I][BluetoothA2DPCommon.h:166] set_volume(): [BT_AV] set_volume: 20
21:21:03.360 -> [  1188][I][BluetoothA2DPSource.cpp:555] bt_app_gap_callback(): [BT_AV] event: 10
21:21:03.360 -> =========== After Setup Start ============
21:21:03.360 -> INT[E R N1A1L9 9M]e[mDo]r[yB lIuneftoo:o
t
21:21:03.360 -> hA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 10
21:21:03.395 -> ------------------------------------------
21:21:03.395 -> [  1210][I][BluetoothA2DPSource.cpp:555] bt_app_gap_callback(): [BT_AV] event: 10
21:21:03.395 ->   Total Size        :   275540 B ( 269.1 KB)
21:21:03.395 ->   Free Bytes        :   148148 B ( 144.7 KB)
21:21:03.395 ->   Allocated Bytes   :   116680 B ( 113.9 KB)
21:21:03.395 -> [  1230][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 1
21:21:03.395 ->   Minimum Free Bytes:   145088 B ( 141.7 KB)
21:21:03.429 -> [  1242][I][BluetoothA2DPSource.cpp:484] bt_app_gap_callback(): [BT_AV] Discovery started.
21:21:03.429 ->   Largest Free Block:   110580 B ( 108.0 KB)
21:21:03.429 -> ------------------------------------------
21:21:03.429 -> GPIO Info:
21:21:03.429 -> ------------------------------------------
21:21:03.429 ->   GPIO : BUS_TYPE[bus/unit][chan]
21:21:03.429 ->   --------------------------------------  
21:21:03.429 ->      1 : UART_TX[0]
21:21:03.429 ->      3 : UART_RX[0]
21:21:03.467 -> ============ After Setup End =============
21:21:04.036 -> [  1878][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
21:21:04.069 -> [  1887][I][BluetoothA2DPSource.cpp:394] filter_inquiry_scan_result(): [BT_AV] Scanned device: 84:44:af:54:ea:e0
21:21:04.069 -> [  1897][I][BluetoothA2DPSource.cpp:400] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240404
21:21:04.069 -> [  1906][I][BluetoothA2DPSource.cpp:404] filter_inquiry_scan_result(): [BT_AV] --RSSI: -34
21:21:04.069 -> [  1914][I][BluetoothA2DPSource.cpp:424] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
21:21:04.102 -> [  1923][I][BluetoothA2DPSource.cpp:426] filter_inquiry_scan_result(): [BT_AV] --Name: Tmall Genie BOOM(EA:E0)
21:21:04.102 -> [  1933][D][BluetoothA2DPSource.cpp:438] filter_inquiry_scan_result(): [BT_AV] --Checking match: Tmall Genie BOOM(EA:E0)
21:21:04.102 -> [  1944][I][BluetoothA2DPSource.cpp:447] filter_inquiry_scan_result(): [BT_AV] --Result: Target device found
[... cut ...]
21:22:33.365 -> [ 91185][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0xff00
21:22:33.365 -> [ 91196][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
21:22:33.365 -> [ 91206][D][BluetoothA2DPSource.cpp:765] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 65280
21:22:33.410 -> [ 91216][D][BluetoothA2DPSource.cpp:838] bt_app_av_media_proc(): [BT_AV] bt_app_av_media_proc evt 65280

```

pschatzmann, 2024-12-19:
Thanks for the update. I committed a correction. 
Can you help me to test this ?

flywhc, 2024-12-19:
> Thanks for the update. I committed a correction. Can you help me to test this ?

Thank you for the fix. The original problem is fixed.

However, after it reconnected and the BT speaker showed 'Bluetooth connected', no low-frequency sound was played mostly.
I tested a few times, each time I powered off ESP32 and power on it.
* Test 1: no sound after all reconnections
* Test 2: playing sound after reconnection at the minute 2 and then no sound after the 2nd reconnection at  the min 4, no sound after the 3nd reconnection at the 6.

Attached log is the 2nd "Connected: 0" and "Connected: 1".

Additionally You can see many lines of Bluetooth related logs after disconnection such as bt_app_work_dispatch() is called every 10 seconds. Will they impact WiFi connection? If the program deletes BluetoothA2DPSource instance, will it crash?

```
00:04:50.677 -> Connected: 0
00:04:50.677 -> [181334][I][BluetoothA2DPCommon.cpp:68] disconnect(): [BT_AV] disconect a2d: 84:44:af:54:ea:e0
00:04:50.712 -> [181357][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
00:04:50.712 -> [181368][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
00:04:50.712 -> [181378][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0x0
00:04:50.712 -> [181389][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
00:04:50.754 -> [181399][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnecting
00:04:50.754 -> [181412][D][BluetoothA2DPSource.cpp:768] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 0
00:04:50.791 -> [181463][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 16
00:04:50.791 -> [181474][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0x0
00:04:50.825 -> [181474][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 13
00:04:50.825 -> [181493][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTED, evt 0x0
00:04:50.825 -> [181493][I][BluetoothA2DPSource.cpp:549] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT mode:0
00:04:50.857 -> [181513][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
00:04:50.857 -> [181523][D][BluetoothA2DPSource.cpp:640] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnected
00:04:50.857 -> [181536][D][BluetoothA2DPSource.cpp:768] bt_app_av_state_connected_hdlr(): [BT_AV] bt_app_av_state_connected_hdlr evt 0
00:04:50.891 -> [181547][I][BluetoothA2DPSource.cpp:776] bt_app_av_state_connected_hdlr(): [BT_AV] a2dp disconnected
00:04:59.109 -> [189755][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 17
00:04:59.109 -> [189763][I][BluetoothA2DPSource.cpp:555] bt_app_gap_callback(): [BT_AV] event: 17
00:05:00.441 -> [191119][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
00:05:00.474 -> [191130][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
00:05:00.474 -> [191141][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_UNCONNECTED, evt 0xff00
00:05:00.474 -> [191152][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
00:05:00.517 -> [191162][D][BluetoothA2DPSource.cpp:693] bt_app_av_state_unconnected_hdlr(): [BT_AV] bt_app_av_state_unconnected_hdlr evt 65280
00:05:10.474 -> [201119][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
00:05:10.474 -> [201130][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
00:05:10.474 -> [201141][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_UNCONNECTED, evt 0xff00
00:05:10.474 -> [201152][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
00:05:10.517 -> [201162][D][BluetoothA2DPSource.cpp:693] bt_app_av_state_unconnected_hdlr(): [BT_AV] bt_app_av_state_unconnected_hdlr evt 65280
00:05:20.469 -> [211119][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
00:05:20.469 -> [211130][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
00:05:20.469 -> [211141][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_UNCONNECTED, evt 0xff00
00:05:20.469 -> [211152][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
00:05:20.525 -> [211162][D][BluetoothA2DPSource.cpp:693] bt_app_av_state_unconnected_hdlr(): [BT_AV] bt_app_av_state_unconnected_hdlr evt 65280
00:05:30.441 -> [221119][D][BluetoothA2DPSource.cpp:256] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0xff00, param len 0
[... cut ...]
00:06:50.464 -> [301130][D][BluetoothA2DPSource.cpp:306] bt_app_task_handler(): [BT_API] bt_app_task_handler, signal: 0x1, event: 0xff00
00:06:50.464 -> [301141][I][BluetoothA2DPSource.cpp:665] bt_app_av_sm_hdlr(): [BT_AV] bt_app_av_sm_hdlr state APP_AV_STATE_CONNECTING, evt 0xff00
00:06:50.504 -> [301152][D][BluetoothA2DPSource.cpp:632] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks
00:06:50.504 -> [301162][D][BluetoothA2DPSource.cpp:725] bt_app_av_state_connecting_hdlr(): [BT_AV] bt_app_av_state_connecting_hdlr evt 65280
00:06:50.721 -> Connected: 0

```

pschatzmann, 2024-12-31:
aargh, that was quite difficult to resolve!

I committed a [correction](https://github.com/pschatzmann/ESP32-A2DP/commit/84bee9cfa24c5f5f618719dd1ce0576d3e8a2308)!

closed 2024-12-31 by pschatzmann