pschatzmann/ESP32-A2DP #648: Simple no I2S example crashes as soon as music is sent

martinbrennan, 2025-01-07:
### Problem Description

I created a simple a2dp sink sketch using a callback. No Audiotools, no I2s and no DAC - just an empty callback. It runs on the Espressif ESP32-LyraT (I selected Adafruit ESP32 Feather as the board because Lyra is not available). I can connect to it from my Android phone but the program crashes shortly after music is sent.

Here is the stack dump - which appears to mention I2S code which doesn't seem right

ESP Exception Decoder
Sketch: sketch_jan7a FQBN: esp32:esp32:featheresp32

Backtrace: 0x40195a4d:0x3ffd0320 0x40195fad:0x3ffd0340 0x400d2d89:0x3ffd0360 0x40195da5:0x3ffd0380 0x400d22c3:0x3ffd03a0 0x40116c71:0x3ffd03c0 0x4010d62d:0x3ffd0400 0x4010d5c9:0x3ffd0420 0x40095f8a:0x3ffd0450

Decoding stack results
0x40195a4d: BluetoothA2DPOutputAudioTools::set_output(Print&) at /home/martin/Arduino/libraries/ESP32-A2DP-main/src/BluetoothA2DPOutput.h:105
0x40195fad:  is in BluetoothA2DPSink::i2s_write_data(unsigned char const*, unsigned int) (/home/martin/Arduino/libraries/ESP32-A2DP-main/src/BluetoothA2DPSink.cpp:1251).
0x400d2d89:  is in BluetoothA2DPSink::write_audio(unsigned char const*, unsigned int) (/home/martin/Arduino/libraries/ESP32-A2DP-main/src/BluetoothA2DPSink.h:536).
0x40195da5:  is in BluetoothA2DPSink::audio_data_callback(unsigned char const*, unsigned long) (/home/martin/Arduino/libraries/ESP32-A2DP-main/src/BluetoothA2DPSink.cpp:1050).
0x400d22c3:  is in ccall_audio_data_callback(uint8_t const*, uint32_t) (/home/martin/Arduino/libraries/ESP32-A2DP-main/src/BluetoothA2DPSink.cpp:1192).
0x40095f8a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

### Device Description

Espressif ESP32LyraT

### Sketch

```cpp
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

int count = 0;

// Then somewhere in your sketch:
void data_received_callback() {
//  printf ("Data packet received\n");
  count++;
}

void setup() {
  printf ("starting\n");
  a2dp_sink.set_on_data_received(data_received_callback);
//  a2dp_sink.set_stream_reader(read_data_stream);
  a2dp_sink.start("ArduinoBluetooth");
  printf ("initialised\n");
}

void loop() {
  sleep (1);
  printf ("Count %d\n",count);
}

void read_data_stream(const uint8_t *data, uint32_t length)
{
  int16_t *samples = (int16_t*) data;
  uint32_t sample_count = length/2;
  printf ("read_data_stream\n");
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Arduino 2.3.3

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-01-07:
For this case I suggest that you deactivate the output in the config.h
However I would recommend to use [this sketch](https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_audiokit/bt_music_receiver_audiokit.ino). Just replace the AudioKitEs8388V1 driver with a LyratV42 or LyratV43.

For these boards, I alway select the ESP32 Dev Module as Device in Arduino.

closed 2025-01-07 by pschatzmann

martinbrennan, 2025-01-08:
Hello Phil
Thank you for the very quick reply - unfortunately
1) The link for this sketch returns a 404 error
2) I have pasted config.h below and cannot see anything to deactivate - everything is commented
Martin

#pragma once

#ifndef AUTOCONNECT_TRY_NUM
#  define AUTOCONNECT_TRY_NUM 1000
#endif

// Activate I2S Support (legacy i2s)
#ifndef A2DP_LEGACY_I2S_SUPPORT
#  define A2DP_LEGACY_I2S_SUPPORT (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 0, 0))
#endif

// Use https://pschatzmann.github.io/arduino-audio-tools for output
#ifndef A2DP_I2S_AUDIOTOOLS
#  if __has_include("AudioTools.h")
#    define A2DP_I2S_AUDIOTOOLS 1
#  endif
#endif

#ifndef A2DP_SPP_SUPPORT
#  define A2DP_SPP_SUPPORT (ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 0, 0))
#endif

// Maximum write size
#ifndef A2DP_I2S_MAX_WRITE_SIZE 
#  define A2DP_I2S_MAX_WRITE_SIZE 1024 * 5
#endif

// Maximum wait time for status change in 100 ms when calling end()
#ifndef A2DP_DISCONNECT_LIMIT 
#  define A2DP_DISCONNECT_LIMIT 20
#endif

pschatzmann, 2025-01-08:
I renamed the sketch from audiokit to audioboard to be a bit more generic. The link above has been correcteed.
If you paste code please use the proper [code formatting
](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks)

martinbrennan, 2025-01-08:
Hello Phil
Thanks again for the quick reply - I now have a working example.
In fact I used [this sketch](https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_datacallback/bt_music_receiver_datacallback.ino) of yours.
No modification to config.h was required
Thanks for the triple back tick recommendation
Martin