pschatzmann/ESP32-A2DP #650: bt music sender frequently scan BT devices after calling set_connected(false)

flywhc, 2025-01-11:
### Problem Description

* Load the example 'bt_music_sender_disconnect.
* For the sake of saving time, I changed delay(60000) to delay(10000).
* I don't have the device named "LEXON MINO L" so please change it to anything else if you do have this device.
* Turn on the Core Debug level to Debug level.

Then we can see two issues after the device starts:

Issue 1: It shows massive logs repeatly reporting 'Target device not found'  because it cannot find the BT device - I feel the device is less responsive. Should it have a delay() after 'Result: Target device not found'?

Issue 2: after calling a2dp_source.set_connected(false), the BT stack should be quiet but it still keep scanning. This is more serious issue because WIFI won't work even I tried disconnecting BT.

ps. I did some study on co-existing of BT and WiFi. The result is it is possible to make BT and WiFi work together in a rotatory way. Neither BT nor WiFi may have a heavy data load in the co-existing mode, so WiFi is unlikely to work while A2DP is in operation. 
So my idea is to completely shut BT down and allow my phone to connect to WiFi AP, manually change settings then back to BT A2DP mode. In this way, I don't have to install a screen on my mini BT transmitter.

### Device Description

ESP32 WROOM 32E with programmer

### Sketch

```cpp
#include "BluetoothA2DPSource.h"
#define c3_frequency  130.81

BluetoothA2DPSource a2dp_source;
bool state = true;

// The supported audio codec in ESP32 A2DP is SBC. SBC audio stream is encoded
// from PCM data normally formatted as 44.1kHz sampling rate, two-channel 16-bit sample data
int32_t get_data_channels(Frame *frame, int32_t channel_len) {
    static double m_time = 0.0;
    double m_amplitude = 10000.0;  // -32,768 to 32,767
    double m_deltaTime = 1.0 / 44100.0;
    double m_phase = 0.0;
    double double_Pi = PI * 2.0;
    // fill the channel data
    for (int sample = 0; sample < channel_len; ++sample) {
        double angle = double_Pi * c3_frequency * m_time + m_phase;
        frame[sample].channel1 = m_amplitude * sin(angle);
        frame[sample].channel2 = frame[sample].channel1;
        m_time += m_deltaTime;
    }

    return channel_len;
}

void setup() {
  Serial.begin(115200);
  //a2dp_source.set_auto_reconnect(false);
  a2dp_source.start("LEXON MINO L", get_data_channels);  
  a2dp_source.set_volume(20);
}

void loop() {
  delay(10000); // change state every minute
  state = !state;
  Serial.print("Connected: ");
  Serial.println(state);
  a2dp_source.set_connected(state);
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.0.7

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

flywhc, 2025-01-11:
Logs:

17:41:23.779 -> [ 11952][I][BluetoothA2DPSource.cpp:454] filter_inquiry_scan_result(): [BT_AV] --Result: Target device not found
17:41:23.813 -> [ 11993][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
17:41:23.813 -> [ 12002][I][BluetoothA2DPSource.cpp:394] filter_inquiry_scan_result(): [BT_AV] Scanned device: 84:44:af:54:ea:e0
17:41:23.813 -> [ 12012][I][BluetoothA2DPSource.cpp:400] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240404
17:41:23.813 -> [ 12021][I][BluetoothA2DPSource.cpp:404] filter_inquiry_scan_result(): [BT_AV] --RSSI: -51
17:41:23.847 -> [ 12029][I][BluetoothA2DPSource.cpp:424] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
17:41:23.847 -> [ 12038][I][BluetoothA2DPSource.cpp:426] filter_inquiry_scan_result(): [BT_AV] --Name: Tmall Genie BOOM(EA:E0)
17:41:23.847 -> [ 12048][D][BluetoothA2DPSource.cpp:438] filter_inquiry_scan_result(): [BT_AV] --Checking match: LEXON MINO L
17:41:23.847 -> [ 12058][I][BluetoothA2DPSource.cpp:454] filter_inquiry_scan_result(): [BT_AV] --Result: Target device not found
17:41:24.008 -> [ 12193][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
17:41:24.008 -> [ 12202][I][BluetoothA2DPSource.cpp:394] filter_inquiry_scan_result(): [BT_AV] Scanned device: 84:44:af:54:ea:e0
17:41:24.008 -> [ 12212][I][BluetoothA2DPSource.cpp:400] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240404
17:41:24.041 -> [ 12221][I][BluetoothA2DPSource.cpp:404] filter_inquiry_scan_result(): [BT_AV] --RSSI: -50
17:41:24.041 -> [ 12229][I][BluetoothA2DPSource.cpp:424] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
17:41:24.041 -> [ 12238][I][BluetoothA2DPSource.cpp:426] filter_inquiry_scan_result(): [BT_AV] --Name: Tmall Genie BOOM(EA:E0)
17:41:24.041 -> [ 12248][D][BluetoothA2DPSource.cpp:438] filter_inquiry_scan_result(): [BT_AV] --Checking match: LEXON MINO L
17:41:24.084 -> [ 12258][I][BluetoothA2DPSource.cpp:454] filter_inquiry_scan_result(): [BT_AV] --Result: Target device not found
17:41:24.125 -> [ 12333][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
17:41:24.159 -> [ 12342][I][BluetoothA2DPSource.cpp:394] filter_inquiry_scan_result(): [BT_AV] Scanned device: 84:44:af:54:ea:e0
17:41:24.159 -> [ 12352][I][BluetoothA2DPSource.cpp:400] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240404
17:41:24.159 -> [ 12361][I][BluetoothA2DPSource.cpp:404] filter_inquiry_scan_result(): [BT_AV] --RSSI: -51
17:41:24.159 -> [ 12369][I][BluetoothA2DPSource.cpp:424] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
17:41:24.191 -> [ 12378][I][BluetoothA2DPSource.cpp:426] filter_inquiry_scan_result(): [BT_AV] --Name: Tmall Genie BOOM(EA:E0)
17:41:24.191 -> [ 12388][D][BluetoothA2DPSource.cpp:438] filter_inquiry_scan_result(): [BT_AV] --Checking match: LEXON MINO L
17:41:24.191 -> [ 12398][I][BluetoothA2DPSource.cpp:454] filter_inquiry_scan_result(): [BT_AV] --Result: Target device not found
17:41:24.228 -> [ 12413][D][BluetoothA2DPSource.cpp:461] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 0
17:41:24.228 -> [ 12422][I][BluetoothA2DPSource.cpp:394] filter_inquiry_scan_result(): [BT_AV] Scanned device: 84:44:af:54:ea:e0
17:41:24.228 -> [ 12432][I][BluetoothA2DPSource.cpp:400] filter_inquiry_scan_result(): [BT_AV] --Class of Device: 0x240404
17:41:24.261 -> [ 12441][I][BluetoothA2DPSource.cpp:404] filter_inquiry_scan_result(): [BT_AV] --RSSI: -51
17:41:24.261 -> [ 12449][I][BluetoothA2DPSource.cpp:424] filter_inquiry_scan_result(): [BT_AV] --Compatiblity: Compatible
17:41:24.261 -> [ 12458][I][BluetoothA2DPSource.cpp:426] filter_inquiry_scan_result(): [BT_AV] --Name: Tmall Genie BOOM(EA:E0)
17:41:24.261 -> [ 12468][D][BluetoothA2DPSource.cpp:438] filter_inquiry_scan_result(): [BT_AV] --Checking match: LEXON MINO L
17:41:24.294 -> [ 12478][I][BluetoothA2DPSource.cpp:454] filter_inquiry_scan_result(): [BT_AV] --Result: Target device not found

pschatzmann, 2025-01-17:
ad 1: feel free to implement your own logic

ad 2) I can't reproduce your issue:  after beeing disconnected the task handler is just looping in disconnected state: so this works as designed. I you want to end all processing call end() as documented

flywhc, 2025-01-19:
@pschatzmann end() doesn't stop the bluetooth related code running.  I guess I found the way to fix this issue. Would you please confirm:

s_tmr is created but it is not stopped/deleted so that the code is called every 10 seconds even after calling end(true) - this causes my problem.
esp_avrc_ct_init() is called but esp_avrc_ct_deinit() is not called in end() - this doesn't cause my problem but it is potential leak.

So my fix is adding a overrided end() in BluetoothA2DPSource.cpp:

```
void BluetoothA2DPSource::end(bool release_memory) {
  esp_avrc_ct_deinit();
  if(s_tmr != NULL) {
    xTimerDelete(s_tmr, portMAX_DELAY);
    s_tmr = NULL;
  }
  BluetoothA2DPCommon::end(release_memory);
}

```

and add a line in the BluetoothA2DPSource.h:

`virtual void end(bool releaseMemory=false);`

pschatzmann, 2025-01-20:
I moved the functionality that is available both in the source and sink to common to [address this issue](https://github.com/pschatzmann/ESP32-A2DP/pull/657).

closed 2025-01-20 by pschatzmann