pschatzmann/ESP32-A2DP #651: Arduino IDE 2.3.4 - error: 'I2SStream' does not name a type

TerelleM, 2025-01-13:
### Problem Description

Compiler does not recognize "I2SStream" on line 27. Do I need to install another library or is the new Arduino IDE 2.0 not compatible?

Error:
C:\Users\terel\AppData\Local\Temp\.arduinoIDE-unsaved2025013-15096-mf52s5.nhad\bt_music_receiver_and_BLE\bt_music_receiver_and_BLE.ino:27:1: error: 'I2SStream' does not name a type; did you mean 'Stream'?
   27 | I2SStream i2s;
      | ^~~~~~~~~
      | Stream
C:\Users\terel\AppData\Local\Temp\.arduinoIDE-unsaved2025013-15096-mf52s5.nhad\bt_music_receiver_and_BLE\bt_music_receiver_and_BLE.ino:28:29: error: 'i2s' was not declared in this scope
   28 | BluetoothA2DPSink a2dp_sink(i2s);
      |                             ^~~

exit status 1

Compilation error: 'I2SStream' does not name a type; did you mean 'Stream'?

### Device Description

ESP32 DEVKIT V1

### Sketch

```cpp
/*
  Streaming Music from Bluetooth
  
  Copyright (C) 2020 Phil Schatzmann
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

// ==> Example which shows how to cooperate with BLE

#include "BluetoothA2DPSink.h"
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>

#define SERVICE_UUID "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

I2SStream i2s;
BluetoothA2DPSink a2dp_sink(i2s);
BLECharacteristic *pCharacteristic;
char title[160] = {"Undefined"};

void avrc_metadata_callback(uint8_t id, const uint8_t *text) {
  Serial.printf("==> AVRC metadata rsp: attribute id 0x%x, %s\n", id, text);
  if (id == ESP_AVRC_MD_ATTR_TITLE) {
    strncpy(title, (const char *)text, 160);
    pCharacteristic->setValue(title);
  }
}

void setup() {
  // start a2dp in ESP_BT_MODE_BTDM mode
  a2dp_sink.set_default_bt_mode(ESP_BT_MODE_BTDM);
  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
  a2dp_sink.start("MyMusic");
  Serial.println("A2DP Started!");

  // start BLE
  BLEDevice::init("MyMusic");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
      CHARACTERISTIC_UUID, BLECharacteristic::PROPERTY_READ);

  pCharacteristic->setValue(title);
  pService->start();
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(
      0x06); // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  delay(1000); // do nothing
}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

Arduino IDE 2.3.4

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-01-13:
Just read the README !

closed 2025-01-13 by pschatzmann

TerelleM, 2025-01-13:
> Just read the README !

Hi, I did an you mention "Espressif is retiring the legacy I2S API: So with Arduino v3.0.0 (IDF v5) [my old I2S integration](https://github.com/pschatzmann/ESP32-A2DP/wiki/Legacy-I2S-API) will not be available any more. The legacy syntax is still working as long as you don't upgrade."

I've installed the arduino-audio-tools library and still no bueno.

pschatzmann, 2025-01-13:
Maybe you need to restart the IDE ?
If this does not help double check the libraries directory if you installed it correctly!

TerelleM, 2025-01-13:
> Maybe you need to restart the IDE ? If this does not help double check the libraries directory if you installed it correctly!

reinstalling the esp32 board managers seemed to have fixed it. Thanks for the fast reply! Great work on the documentation its much appreciated coming from an engineering background.