pschatzmann/ESP32-A2DP #661: Bluetooth will not be connectable when using ESP32-A2DP with ESP32-HUB75-MatrixPanel-DMA

makerawei, 2025-01-24:
### Problem Description

When I use ESP32-A2DP with [ESP32-HUB75-MatrixPanel-DMA](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA).  
I can search the ESP32 bluetooth device, but can not connect to it.

### Device Description

ESP32

### Sketch

```cpp
auto cfg = i2s.defaultConfig();
  cfg.pin_bck = 2;  //BCLK
  cfg.pin_ws = 33;   //LRC
  cfg.pin_data = 32; //DIN/DOUT
  i2s.begin(cfg);
  a2dp_sink.set_i2s_port(I2S_NUM_1);
  a2dp_sink.set_on_connection_state_changed(connection_state_changed);
  a2dp_sink.set_on_audio_state_changed(audio_state_changed);
  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
  a2dp_sink.set_stream_reader(read_data_stream);
  a2dp_sink.start("ESP32-44");
  Serial.println("waiting for connection...");
  
  //*
  #if SWAP_BLUE_GREEN
  mxconfig.gpio.b1 = 26;
  mxconfig.gpio.b2 = 12;
  mxconfig.gpio.g1 = 27;
  mxconfig.gpio.g2 = 13;
#endif
  mxconfig.gpio.e = 18;
  mxconfig.clkphase = false;
  display = new MatrixPanel_I2S_DMA(mxconfig);
  display->begin();
  display->setBrightness8(90); // 0-255
  display->setRotation(2);     // 0,1,2,3
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

4.5.1

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-01-24:
Use the set_valid_cod_service() method to define the cod of your device!

closed 2025-01-24 by pschatzmann

pschatzmann, 2025-01-24:
Just set the log level to info: This will report the Class of Device.
The you can allow it with the method call metioned above.

This is necessary if it is not one of a 
- 
- ESP_BT_COD_SRVC_RENDERING,
- ESP_BT_COD_SRVC_AUDIO,
- ESP_BT_COD_SRVC_TELEPHONY

makerawei, 2025-01-24:
#include "BluetoothA2DPSource.h"  
BluetoothA2DPSource a2dp_source;  
a2dp_source.set_valid_cod_service(ESP_BT_COD_SRVC_AUDIO);  

----  
Is the above right ?

pschatzmann, 2025-01-24:
No: audio is already allowed: you need to use the non standard value of your device !

makerawei, 2025-01-24:
Yes  
But I dont know what the `YourValue` should be.

pschatzmann, 2025-01-24:
Me neither: you need to activate the log!

makerawei, 2025-01-24:
Oh, got it !  
I'll try later when I known how to activate the log.  
could you tell me? ~.~

makerawei, 2025-01-24:
here is my log, please help me ~~
```
21:31:09.692 -> [  1432][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:429] begin(): [begin()] Using GPIO 23 for A_PIN
21:31:09.692 -> [  1440][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:430] begin(): [begin()] Using GPIO 19 for B_PIN
21:31:09.724 -> [  1448][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:431] begin(): [begin()] Using GPIO 5 for C_PIN
21:31:09.724 -> [  1456][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:432] begin(): [begin()] Using GPIO 17 for D_PIN
21:31:09.724 -> [  1465][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:433] begin(): [begin()] Using GPIO 18 for E_PIN
21:31:09.724 -> [  1473][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:434] begin(): [begin()] Using GPIO 4 for LAT_PIN
21:31:09.756 -> [  1481][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:435] begin(): [begin()] Using GPIO 15 for OE_PIN
21:31:09.756 -> [  1490][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:436] begin(): [begin()] Using GPIO 16 for CLK_PIN
21:31:09.756 -> [  1498][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:38] allocateDMAmemory(): [I2S-DMA] Free heap: 48628
21:31:09.756 -> [  1507][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:39] allocateDMAmemory(): [I2S-DMA] Free SPIRAM: 0
21:31:09.787 -> [  1516][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:43] allocateDMAmemory(): [I2S-DMA] allocating rowBitStructs with pixel_color_depth_bits of 8
21:31:09.787 -> [  1529][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:58] allocateDMAmemory(): [I2S-DMA] CRITICAL ERROR: Not enough memory for requested colour depth! Please reduce pixel_color_depth_bits value.
21:31:09.820 -> 
21:31:09.820 -> [  1546][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:59] allocateDMAmemory(): [I2S-DMA] Could not allocate rowBitStruct 17!.
21:31:09.820 -> 
21:31:09.820 -> [  1557][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:629] setBrightness(): [setBrightness()] Tried to set output brightness before begin()
21:31:09.820 -> =========== After Setup Start ============
21:31:09.820 -> INTERNAL Memory Info:
21:31:09.852 -> ------------------------------------------
21:31:09.852 ->   Total Size        :   243444 B ( 237.7 KB)
21:31:09.852 ->   Free Bytes        :    12572 B (  12.3 KB)
21:31:09.852 ->   Allocated Bytes   :   219848 B ( 214.7 KB)
21:31:09.852 ->   Minimum Free Bytes:    12276 B (  12.0 KB)
21:31:09.852 ->   Largest Free Block:    10228 B (  10.0 KB)
21:31:09.852 -> ------------------------------------------
21:31:09.852 -> GPIO Info:
21:31:09.852 -> ------------------------------------------
21:31:09.885 ->   GPIO : BUS_TYPE[bus/unit][chan]
21:31:09.885 ->   --------------------------------------  
21:31:09.885 ->      1 : UART_TX[0]
21:31:09.885 ->      3 : UART_RX[0]
21:31:09.885 -> ============ After Setup End =============
```

makerawei, 2025-01-24:
```
21:31:09.692 -> [  1432][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:429] begin(): [begin()] Using GPIO 23 for A_PIN
21:31:09.692 -> [  1440][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:430] begin(): [begin()] Using GPIO 19 for B_PIN
21:31:09.724 -> [  1448][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:431] begin(): [begin()] Using GPIO 5 for C_PIN
21:31:09.724 -> [  1456][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:432] begin(): [begin()] Using GPIO 17 for D_PIN
21:31:09.724 -> [  1465][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:433] begin(): [begin()] Using GPIO 18 for E_PIN
21:31:09.724 -> [  1473][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:434] begin(): [begin()] Using GPIO 4 for LAT_PIN
21:31:09.756 -> [  1481][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:435] begin(): [begin()] Using GPIO 15 for OE_PIN
21:31:09.756 -> [  1490][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:436] begin(): [begin()] Using GPIO 16 for CLK_PIN
21:31:09.756 -> [  1498][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:38] allocateDMAmemory(): [I2S-DMA] Free heap: 48628
21:31:09.756 -> [  1507][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:39] allocateDMAmemory(): [I2S-DMA] Free SPIRAM: 0
21:31:09.787 -> [  1516][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:43] allocateDMAmemory(): [I2S-DMA] allocating rowBitStructs with pixel_color_depth_bits of 8
21:31:09.787 -> [  1529][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:58] allocateDMAmemory(): [I2S-DMA] CRITICAL ERROR: Not enough memory for requested colour depth! Please reduce pixel_color_depth_bits value.
21:31:09.820 -> 
21:31:09.820 -> [  1546][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:59] allocateDMAmemory(): [I2S-DMA] Could not allocate rowBitStruct 17!.
21:31:09.820 -> 
21:31:09.820 -> [  1557][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:629] setBrightness(): [setBrightness()] Tried to set output brightness before begin()
21:31:09.820 -> =========== After Setup Start ============
21:31:09.820 -> INTERNAL Memory Info:
21:31:09.852 -> ------------------------------------------
21:31:09.852 ->   Total Size        :   243444 B ( 237.7 KB)
21:31:09.852 ->   Free Bytes        :    12572 B (  12.3 KB)
21:31:09.852 ->   Allocated Bytes   :   219848 B ( 214.7 KB)
21:31:09.852 ->   Minimum Free Bytes:    12276 B (  12.0 KB)
21:31:09.852 ->   Largest Free Block:    10228 B (  10.0 KB)
21:31:09.852 -> ------------------------------------------
21:31:09.852 -> GPIO Info:
21:31:09.852 -> ------------------------------------------
21:31:09.885 ->   GPIO : BUS_TYPE[bus/unit][chan]
21:31:09.885 ->   --------------------------------------  
21:31:09.885 ->      1 : UART_TX[0]
21:31:09.885 ->      3 : UART_RX[0]
21:31:09.885 -> ============ After Setup End =============
```

makerawei, 2025-01-24:
21:31:08.268 -> [    16][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
21:31:08.298 -> =========== Before Setup Start ===========
21:31:08.298 -> Chip Info:
21:31:08.298 -> ------------------------------------------
21:31:08.298 ->   Model             : ESP32
21:31:08.330 ->   Package           : D0WD-Q5
21:31:08.330 ->   Revision          : 3.00
21:31:08.330 ->   Cores             : 2
21:31:08.330 ->   CPU Frequency     : 240 MHz
21:31:08.330 ->   XTAL Frequency    : 40 MHz
21:31:08.330 ->   Features Bitfield : 0x00000032
21:31:08.362 ->   Embedded Flash    : No
21:31:08.362 ->   Embedded PSRAM    : No
21:31:08.362 ->   2.4GHz WiFi       : Yes
21:31:08.362 ->   Classic BT        : Yes
21:31:08.362 ->   BT Low Energy     : Yes
21:31:08.362 ->   IEEE 802.15.4     : No
21:31:08.362 -> ------------------------------------------
21:31:08.394 -> INTERNAL Memory Info:
21:31:08.394 -> ------------------------------------------
21:31:08.394 ->   Total Size        :   243444 B ( 237.7 KB)
21:31:08.394 ->   Free Bytes        :   205096 B ( 200.3 KB)
21:31:08.394 ->   Allocated Bytes   :    33052 B (  32.3 KB)
21:31:08.427 ->   Minimum Free Bytes:   200304 B ( 195.6 KB)
21:31:08.427 ->   Largest Free Block:   110580 B ( 108.0 KB)
21:31:08.427 -> ------------------------------------------
21:31:08.427 -> Flash Info:
21:31:08.427 -> ------------------------------------------
21:31:08.427 ->   Chip Size         :  4194304 B (4 MB)
21:31:08.459 ->   Block Size        :    65536 B (  64.0 KB)
21:31:08.459 ->   Sector Size       :     4096 B (   4.0 KB)
21:31:08.459 ->   Page Size         :      256 B (   0.2 KB)
21:31:08.459 ->   Bus Speed         : 80 MHz
21:31:08.459 ->   Bus Mode          : DIO
21:31:08.492 -> ------------------------------------------
21:31:08.492 -> Partitions Info:
21:31:08.492 -> ------------------------------------------
21:31:08.492 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
21:31:08.524 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
21:31:08.524 ->                app0 : addr: 0x00010000, size:  3072.0 KB, type:  APP, subtype: OTA_0
21:31:08.556 ->              spiffs : addr: 0x00310000, size:   896.0 KB, type: DATA, subtype: SPIFFS
21:31:08.556 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
21:31:08.588 -> ------------------------------------------

21:31:08.588 -> Software Info:
21:31:08.588 -> ------------------------------------------
21:31:08.588 ->   Compile Date/Time : Jan 24 2025 21:30:03
21:31:08.588 ->   Compile Host OS   : macosx
21:31:08.620 ->   ESP-IDF Version   : v5.3.2-282-gcfea4f7c98-dirty
21:31:08.620 ->   Arduino Version   : 3.1.1
21:31:08.620 -> ------------------------------------------
21:31:08.620 -> Board Info:
21:31:08.620 -> ------------------------------------------
21:31:08.620 ->   Arduino Board     : ESP32_DEV
21:31:08.653 ->   Arduino Variant   : doitESP32devkitV1
21:31:08.653 ->   Arduino FQBN      : esp32:esp32:esp32doit-devkit-v1:UploadSpeed=921600,FlashFreq=80,DebugLevel=debug,EraseFlash=none
21:31:08.653 -> ============ Before Setup End ============
21:31:08.750 -> [   503][D][a2dpmusic.ino:181] setup(): [test] this is a test log
21:31:08.783 -> [   510][D][BluetoothA2DPSink.cpp:89] start(): [BT_AV] start
21:31:08.783 -> [   515][I][BluetoothA2DPCommon.cpp:267] log_free_heap(): [BT_AV] Available Heap: 141124
21:31:08.783 -> [   523][I][BluetoothA2DPSink.cpp:101] start(): [BT_AV] Device name will be set to 'ESP32-44'
21:31:08.783 -> [   532][D][BluetoothA2DPSink.cpp:1060] init_nvs(): [BT_AV] init_nvs
21:31:08.783 -> [   538][D][BluetoothA2DPCommon.cpp:163] get_last_connection(): [BT_AV] get_last_connection
21:31:08.815 -> [   546][E][BluetoothA2DPCommon.cpp:197] read_address(): [BT_AV] NVS OPEN ERROR
21:31:08.815 -> [   553][E][BluetoothA2DPCommon.cpp:206] read_address(): [BT_AV] nvs_get_blob failed
21:31:08.815 -> [   561][D][BluetoothA2DPCommon.cpp:169] get_last_connection(): [BT_AV] => 00:00:00:00:00:00
21:31:08.815 -> [   569][I][BluetoothA2DPSink.cpp:164] init_i2s(): [BT_AV] init_i2s
21:31:08.848 -> [   575][I][BluetoothA2DPOutput.cpp:196] begin(): [BT_AV] begin
21:31:08.848 -> [   582][D][BluetoothA2DPSink.cpp:184] init_bluetooth(): [BT_AV] init_bluetooth
21:31:08.912 -> [   663][I][BluetoothA2DPSink.cpp:189] init_bluetooth(): [BT_AV] controller initialized
[... 56 lines cut ...]
21:31:09.756 -> [  1498][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:38] allocateDMAmemory(): [I2S-DMA] Free heap: 48628

21:31:09.756 -> [  1507][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:39] allocateDMAmemory(): [I2S-DMA] Free SPIRAM: 0
21:31:09.787 -> [  1516][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:43] allocateDMAmemory(): [I2S-DMA] allocating rowBitStructs with pixel_color_depth_bits of 8
21:31:09.787 -> [  1529][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:58] allocateDMAmemory(): [I2S-DMA] CRITICAL ERROR: Not enough memory for requested colour depth! Please reduce pixel_color_depth_bits value.
21:31:09.820 -> 
21:31:09.820 -> [  1546][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:59] allocateDMAmemory(): [I2S-DMA] Could not allocate rowBitStruct 17!.
21:31:09.820 -> 
21:31:09.820 -> [  1557][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:629] setBrightness(): [setBrightness()] Tried to set output brightness before begin()
21:31:09.820 -> =========== After Setup Start ============
21:31:09.820 -> INTERNAL Memory Info:
[... cut ...]
21:31:09.852 ->   Minimum Free Bytes:    12276 B (  12.0 KB)
21:31:09.852 ->   Largest Free Block:    10228 B (  10.0 KB)
21:31:09.852 -> ------------------------------------------
21:31:09.852 -> GPIO Info:
21:31:09.852 -> ------------------------------------------
21:31:09.885 ->   GPIO : BUS_TYPE[bus/unit][chan]
21:31:09.885 ->   --------------------------------------  
21:31:09.885 ->      1 : UART_TX[0]
21:31:09.885 ->      3 : UART_RX[0]
21:31:09.885 -> ============ After Setup End =============

makerawei, 2025-01-24:
the text above is my log, please help me ~.~ thank you very much!

pschatzmann, 2025-01-24:
You obviously don't have enough RAM for the MatrixPanel!

Did you activate PSRAM?

But test first just with a simple A2DP sketch! If this is working you are asking for help at the wrong place!

makerawei, 2025-01-24:
I fixed the RAM not enough error, but I can not connect to ESP32 device. logs as below. Please help me~.~  
22:54:39.062 -> [   525][D][a2dpmusic.ino:181] setup(): [test] this is a test log
22:54:39.062 -> [   532][D][BluetoothA2DPSink.cpp:89] start(): [BT_AV] start
22:54:39.062 -> [   537][I][BluetoothA2DPCommon.cpp:267] log_free_heap(): [BT_AV] Available Heap: 153956
22:54:39.062 -> [   545][I][BluetoothA2DPSink.cpp:101] start(): [BT_AV] Device name will be set to 'ESP32-44'
22:54:39.094 -> [   553][D][BluetoothA2DPSink.cpp:1060] init_nvs(): [BT_AV] init_nvs
22:54:39.094 -> [   559][D][BluetoothA2DPCommon.cpp:163] get_last_connection(): [BT_AV] get_last_connection
22:54:39.094 -> [   568][E][BluetoothA2DPCommon.cpp:197] read_address(): [BT_AV] NVS OPEN ERROR
22:54:39.094 -> [   575][E][BluetoothA2DPCommon.cpp:206] read_address(): [BT_AV] nvs_get_blob failed
22:54:39.094 -> [   582][D][BluetoothA2DPCommon.cpp:169] get_last_connection(): [BT_AV] => 00:00:00:00:00:00
22:54:39.127 -> [   590][I][BluetoothA2DPSink.cpp:164] init_i2s(): [BT_AV] init_i2s
22:54:39.127 -> [   596][I][BluetoothA2DPOutput.cpp:196] begin(): [BT_AV] begin
22:54:39.127 -> [   603][D][BluetoothA2DPSink.cpp:184] init_bluetooth(): [BT_AV] init_bluetooth
22:54:39.193 -> [   683][I][BluetoothA2DPSink.cpp:189] init_bluetooth(): [BT_AV] controller initialized
22:54:39.521 -> [  1002][I][BluetoothA2DPSink.cpp:198] init_bluetooth(): [BT_AV] bluedroid initialized
22:54:39.617 -> [  1088][I][BluetoothA2DPSink.cpp:207] init_bluetooth(): [BT_AV] bluedroid enabled
22:54:39.618 -> [  1095][D][BluetoothA2DPSink.cpp:320] app_task_start_up(): [BT_AV] app_task_start_up
22:54:39.618 -> [  1103][D][BluetoothA2DPSink.cpp:1175] ccall_app_task_handler(): [BT_AV] ccall_app_task_handler
22:54:39.649 -> [  1112][D][BluetoothA2DPSink.cpp:276] app_task_handler(): [BT_AV] app_task_handler
22:54:39.649 -> [  1119][D][BluetoothA2DPSink.cpp:232] app_work_dispatch(): [BT_API] app_work_dispatch event 0x0, param len 0
22:54:39.649 -> [  1129][D][BluetoothA2DPSink.cpp:262] app_send_msg(): [BT_AV] app_send_msg
22:54:39.649 -> [  1136][D][BluetoothA2DPSink.cpp:284] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x0
22:54:39.681 -> [  1144][D][BluetoothA2DPSink.cpp:287] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
22:54:39.681 -> [  1155][D][BluetoothA2DPSink.cpp:255] app_work_dispatched(): [BT_AV] app_work_dispatched
22:54:39.681 -> [  1162][D][BluetoothA2DPSink.cpp:1208] ccall_av_hdl_stack_evt(): [BT_AV] ccall_av_hdl_stack_evt
22:54:39.681 -> [  1171][D][BluetoothA2DPSink.cpp:908] av_hdl_stack_evt(): [BT_AV] av_hdl_stack_evt evt 0
22:54:39.714 -> [  1179][D][BluetoothA2DPSink.cpp:913] av_hdl_stack_evt(): [BT_AV] av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
22:54:39.714 -> [  1190][D][BluetoothA2DPSink.cpp:923] av_hdl_stack_evt(): [BT_AV] AVRCP controller initialized!
22:54:39.714 -> [  1190][D][BluetoothA2DPSink.cpp:1202] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
22:54:39.745 -> [  1207][I][BluetoothA2DPSink.cpp:431] app_gap_callback(): [BT_AV] event: 10
22:54:39.745 -> [  1207][D][BluetoothA2DPSink.cpp:969] av_hdl_stack_evt(): [BT_AV] set_scan_mode_connectable(true)
22:54:39.745 -> [  1222][I][BluetoothA2DPCommon.cpp:368] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
22:54:39.745 -> [  1227][D][BluetoothA2DPSink.cpp:1188] ccall_app_a2d_callback(): [BT_AV] ccall_app_a2d_callback
22:54:39.777 -> [  1240][D][BluetoothA2DPSink.cpp:983] app_a2d_callback(): [BT_AV] app_a2d_callback
22:54:39.777 -> [  1241][I][BluetoothA2DPSink.cpp:158] start(): [BT_AV] IDF Version 5.3
22:54:39.777 -> [  1254][D][BluetoothA2DPSink.cpp:1006] app_a2d_callback(): [B[T _ A1V2]5 4a]p[pI_]a[2Bdl_uceatlolobtahcAk2 DEPSCPo_mAm2oDn_.AcUpD:267F _EV_
22:54:39.777 -> ee_heap(): [BT_AV] Available Heap: 51516
22:54:39.809 -> waiting for connection...

makerawei, 2025-01-24:
23:05:05.746 -> [ 99939][I][BluetoothA2DPSink.cpp:431] app_gap_callback(): [BT_AV] event: 16
23:05:13.980 -> [108198][D][BluetoothA2DPSink.cpp:1202] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
23:05:14.011 -> [108207][I][BluetoothA2DPSink.cpp:431] app_gap_callback(): [BT_AV] event: 17