pschatzmann/ESP32-A2DP #673: Task watchdog got triggered

hientv1999, 2025-02-22:
### Problem Description

I tested your example bt_music_receiver_playing_status_callbacks. It works fine
However, when I test your example bt_music_receiver_arduino_i2s_3 with a modified GPIO to work on my custom board, I got below error:

```
[   497][E][BluetoothA2DPCommon.cpp:270] read_address(): [BT_AV] NVS OPEN ERROR
[   504][E][BluetoothA2DPCommon.cpp:279] read_address(): [BT_AV] nvs_get_blob failed
E (28742) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (28742) task_wdt:  - IDLE0 (CPU 0)
E (28742) task_wdt: Tasks currently running:
E (28742) task_wdt: CPU 0: BTC_TASK
E (28742) task_wdt: CPU 1: loopTask
E (28742) task_wdt: Aborting.
E (28742) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400dfa46:0x3ffd8be0 0x400da2c3:0x3ffd8c20 0x400da32a:0x3ffd8c60 0x40227945:0x3ffd8c90 0x402278b1:0x3ffd8cb0 0x4022780d:0x3ffd8cd0 0x400d9255:0x3ffd8cf0 0x402275dd:0x3ffd8d10 0x400d8137:0x3ffd8d30 0x4014cd1d:0x3ffd8d50 0x401453a9:0x3ffd8d90 0x40145345:0x3ffd8db0 0x40099c96:0x3ffd8de0

ELF file SHA256: 412ba94262c656c0

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr
Core  0 register dump:
PC      : 0x40083f38  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc6c70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x3ffd8b20  A8      : 0x00000000  A9      : 0x00000008  
A10     : 0x00000000  A11     : 0xfa1e13f8  A12     : 0xfa1e13f8  A13     : 0x00060d21
A14     : 0xfffbfff0  A15     : 0x00000046  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40092d8c  LEND    : 0x40092da2  LCOUNT  : 0xffffffff

Backtrace: 0x40083f35:0x3ffc6c70 0x40080303:0x3ffc6ca0 0x65a5a5a2:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 412ba94262c656c0

Re-entered core dump! Exception happened during core dump!
Rebooting...
ets Jul 29 2019 12:21:46
```

### Device Description

Espressif Systems ESP32-WROVER-IE-N16R8

### Sketch

```cpp
#include "ESP_I2S.h"
#include "BluetoothA2DPSink.h"

I2SClass i2s;

BluetoothA2DPSink a2dp_sink(i2s);

void setup() {

      Serial.begin(115200);
    digitalWrite(4, LOW);
    i2s.setPins(26, 12, 14);
    if (!i2s.begin(I2S_MODE_STD, 44100, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO, I2S_STD_SLOT_BOTH)) {
      Serial.println("Failed to initialize I2S!");
      while (1); // do nothing
    }

    a2dp_sink.start("MyMusic");
}

void loop() {
  delay(1000);
}
```

### Other Steps to Reproduce

These pins should be fine for I2S output as I have used the same pin layout for a different I2S project.

I tested on both Android and iPhone. They both crash the ESP32

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-02-22:
I have never seen anything like this so try:
 
- to replace I2SClass with the I2SStream from the AudioTools
- to check with another ESP32-WROVER-IE-N16R8 if this is a hardware bug
- to reproduce the issue with Arduino to check if this an issue with PlatformIO
- set the log level to Info to get a better understanding where the issue happens
- try call a2dp_sink.set_auto_reconnect(true); and run the sketch once 

I am closing this issue because I can't reproduce your issue.

pschatzmann, 2025-02-22:
I optimized the logic a bit so that the NVS is not used when not needed: this should make your nvs errors go away...

hientv1999, 2025-02-22:
I switched to Arduino 3.1.3 and it seems to no longer crash

closed 2025-02-22 by hientv1999