pschatzmann/ESP32-A2DP #683: A2DP Sink and Source Incompatibility

phatboydave, 2025-03-19:
### Problem Description

The issue I am encountering is that there is some fundamental bug (or intentional lack of support) which is not allowing both a BluetoothA2DPSink object and a BluetoothA2DPSource object to be defined in the same program.

When testing, I tried both of the most basic configurations and they both individually worked fine (connected to iPhone 16, and to a Bose bluetooth speaker). However, when combined, even if started sequentially (ensuring I have connected to the sink properly before running start() on the source), both features simply fail.

I am not sure if this is intentional or a mistake (I hope it is the former). The goal of my project is to create an audio bluetooth hub to synchronize various speakers to my iPhone audio which is why I ran into this issue.

To easily recreate this issue, just define both variables in the global space as can be seen in the "Sketch Code" section.

Thank you in advance for taking the time to consider this, and I deeply appreciate the work done on this project!

### Device Description

ESP32-WROOM-32 Dev Board (recently bought on Amazon).

### Sketch

```cpp
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"
#include "BluetoothA2DPSource.h"

I2SStream data_stream;

BluetoothA2DPSink a2dp_sink(data_stream);
BluetoothA2DPSource a2dp_source;

void setup() { /* etc... */ }

void loop() { /* etc... */ }
```

### Other Steps to Reproduce

It should be rather easy to reproduce this error. Not exactly sure where the fundamental conflict is coming from though.

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

ESP32-D0WD-V3 (revision v3.1)

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-19:
This is not supported!

closed 2025-03-19 by pschatzmann

phatboydave, 2025-03-19:
Oh I see thank you. Just so I do not waste any more time on this; do you think it would be possible to at least host multiple A2DP sinks on one ESP32 and then I2C that over to another who will distrubute synchronously to various connected devices through sources?

SamObde, 2025-04-06:
I have a product im working on that has 8 esp32s on it and you can choose to use them as sources or sinks with a mux for choosing the active source.  Almost got it all working - just had PCB made.

One thing you will run into is each speaker has its own buffer / delays - so unless you sync them they will be slightly out of sync.