pschatzmann/ESP32-A2DP #684: crash after start playing

harryberlin, 2025-03-23:
### Problem Description

my program crashes at begin start playing.
connecting works fine.
start playing on smartphone. -> crash

Could you give some help, to find the methode inside the sketch?
maybe a callback method?

latest library version of esp32-a2dp and audio-tools

```
22:05:14.409 -> CA:CE:48:46:B6:B0:99
22:05:14.409 -> Connected::C2
22:05:14.503 -> 
22:05:14.503 -> assert failed: xRingbufferSend ringbuf.c:1027 (pxRingbuffer)
22:05:14.503 -> 
22:05:14.503 -> 
22:05:14.503 -> Backtrace: 0x400839d1:0x3ffe65c0 0x400958a1:0x3ffe65e0 0x4009b395:0x3ffe6600 0x400953c6:0x3ffe6730 0x400dc3e6:0x3ffe6760 0x402187dd:0x3ffe6790 0x400db763:0x3ffe67b0 0x40149dd1:0x3ffe67d0 0x40141d61:0x3ffe6810 0x40141cf7:0x3ffe6830
22:05:14.549 -> 
22:05:14.549 -> 
22:05:14.549 -> 
22:05:14.549 -> 
22:05:14.549 -> ELF file SHA256: 3e53ab5b65a5bf15
22:05:14.549 -> 
22:05:14.549 -> E0ISR
22:05:14.784 -> Rebooting...
22:05:14.784 -> ets Jul 29 2019 12:21:46
22:05:14.784 -> 
22:05:14.784 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:05:14.784 -> configsip: 0, SPIWP:0xee
22:05:14.784 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:05:14.784 -> mode:DIO, clock div:1
22:05:14.784 -> load:0x3fff0030,len:1344
22:05:14.784 -> load:0x40078000,len:13964
22:05:14.784 -> load:0x40080400,len:3600
22:05:14.784 -> entry 0x400805f0
22:05:15.393 -> BUILD:Mar 22 2025,21:26:50
22:05:15.393 -> DA
22:05:15.393 -> BTSerial:1
22:05:16.893 -> Connecting::C1
22:05:16.939 -> M0
22:05:18.392 -> CA:CE:48:46:B6:B0:99
22:05:18.392 -> Connected::C2
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x1, 
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x2, 
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x4, 
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x8, 1
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x10, 1
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x20, 
22:05:18.580 -> ==> AVRC metadata rsp: attribute id 0x40, 0
22:05:18.580 -> ==> Playing time is 0 ms (0 seconds)
22:05:18.580 -> Volume changed: 0
22:05:18.626 -> Volume change: 15
22:05:19.423 -> Volume changed: 15

```

decoded backtrace:

```

Decoding stack results
0x400839d1: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 408
0x400958a1: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 137
0x4009b395: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x400953c6: xRingbufferSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ringbuf/ringbuf.c line 1028
0x400dc3e6: BluetoothA2DPSinkQueued::write_audio(unsigned char const*, unsigned int) at C:\Users\Christian\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSinkQueued.cpp line 113
0x402187dd: BluetoothA2DPSink::audio_data_callback(unsigned char const*, unsigned int) at C:\Users\Christian\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp line 985
0x400db763: ccall_audio_data_callback(uint8_t const*, uint32_t) at C:\Users\Christian\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp line 30
0x40149dd1: btc_a2dp_sink_data_ready at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/btc/profile/std/a2dp/btc_a2dp_sink.c line 158
0x40141d61: osi_thread_generic_event_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/thread.c line 425
0x40141cf7: osi_thread_run at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/thread.c line 165

```

### Device Description

ESP32-WROOM-32

### Sketch

```cpp
tbd
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.16, because the updated version blow up the program size

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-24:
I am not supporting these old releases: please upgrade to an acutal release. A regular Sketch is only using 1.8 M, but the ESP32 has 4 M PROGMEM, so there is more then enough space left: just use a proper partition scheme.

You usually get these kind of errors if you run out of RAM: so if you run low on RAM you should better use the BluetoothA2DPSink instread of the BluetoothA2DPSinkQueued.

I am closing this issue because I can't reproduce it...

closed 2025-03-24 by pschatzmann

harryberlin, 2025-03-24:
updated espress idf to 3.1.3 and got the crash also.
switched to BluetoothA2DPSink.
no crash but sometimes few clicks during playback or volume change. that's why a i have swapped to BluetoothA2DPSinkQueued before.
PSRAM don't exist.

harryberlin, 2025-03-24:
got crash again:
```
22:28:23.003 -> 
22:28:23.003 -> assert failed: host_recv_pkt_cb hci_hal_h4.c:580 (0)
22:28:23.003 -> 
22:28:23.003 -> 
22:28:23.003 -> Backtrace: 0x40082525:0x3ffe52a0 0x40094e79:0x3ffe52c0 0x4009b15e:0x3ffe52e0 0x40163db9:0x3ffe5410 0x401fe02e:0x3ffe5430 0x401e9d26:0x3ffe5450 0x4001791d:0x3ffe5470 0x401ea741:0x3ffe5490 0x401ea81d:0x3ffe54d0 0x400188f5:0x3ffe5500 0x401ea063:0x3ffe5540 0x401ef9d5:0x3ffe5560 0x4008a69e:0x3ffe5580 0x40019d11:0x3ffe55b0 0x40055b4d:0x3ffe55d0 0x401e8333:0x3ffe55f0 0x401e8995:0x3ffe5610 0x40095f32:0x3ffe5640
22:28:23.049 -> 
22:28:23.049 -> 
22:28:23.049 -> 
22:28:23.049 -> 
22:28:23.049 -> ELF file SHA256: 56e364b4b
22:28:23.049 -> 
22:28:23.331 -> Rebooting...
22:28:23.331 -> ets Jul 29 2019 12:21:46
22:28:23.331 -> 
22:28:23.331 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:28:23.331 -> configsip: 0, SPIWP:0xee
22:28:23.331 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:28:23.331 -> mode:DIO, clock div:1
22:28:23.331 -> load:0x3fff0030,len:4916
22:28:23.331 -> load:0x40078000,len:16492
22:28:23.331 -> load:0x40080400,len:4
22:28:23.331 -> load:0x40080404,len:3524
22:28:23.331 -> entry 0x400805b8
22:28:23.893 -> BUILD:Mar 24 2025,22:27:22
22:28:23.893 -> DA
22:28:23.893 -> BTSerial:1
22:28:25.393 -> Connecting::C1
22:28:25.486 -> M0
22:28:27.080 -> CA:CE:48:46:B6:B0:99
22:28:27.080 -> Connected::C2
```

```
0x40082525: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 463
0x40094e79: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 92
0x4009b15e: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x40095f32: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139
```

harryberlin, 2025-03-24:
after hardware reset and playing audio, there is very bad audio for nearly 30 seconds, after this the audio is playing good.

pschatzmann, 2025-03-24:
And what about my comment about the RAM ? Please log your RAM e.g. with

void loop(){
   checkMemory(true);
   delay(2000);
}  

My best bet is still that you just run out of RAM: so if you still use the BluetoothA2DPSinkQueued, I suggest to decrease the queue size.