pschatzmann/ESP32-A2DP #685: Problems with volume control and callbacks

ArthurKoba, 2025-03-23:
### Problem Description

Hello. I recently returned to the development of my light and music system and noticed that the library was significantly reworked. Due to the fact that my speaker system has changed, I decided to add SPDIF for galvanic isolation and have not yet checked whether there are problems when using standard i2s.

Now about the problems:
1. Volume control does not work. Neither the phone nor the laptop (win11) receive volume change events, as it was in the old version.
2. Based on the first, callbacks about volume changes are not called.
3. The firmware also crashes when audio playback is stopped (Double exception).
4. When installing a callback to change the volume when connected, the firmware crashes.

When I started collecting information and tried the standard i2s, I saw that the volume is adjustable. When I switched to spdif, the problems went away. At first I didn't understand how the bug was fixed and what the problem was, but I noticed that when preparing the example for issue, instead of delay in the loop function, I used `vTaskDelete(nullptr);` which stops the Arduino `loop` task. If the task is not completed and delay is used, all the problems appear and are tracked.

This may be due to the fact that the heap memory is running out. I have seen many times how only ~ 5000 RAM remained, although normally about 100000... Potentially, the Arduino task takes up too much memory and can lead to this problem. I don't know, I personally don't use loop, I create tasks myself, but someone may need it and this can lead to problems.

### Device Description

Board: ESP32-DevKit with type-c and ch341 (esp-wroom-34)
Phone: Mi 10T Pro on CrDroid firmware, Android 14.
Laptop: RedmiBook 16 pro 2024, bt - intel ax211, windows 11

PACKAGES:
framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
audio-tools @ 1.1.0+sha.350fefc
ESP32-A2DP @ 1.8.7+sha.8faaafa

### Sketch

```cpp
#include "BluetoothA2DPSink.h"
#include "AudioTools/AudioLibs/SPDIFOutput.h"

#define SERIAL_SPEED 1000000

SPDIFOutput stream;

BluetoothA2DPSink a2dp_sink(stream);

void volume_cb(int vol) {
    Serial.printf("New volume: %u\n", vol);
}

void setup() {
    Serial.begin(SERIAL_SPEED);
    Serial.println("\nStart Test Entry!");

    auto cfg = stream.defaultConfig();
    cfg.buffer_size = 384;
    cfg.buffer_count = 30;
    cfg.pin_data = 23;
    stream.begin(cfg);

    a2dp_sink.set_on_volumechange(volume_cb);
    a2dp_sink.set_volume(64);
    a2dp_sink.start("MyMusic");
}

void loop() {
//    vTaskDelete(nullptr); // Fixed bugs in issue (close arduino loop task without error)
    delay(1000); // bugs in issue
}
```

### Other Steps to Reproduce

Log with error when stopping:
```
[ 47910][D][BluetoothA2DPSink.cpp:956] audio_data_callback(): [BT_AV] audio_data_callback
[ 47929][D][BluetoothA2DPSink.cpp:956] audio_data_callback(): [BT_AV] audio_data_callback
[ 47938][D][BluetoothA2DPSink.cpp:919] app_a2d_callback(): [BT_AV] app_a2d_callback
[ 47939][D][BluetoothA2DPSink.cpp:928] app_a2d_callback(): [BT_AV] app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
[... 113 lines cut ...]
A14     : 0x3ffb6740  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000002
EXCVADDR: 0xffffffe0  LBEG    : 0x40092679  LEND    : 0x40092689  LCOUNT  : 0xfffffffb

Backtrace: 0x400951f3:0x3fff8930 0x4009594d:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x400951f3
:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x400951f
3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x400951
f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x40095
1f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x4009
51f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x400
951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x40
0951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x4
00951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0x
400951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20 0
x400951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20
0x400951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 0x400951f3:0x3fff8a00 0x400951f3:0x3fff8a20
 0x400951f3:0x3fff8a40 0x4008007d:0x3fff8930 0x40099e61:0x3fff8970 0x400951f3:0x3fff8990 0x400951f3:0x3fff89b0 0x400951f3:0x3fff89d0 |<-CONTINUES
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-24:
I corrected the includes in your sketch that were giving compile errors and tested it with the lates 3.2 Arduino Core.
I could not reproduce any issues and I was getting the following with log level Warning:

New volume: 85
New volume: 94
New volume: 85
New volume: 77
[ 27704][W][BluetoothA2DPOutput.cpp:255] set_output_active(): [BT_AV] i2s_stop
[ 35328][W][BluetoothA2DPOutput.cpp:255] set_output_active(): [BT_AV] i2s_stop
[141143][W][BluetoothA2DPOutput.cpp:255] set_output_active(): [BT_AV] i2s_stop
[144864][W][BluetoothA2DPOutput.cpp:255] set_output_active(): [BT_AV] i2s_stop

I was changing the volume on the phone and I was stopping the playback on the phone both by hitting the stop button and by disconnecting from Bluetooth...

closed 2025-03-24 by pschatzmann

harryberlin, 2025-03-24:
maybe topic fits here, looks like these callbacks don't work anymore:
```
a2dp_sink.set_avrc_rn_volumechange(avrc_rn_volume_change_callback);
a2dp_sink.set_avrc_rn_volumechange_completed(avrc_rn_volume_changed_callback);

void avrc_rn_volume_change_callback(int volume) {
  Serial.printf("Volume change: %d\n", volume);
}

void avrc_rn_volume_changed_callback(int volume) {
  Serial.printf("Volume changed: %d\n", volume);
}
```
no notification after volume change