pschatzmann/ESP32-A2DP #688: Eventually out of heap memory

WillyBilly06, 2025-03-26:
### Problem Description

Apparently, the amount of IRAM that's assigned to run A2Dp is only 81Kb-82KB. Whenever I run it using Windows and switching tabs back and forth, the amount of free ram depletes until it cannot allocate 8192 bytes, which is an error from the file Allocator.h. Are there any ways that I can clear or maybe free the heap memory while the A2DP is running and while I'm streaming music to ESP32?. This module has no PSRAM so I cannot allocate it to use for the application

### Device Description

LILY TTGO T-DISPLAY

### Sketch

```cpp
/*
  Streaming Music from Bluetooth
  
  Copyright (C) 2020 Phil Schatzmann
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

// ==> Example A2DP Receiver which uses the A2DP I2S output with 32 bits

#include "AudioTools.h"
#include "BluetoothA2DPSink.h"
#include "C:\Users\BillyDaKidz\Documents\Arduino\libraries\audio-tools\src\AudioTools\CoreAudio\AudioBasic\Collections\Allocator.h"
AudioInfo info(44100, 2, 24);
I2SStream out;
NumberFormatConverterStream convert(out);
BluetoothA2DPSink a2dp_sink(convert);
Allocator allocator;
void setup() {
  Serial.begin(115200);
  // AudioToolsLogger.begin(Serial, AudioLogger::Info);

  // Configure i2s to use 32 bits
  auto cfg = out.defaultConfig();
  cfg.copyFrom(info);
  cfg.pin_bck = 27;
  cfg.pin_data = 26;
  cfg.pin_ws = 25;
  out.begin(cfg);

  // Convert from 16 to 32 bits
  convert.begin(16, 24);

  // start a2dp
  a2dp_sink.start("DeviceName");  
  
}

void loop() {
delay(400);
}
```

### Other Steps to Reproduce

Connect the module, play music/video on youtube and then foward/backward the page and the issue will appear

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.1.3

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-26:
I have no clue what you are talking about: What does Windows and switching tabs back and forth actually mean ?

WillyBilly06, 2025-03-26:
Oh, like if you playing something on Youtube and then click on another video. If you "Backward" to the previous video and then "Forward" to the video you clicked. By doing it repetitively, the heap memory of ESP32 eventually run out until you need to restart the ESP32. 

Or what you can do is go to "Manage output audio device", click to your ESP32's bluetooth name and then turn on/off Audio Enhancement. Eventually the ESP32 will stop playing sound because of out of heap memory and it will output "Failed to allocate 8192 bytes" in Arduino IDE Serial Monitor.

pschatzmann, 2025-03-26:
I don't have Windows, so I need to try to reproduce this from my Android phone.
Do you have a backtrace ?

pschatzmann, 2025-03-26:
I committed a correction to the NumberFormatConverterStream in the AudioTools. Objects were allocated, but never released

WillyBilly06, 2025-03-26:
Confirmed that the issue is fixed! Thank you for all your hard work to create this library for the community!

closed 2025-03-26 by WillyBilly06