pschatzmann/ESP32-A2DP #692: apll not working with pioarduino

cQszasza, 2025-03-30:
### Problem Description

Hi! 
First of all, I'm a complete noob when it comes to programming so it is very well possible that the problem is on my side. But!
Can someone please confirm that the apll is working with idf>5? 
I'm using the 32bits a2dp to i2s example code so I'm not going to copy my sketch here, it is untouched, and it makes it easy to reproduce the problem. 
Using PlatformIO, just put `platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip` in the ini file. 
The apll stops working and as a result I have a horrible jitter on the clocks. Without the apll the audio clocks are all over the place so it is quite easy to spot it on a scope(just trigger on LRCK and check BCLK on the second channel), and if I have a DAC connected this results in a very high audio noise floor. 
I need this idf version for something else in my project so please don't just say to use `platform = espressif32` in the ini file. By the way this does not complile for me anyway, `platform = https://github.com/platformio/platform-espressif32.git` does complile but I have a different problem with this one. 
So, how can I force the apll on with the forementioned idf version? 
Thank you for this awesome library!

### Device Description

ESP32 Wroom with a scope attached on LRCK and BCLK.

### Sketch

```cpp
bt_music_receiver_32bits.ino compiled in PlatformIO
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

currently espressif Arduino 3.2.0 and IDF 5.4.1

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

cQszasza, 2025-03-30:
Obviously forgot to mention that you do need to turn on apll for this test...

pschatzmann, 2025-03-30:
I removed my support for use_apll because that was just not give any valid audio signal!

You can double check if in your case you get what you expect: just make sure that the getClockSource() method returns I2S_CLK_SRC_APLL. If you confirm that this works for you, I can add the functionality back....

cQszasza, 2025-03-31:
Thank you!
This seems to be fine apart from one thing, it gives me FS*512 (around 22.5 Mhz) on the MCLK pin instead of the "usual" FS*256. I'm using MCLK to clock a DSP chip and I can't really change my hardware, its an old DSP chip and its not very flexible when it comes to clocks... Can you help me where to look for changing the MCLK frequency? The pll seems to be working fine.

cQszasza, 2025-04-01:
I think I found something altough this should be in the audio lib repo and not here I belive...
In I2SESP32V1.h row 282 there is this: `clk_cfg.mclk_multiple = I2S_MCLK_MULTIPLE_256;` If I change this to 128 the MCLK is correct for FS256, so there is a doubling going on somewhere... I hope this helps. 
Everithing else works otherwise so I think you can add back the apll feautre, the audio is good, for me anyways.

pschatzmann, 2025-04-01:
The [pull request has been merged](https://github.com/pschatzmann/arduino-audio-tools/pull/1989)

closed 2025-04-01 by pschatzmann