pschatzmann/ESP32-A2DP #720: Source does not send data frames after seconds from the start of the stream.

semihtuncer, 2025-06-20:
### Problem Description

While all the connections are successful and sd card is not faulty, reading the audio data correctly, there is no data frames passing to the sink. There is no sound coming from the device that is connected.

### Device Description

- ESP32-CAM AI THINKER. (NO CAMERA MODULE INSTALLED, USING JUST FOR THE SD CARD MODULE.)
- Airpods Pro (CONNECTS TO THE SOURCE.)

### Sketch

```cpp
#include "FS.h"
#include "SPI.h"
#include "SD_MMC.h"
#include "soc/soc.h"
#include "driver/rtc_io.h"
#include "soc/rtc_cntl_reg.h"
#include "BluetoothA2DPSource.h"

const char* wifiName = "xxx";
const char* wifiPwd = "xxx";
const char* bluetoothDeviceName = "xxx";

File file;
BluetoothA2DPSource source;

int songNDX = 0;
int playlistNDX = 0;
bool isPlaying = false;

float random_float() {
    return ((float)rand() / RAND_MAX) * 2.0f - 1.0f;
}

String getFileNameFromNDX(int ndx) {
  return "/" + String(ndx) + ".wav";
}
void changeSongToNDX(int ndx) {
  if (file)
    file.close();

  delay(50);

  String playFile = getFileNameFromNDX(ndx);
  Serial.print("playing file: ");
  Serial.println(playFile);
  
  file = SD_MMC.open(playFile, FILE_READ);
  
  if (file) {
    file.seek(44);
  } else {
    Serial.println("failed to open file");
    isPlaying = false;
  }
}

int32_t getDataFrames(Frame* data, int32_t frameCount) {
  if (!isPlaying) {
    for (int i = 0; i < frameCount; i++) {
      data[i].channel1 = 0;
      data[i].channel2 = 0;
    }
    return frameCount;
  }

  if (!file) {
    changeSongToNDX(songNDX);
    return frameCount;
  }

  if (!file.available()) {
    file.close();
    delay(50);
    songNDX++;
    changeSongToNDX(songNDX);
    return frameCount;
  }

  size_t sample = file.read((uint8_t*)data, frameCount * sizeof(Frame));
  delay(1);

  Serial.printf("song NDX: %d | file available: %d\n", songNDX, file.available());

  return frameCount;
}

// int32_t getDataFrames(Frame* data, int32_t frameCount) {
//   if (!isPlaying) {
//     for (int i = 0; i < frameCount; i++) {
//       data[i].channel1 = 0;
//       data[i].channel2 = 0;
//     }
//     return frameCount;
//   }

//   for (int i = 0; i < frameCount; i++) {
//     float sample = random_float();
//     int16_t pcm = static_cast<int16_t>(sample * 32767);
//     data[i] = Frame(pcm);
//   }
//   Serial.println("a");
//   return frameCount;
// }

void onTogglePlaying() {
  isPlaying = !isPlaying;
}
void onNextSong() {
  songNDX = songNDX + 1;
  changeSongToNDX(songNDX);
}
void onPreviousSong() {
  if(songNDX == 0)
    return;

  songNDX = songNDX - 1;
  changeSongToNDX(songNDX);
}
void handleButtons(uint8_t id, bool isReleased) {
  if (isReleased) {
    Serial.print("button pressed: ");
    Serial.println(id);

    switch (id) {
      case 70:
        {
          onTogglePlaying();
        }
        break;
      case 75:
        {
          onNextSong();
        }
        break;
      case 76:
        {
          onPreviousSong();
        }
        break;
    }
  }
}

bool hasGreeted = false;
void startConnection() {
  source.set_auto_reconnect(true);
  source.set_data_callback_in_frames(getDataFrames);
  source.set_avrc_passthru_command_callback(handleButtons);

  source.start(bluetoothDeviceName);
  source.set_volume(127);

  hasGreeted = false;

  Serial.println("started bluetooth");
}
void stopConnection() {
  source.disconnect();
  source.end();

  hasGreeted = false;

  Serial.println("stopped bluetooth");
}
void connectionGreeting() {
  if (hasGreeted)
    return;

  hasGreeted = true;
  Serial.print("connection successful with: ");
  Serial.println(bluetoothDeviceName);
}

void setup() {
  Serial.begin(115200);

  esp_sleep_enable_timer_wakeup(0);
  esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); 

  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);

  Serial.println("initializing SD card");
  if (!SD_MMC.begin("/sdcard", true)) {
    Serial.println("SD card initialization failed");
    return;
  }
  Serial.println("SD card initialized");

  startConnection();
}
void loop() {
  if (source.is_connected()) {
    connectionGreeting();
  }
}
```

### Other Steps to Reproduce

The commented part where I just stream a white noise works properly.

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

2.0.18

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-06-21:
That's how the system behaves, if you don't send valid pcm data: Double check that your wav files are really 16 bits, 2 channels!
I suggest that you add some print statements and look at the data in the Serial Plotter.
 
The AudioTools project [contains plenty of working examples](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-communication/a2dp)

semihtuncer, 2025-06-22:
Thanks for the quick reply, but what I am concerned is that if I wire up another esp32 like (olimex esp32-poe-ios or esp32-d1mini) this exact code works. So I thought my audio files were correct. Do I need to decode the mp3 file with a tool and stream the raw data?

pschatzmann, 2025-06-23:
Please follow the advice given in [this Wiki](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working).
In addition use the logging to monitor what's going on:
- Activate the [ESP32 Logging to monitor A2DP](https://github.com/pschatzmann/ESP32-A2DP?tab=readme-ov-file#logging)
- Activate the [AudioTools logger](https://github.com/pschatzmann/arduino-audio-tools?tab=readme-ov-file#logging)

closed 2025-07-29 by pschatzmann