pschatzmann/ESP32-A2DP #723: set_avrc_rn_volumechange callback is very slow

Zipdox2, 2025-07-06:
### Problem Description

I registered a callback with set_avrc_rn_volumechange, but the callback is fired at most once per second. When I hold down the volume button on my phone to raise the volume from 0 to 127, I takes about 30 seconds for all the events to fire.

### Device Description

ESP-WROOM-32

### Sketch

```cpp
#include <BluetoothA2DPSink.h>

BluetoothA2DPSink bt_sink;

void read_data_stream(const uint8_t *data, uint32_t length){
  // Do nothing
}

void volume_change(int volume){
    Serial.printf("Volume: %d\n", volume);
}

void setup(void) {
    Serial.begin(115200);

    bt_sink.set_stream_reader(read_data_stream, false);
    bt_sink.set_avrc_rn_volumechange(volume_change);
    bt_sink.start("Speaker");
}

void loop(void) {

}
```

### Provide your Version of the EP32 Arduino Core (or the IDF Version)

3.2.1

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

Zipdox2, 2025-07-06:
Maybe it has something to do with the s_volume_lock mutex?

pschatzmann, 2025-07-07:
This is my understaning as well: the issue is with the AudioSource!